thesofproject/sof #1784: [BUG][SDW] 'DMIC16k: ASoC: trigger FE failed -110' then '0x60030000: GLB_STREAM_MSG: PCM_FREE' ipc timed out when do multi-pipeline test.

ClarexZhou, 2019-08-30:
**Describe the bug**
'DMIC16k: ASoC: trigger FE failed -110' then '0x60030000: GLB_STREAM_MSG: PCM_FREE' ipc timed out when do multi-pipeline test.

**To Reproduce**
1. Terminal A:   aplay -Dhw:0,0 -r48000 -c2 -fs32_le wav/long.wav -vv -i
2. Terminal B:   arecord  -Dhw:0,1 -r48000 -c2 -fs32_le 1.wav -vv -i
3. Terminal C:  arecord  -Dhw:0,3 -r48000 -c2 -fs16_le 1.wav -vv -i

**frequency**
2/2

**Actual result**
All pipeline stopped for Input/output error.  
'DMIC16k: ASoC: trigger FE failed -110' then '0x60030000: GLB_STREAM_MSG: PCM_FREE' ipc timed out 

**Expected result**
All pipelines should work ok.

**Environment**
linux: https://github.com/thesofproject/linux/commits/integration/soundwire-alpha commit: a8f3408
sof: https://github.com/thesofproject/sof/commits/sdw-001-drop-stable commit: 0223d94
Topology: file: sof-cml-rt700.m4 same path as firmware
platform: CML SDW RVP

**Error sof logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         DMIC         105296289.010417 105296288.000000 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296298.645833         9.635417 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296319.739583        21.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296330.833333        11.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296345.729167        14.895833 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296356.822917        11.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296371.666667        14.843750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296382.708333        11.041667 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296397.604167        14.895833 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296408.645833        11.041667 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296423.541667        14.895833 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296434.635417        11.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296449.479167        14.843750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296460.572917        11.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296475.468750        14.895833 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296486.562500        11.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296501.458333        14.895833 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296512.500000        11.041667 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296527.343750        14.843750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296538.385417        11.041667 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296553.281250        14.895833 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296564.375000        11.093750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296579.218750        14.843750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296590.260417        11.041667 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296605.104167        14.843750 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
    0      1         DMIC         105296616.145833        11.041667 rs/intel/cavs/dmic.c:1506	dmic_irq_handler() error: full fifo B or PDM overrrun
```
**dmesg**
```
[   52.938544] plb: soc_pcm_hw_params component name int-sdw.1
[   52.938545] plb: soc_pcm_hw_params component name sdw:1:25d:700:0:0
[   52.938546] plb: soc_pcm_hw_params component name 0000:00:1f.3
[   52.938548] plb: soc_pcm_hw_params codec_dai name snd-soc-dummy-dai
[   52.938549] plb: soc_pcm_hw_params cpu_dai name SDW1-mics 1
[   52.938550] plb: soc_pcm_hw_params component name 0000:00:1f.3
[   52.938552] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 1 dir 1
[   52.938557] sof-audio-pci 0000:00:1f.3: generating page table for 000000008191e47e size 0xffc0 pages 16
[   52.938567] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   52.938580] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   52.938582] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[   52.938582] sof-audio-pci 0000:00:1f.3: periods:4
[   52.938591] sof-audio-pci 0000:00:1f.3: stream_tag 2
[   52.938594] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[... 97 lines cut ...]
[   79.188688] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   79.188692] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   79.188707] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   79.188710] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   79.188784] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   79.692354] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   79.692373] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   79.692632] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   79.692649] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   79.692652] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   79.692667] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   79.692671] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   79.692683]  SDW1-mics: ASoC: trigger FE failed -110
[   79.692749] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   80.196383] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   80.196404] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   80.196663] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   80.196687] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   80.196691] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   80.196707] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask
```

[multipipline_ipcTimedout_logger.txt](https://github.com/thesofproject/sof/files/3558122/multipipline_ipcTimedout_logger.txt)
[multipipline_ipcTimedout_errorlogger.txt](https://github.com/thesofproject/sof/files/3558123/multipipline_ipcTimedout_errorlogger.txt)
[multipipline_ipcTimedout_dmesg.txt](https://github.com/thesofproject/sof/files/3558124/multipipline_ipcTimedout_dmesg.txt)

tlauda, 2019-10-25:
@ClarexZhou Do we still have a repro?

sinahuang, 2019-10-25:
issue not reproduce on CML RVP with codec ALC700 in SDW mode 
test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 1c3de3d 
FW: https://github.com/thesofproject/sof/commits/master commit: 89ef0af 
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: CML RVP with codec ALC700 in SDW mode

ClarexZhou, 2019-10-25:
Close for not reproduced

closed 2019-10-25 by ClarexZhou