thesofproject/sof #1785: [BUG][SDW]Aplay/arecord shows prepare error after switch to background then switch back foreground.

YvonneYang2, 2019-08-30:
**Describe the bug**
aplay/arecord will show prepare error after switch to background then switch back foreground. Dmesg shows  SDW0-Playback: ASoC: backend prepare failed -22. Only reboot can recover the issue.

**To Reproduce**

1.aplay -Dhw:0,0 -r48000 -c2 -fs16_le 1.wav -vv -i or arecord -Dhw:0,1 -r48000 -c2 -fs32_le 1.wav -vv -i
2.Do "ctrl+z" to backgound, then do "fg" to foreground
3. Repeat step2

**frequency**
100%

**Expected result**
aplay normally after back to foregeound

**Actual result**
Console shows prepare error after tried to back to foreground

**Environment**
Firmware: 0223d94 https://github.com/thesofproject/sof/commits/sdw-001-drop-stable 
+PR https://github.com/thesofproject/sof/pull/1781
Kernel: a8f3408 https://github.com/thesofproject/linux/commits/integration/soundwire-alpha 

**dmesg**
```
[  308.982274] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[  308.982294] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  308.982301] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  308.982507] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  308.982508] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  308.982852] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  308.982855] plb: soc_pcm_trigger component name snd-soc-dummy
[  308.982859] plb: soc_pcm_trigger cpu_dai name SDW0-speakers 0
[  308.982863] plb: soc_pcm_trigger codec_dai name rt700-aif1
[  308.982866] plb: soc_pcm_trigger component name int-sdw.0
[  308.982869] plb: soc_pcm_trigger component name sdw:0:25d:700:0:0
[  308.982871] plb: soc_pcm_trigger component name 0000:00:1f.3
[  308.982874] plb: soc_pcm_trigger cpu_dai name SDW0 Pin2
[  309.580984] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  309.581025] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  309.681517] int-sdw int-sdw.0: Compute bus params failed: -22
[  309.681528] int-sdw int-sdw.0: Compute params failed: -22
[  309.681532] Prepare for stream:SDW0 Pin2-Playback failed: -22
[  309.681537] int-sdw int-sdw.0: ASoC: cpu DAI prepare error: -22
[  309.681541]  SDW0-Playback: ASoC: backend prepare failed -22
[  309.682387] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  309.682969] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  309.683194] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
[  310.115857] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  310.115898] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  313.013833] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  313.056218] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  313.056675] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  313.056819] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  313.056861] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[  313.056871] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  313.057051] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
```
[pause_release_dmesg.log](https://github.com/thesofproject/sof/files/3558171/pause_release_dmesg.log)
[pause_release_logger.log](https://github.com/thesofproject/sof/files/3558172/pause_release_logger.log)

bardliao, 2019-09-06:
@YvonneYang2 This is actually a kernel issue.

closed 2019-09-06 by bardliao

YvonneYang2, 2019-09-06:
Will track in https://github.com/thesofproject/linux/issues/1178