thesofproject/sof #1789: [BUG][cml] Paplay has no audio output when playing captured fs24_le audio.

ClarexZhou, 2019-08-30:
**Describe the bug**
Paplay has no audio output when playing captured fs24_le audio. Audio can be played via aplay successfully.

**To Reproduce**
1. arecord -Dhw:0,1 -r48000 -c2 -fs24_le 1.wav -vv -i
2. paplay 1.wav

**frequency**
100%

**Actual result**
Paplay has no audio output when playing captured fs24_le audio. Audio can be played via aplay successfully.

**Expected result**
audio should be played normally.

**Environment**
linux: https://github.com/thesofproject/linux/commits/integration/soundwire-alpha commit: a8f3408
sof: https://github.com/thesofproject/sof/commits/sdw-001-drop-stable commit: 0223d94
Topology: file: sof-cml-rt700.m4 same path as firmware
platform: CML SDW RVP

No error messages.
[paplay_soflogger.txt](https://github.com/thesofproject/sof/files/3558271/paplay_soflogger.txt)
[paplay_dmesg.txt](https://github.com/thesofproject/sof/files/3558272/paplay_dmesg.txt)

aiChaoSONG, 2019-09-02:
paplay needs correct ucm to funciton well, run `pavucontrol` to check if there is a playback device.

ClarexZhou, 2019-09-02:
There is a playback device.  Update result to "volume is too low for fs24_le audio via paplay", this audio can output normally via aplay.  Also fs16_le and fs32_le audio can be output via paplay normally.

lgirdwood, 2019-09-06:
Low impact since it's paplay only.

plbossart, 2019-09-06:
this should be fixed with the latest soundwire firmware, the hardware does require the data to be 32-bit MSB aligned and it's now handled. without the firmware change, the sound would be attenuated by 6.02*8=48dB...

slawblauciak, 2019-09-09:
This should no longer occur. Please confirm

ClarexZhou, 2019-09-12:
issue still seen 
**Environment**
linux: https://github.com/thesofproject/linux/tree/topic/soundwire commit: bccf4a8a
sof: https://github.com/thesofproject/sof/commits/sdw-001-drop-stable commit: d9c0463
Topology: file: sof-cml-rt700.m4 same path as firmware
platform: CML SDW RVP