thesofproject/sof #1793: [BUG] sof memory allocation error with Wov topology

aiChaoSONG, 2019-09-03:
**Describe the bug**
in testing wov with kernel `topic/sof-dev` and sof `master`,  there will be a mem allocation error, after running arecord, make some sound, there will be a *wrong host params* error in the sof-logger output. these errors may cause the issue of wov cannot triggered. 

**To Reproduce**
1. run sof-logger `sudo sof-logger -l sof-cnl.ldc -t`
2. run arecord, `arecord -Dhw:0,8 -M -N -r16000 -c2 -f s16_le -vvv -d5 wov.wav`
3. you will see error message.(with `sudo sof-logger -l sof-cnl.ldc` you can see the message clearly)

**Reproduction Rate**
100%

**Expected behavior**
memory should be allocated successfully

**Environment**
1) Kernel: `topic/sof-dev f95e4ebf`
2) sof: `master 90873c5d`
3) Topology: sof-cml-rt5682-max98357a.tplg
4) Platform: CML Chrome

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM          58674154.270833  58674156.000000      src/lib/alloc.c:273 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1          KPB          72877593.125000  14203439.000000      src/audio/kpb.c:852 	kpb_init_draining() error: wrong host params.
```

mrajwa, 2019-09-03:
@aiChaoSONG, the allocator error is expected although we have to think about making it more warning like.
As for 
> kpb_init_draining() error: wrong host params. 
I see you haven't specified --period-size in arecord so it uses the default one and FW calculated that this is not sufficient to drain the data. Try with --period-size=400 for example.

aiChaoSONG, 2019-09-03:
closed for invalid

closed 2019-09-03 by aiChaoSONG