thesofproject/sof #1796: [FEATURE] DCR implementation API discussion

cujomalainey, 2019-09-03:
**Is your feature request related to a problem? Please describe.**
Enable DC Removal filters in SOF

**Describe the solution you'd like**
This bug is to figure out the fine details of the implementation

**Describe alternatives you've considered**
N/A

**Additional context**
@singalsu figured we could figure out the details on the API and implementation. Then we can assign this back to me to track implementation. I was curious what you think would be the best implementations for this in terms of API and controls.
e.g.
* integer/binary controls
* uniform or independent channel control for the filter parameter

singalsu, 2019-09-05:
The simplest way for DCR would be to use the processing component type with binary controls. A separate tool or if simple task a m4 macro within topology could create the blob data. 

This approach would need only add of SOF_COMP_DCR into sof_comp_type enums and SOF_PROCESS_DCR in sof_ipc_process_type enums in topology.h and a line to sof_process[] array initilization in topology.c.

lgirdwood, 2019-09-06:
@lbetlej is there an existing API in CAVS ?

lgirdwood, 2019-09-06:
@cujomalainey do you have a schedlue in mind ?  1.5 / 1.6 timeframe ?
@mengdonglin @michalgrodzicki fyi

cujomalainey, 2019-09-06:
I do not currently, my fall is looking like it is going to be quite busy, I'm hoping early in the new year.

cujomalainey, 2019-09-20:
We are working on a plan to get this upstream, will fill you in @lgirdwood at ELC

closed 2020-12-03 by cujomalainey