thesofproject/sof #1800: [BUG] IPC error -22 on tigger_pause on topology with DEMUX component

zrombel, 2019-09-04:
**Describe the bug**
Topology:
```
        pipe_cap
    +-------------------------------+
    | +------+   +------+   +-----+ |   +------+
    | | Host |<--| Buff |<--| Dai |<----| SSPx |<--------------------------<
    | +------+   +------+   +-----+ |   +------+                           |
    +-------------------------------+                                      |
                                                  pipe_plb                 |
                                            +---------------------+        |
                                            | +------+   +------+ |     +------+
                                         +--->| Buff |-->| Dai1 |------>| SSPx |
                                         |  | +------+   +------+ |     +------+
         pipe_plb                        |  +---------------------+
    +---------------------------------+  |
    | +------+   +------+   +-------+>---+
    | | Host |-->| Buff |-->| DeMux | |
    | +------+   +------+   +------+>----+
    +---------------------------------+  |        pipe_plb
                                         |  +---------------------+
                                         |  | +------+   +------+ |     +--------+
                                         +--->| Buff |-->| Dai2 |------>| SSPx+1 |
                                            | +------+   +------+ |     +--------+
                                            +---------------------+        |
        pipe_cap                                                           |
    +-------------------------------+                                      |
    | +------+   +------+   +-----+ |   +--------+                         |
    | | Host |<--| Buff |<--| Dai |<----| SSPx+1 |<------------------------<
    | +------+   +------+   +-----+ |   +--------+
    +-------------------------------+
```
Description:
After successful play and capture on all pipelines and sending first trigger_pause to pipeline containing DAI1, when sending trigger_pause to pipeline containing DAI2 FW response with error -22. In trace log this entries can be found

Trace log:
```
    0      2          IPC         112046857.552083       397.864594 /src/ipc/handler.c:452 	ipc: comp 8 -> trigger cmd 0x60000
    0      2         PIPE 3.10    112046862.500000         4.947917 /src/audio/pipeline.c:634 	pipeline_trigger()
    0      2          DAI 3.8     112046866.510417         4.010417 /src/audio/dai.c:483 	dai_comp_trigger(), command = 2
    0      2          DAI 3.8     112046870.572917         4.062500 /src/audio/dai.c:547 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         112046874.166667         3.593750 /src/drivers/dw/dma.c:465 	dw_dma_stop(): dma 0 channel 0 stop
    0      2          SSP         112046887.083333        12.916667 /src/drivers/intel/cavs/ssp.c:877 	ssp_trigger() cmd 0
    0      2          SSP         112046935.000000        47.916668 /src/drivers/intel/cavs/ssp.c:858 	ssp_stop(), TX stop
    0      1          DAI 4.11    112047396.458333       461.458344 /src/audio/dai.c:575 	dai_check_for_xrun() error: underrun due to no data available
    0      1         COMP         112047401.718750         5.260417 /src/include/sof/audio/component.h:742 	comp_underrun() error: dev->comp.id = 11, source->avail = 0, copy_bytes = 0
    0      2         PIPE 4.13    112047407.291667         5.572917 /src/audio/pipeline.c:634 	pipeline_trigger()
    0      2          DAI 4.11    112047411.145833         3.854167 /src/audio/dai.c:483 	dai_comp_trigger(), command = 8
    0      2          DAI 4.11    112047415.000000         3.854167 /src/audio/dai.c:541 	dai_comp_trigger(), XRUN
    0      2          DAI 4.11    112047418.281250         3.281250 /src/audio/dai.c:547 	dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         112047421.770833         3.489583 /src/drivers/dw/dma.c:465 	dw_dma_stop(): dma 1 channel 0 stop
    0      2          SSP         112047433.906250        12.135417 /src/drivers/intel/cavs/ssp.c:877 	ssp_trigger() cmd 0
    0      2          SSP         112047480.989583        47.083332 /src/drivers/intel/cavs/ssp.c:858 	ssp_stop(), TX stop
    0      1         PIPE         112047486.979167         5.989583 /src/audio/pipeline.c:790 	pipeline_copy() error: ret = -61
    0      1         PIPE 4.13    112047491.927083         4.947917 /src/audio/pipeline.c:1005	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      2          IPC         112056605.364583      9113.437500 /src/ipc/handler.c:452 	ipc: comp 11 -> trigger cmd 0x60000
    0      2         PIPE 4.13    112056610.572917         5.208333 /src/audio/pipeline.c:634 	pipeline_trigger()
    0      2          DAI 4.11    112056614.479167         3.906250 /src/audio/dai.c:483 	dai_comp_trigger(), command = 2
    0      1         COMP         112056618.281250         3.802083 /src/audio/component.c:182 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1          IPC         112056623.489583         5.208333 /src/audio/pipeline.c:661 	pipeline_trigger() error: ret = -22, host->comp.id = 11, cmd = 2
    0      1          IPC         112056629.062500         5.572917 /src/ipc/handler.c:486 	ipc: comp 11 trigger 0x60000 failed -22
```

**Reproduction Rate**
5%

**Expected behavior**
Stream should pause and stop gracefully. 

**Environment**
1) Found on FW Master branch commit 90873 and erlier 
2) Tested with slim driver. Test: TestDemuxSimple8000Hz24b32b2ch

lgirdwood, 2019-09-08:
@zrombel can you update the diagram and desciption to number your PCM's and SSPs to make it easy to follow. Thanks.

zrombel, 2019-09-09:
Sure, pipelines id and component id on schematics matches log.

jdrzaszcz, 2020-03-12:
Still occur on master branch, commit: b5eb50df955dabc7e1a34c757ae5662c620d6dde

zrombel, 2020-04-21:
According to @tlauda suggestions test were changed and at this point trigger_start, trigger_pause and trigger_stop is only send to PIPE 5 and then internally propagated by FW to PIPE 3 and 4. After test changes bug cannot be reproduced. Closing Issue.

closed 2020-04-21 by zrombel