thesofproject/sof #1802: [BUG][ICL]show I/O error when unplug headset during capture

dengyangchao, 2019-09-05:
**Describe the bug**
show I/O error when unplug headset during capture

**To Reproduce**
1. Reboot system via "sudo reboot"
2. Wait it to suspend status
3. Arecord via "arecord -Dhw:0,0 -r48000 -c2 -fs16_le 1.wav -vv -i"
4. Unplug headset

**Reproduce rate**
100%

**Expected behavior**
the arecord can still process

**Impact**
show I/O error

**Environment**
Firmware: 2036b0f0 https://github.com/thesofproject/sof/commits/master
Kernel: 45c901ba https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-4ch.tplg same as firmware
platform：ICL-RVP

**dmesg**
```
[  142.491709] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x50010000: GLB_COMP_MSG: SET_VALUE
[  142.491712] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  142.491774] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  142.721831] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 1
[  142.721835] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  142.721837] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  142.721839] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  142.722188] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  142.722396] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  142.722467] sof-audio-pci 0000:00:1f.3: format_val=49, rate=48000, ch=2, format=10
[  142.722477] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 0 dir 1
[  142.722481] sof-audio-pci 0000:00:1f.3: generating page table for 00000000cdd50e7e size 0xffc0 pages 16
[  142.722506] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  142.722551] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  142.722556] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  142.722557] sof-audio-pci 0000:00:1f.3: periods:4
[  142.722596] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  142.722611] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  142.722919] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  142.722950] sof-audio-pci 0000:00:1f.3: pcm: stream dir 1, posn mailbox offset is 790528
[  142.726153] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 1
[  142.726468] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  142.726473] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  142.726863] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  142.726872] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  142.990027] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  142.990080] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  143.489983] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  143.490039] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  152.489876] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  152.489929] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  153.224952] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 0
[  153.225510] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  153.225535] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  153.225753] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  153.225763] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  153.225995] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  153.226006] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  153.226019] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  153.226278] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  153.226321] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  153.226563] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  153.226664] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 1
[  153.989843] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  153.989896] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  155.727500] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  155.764669] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  155.764903] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  155.764953] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  155.764957] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[  155.764959] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  155.764996] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
```
[dmesg.log](https://github.com/thesofproject/sof/files/3578071/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3578072/logger.log)

lgirdwood, 2019-09-06:
@dengyangchao looks like kernel error since FW has no interaction with headset IRQ. Can you paste cmd line output to confirm.

dengyangchao, 2019-09-09:
[image]

it seems kernel issue, so close it and open kernel issue to track

closed 2019-09-09 by dengyangchao

dengyangchao, 2019-09-09:
https://github.com/thesofproject/linux/issues/1187  track on #1187

dengyangchao, 2019-09-10:
@lgirdwood  add sof error log, please check again
```
  0      1          DAI 2.13    101815179.947917 101815176.000000      src/audio/dai.c:609 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         101815189.531250         9.583333 of/audio/component.h:758 	comp_overrun() error: dev->comp.id = 13, sink->free = 768, copy_bytes = 0
    0      1         PIPE         101815238.489583        48.958332 src/audio/pipeline.c:811 	pipeline_copy() error: ret = -61, start->comp.id = 13, dir = 0
    0      1         PIPE 2.14    101815249.687500        11.197917 src/audio/pipeline.c:994 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP         105220599.791667   3405350.000000 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
```