thesofproject/sof #1811: [BUG][BYT]Capture failed (arecord: set_params:1363: Unable to install hw params) on byt nocodec platform.

Jiangxinx, 2019-09-10:
**Describe the bug**
arecord: set_params:1363: Unable to install hw params on byt nocodec platform.

**To Reproduce**
arecord -Dhw:0,0 -fs16_LE -c2 -r 48000 1.wav -vv -i

**Reproduction Rate**
100%

**output**
```
Recording WAVE '1.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
arecord: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: 4
BUFFER_TIME: 340000
BUFFER_SIZE: 16320
BUFFER_BYTES: 65280
TICK_TIME: 0
```

**dmesg**
```
[   16.963364] systemd-journald[222]: File /var/log/journal/810813fff71a4647924f113b7ac7bf25/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   18.798220] igb 0000:03:00.0 enp3s0: igb: enp3s0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[   18.798784] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[   81.398328] sof-audio-acpi 80860F28:01: pcm: open stream 0 dir 1
[   81.398336] sof-audio-acpi 80860F28:01: period min 192 max 16384 bytes
[   81.398340] sof-audio-acpi 80860F28:01: period count 2 max 4
[   81.398343] sof-audio-acpi 80860F28:01: buffer max 65536 bytes
[   81.399814] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[   81.399821] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[   81.399828] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[   81.399831] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[   81.399836] sof-audio-acpi 80860F28:01: rate_min: 48000 rate_max: 48000
[   81.399839] sof-audio-acpi 80860F28:01: channels_min: 2 channels_max: 2
[   81.399848] sof-audio-acpi 80860F28:01: pcm: hw params stream 0 dir 1
[   81.399867] sof-audio-acpi 80860F28:01: generating page table for 0000000090564e99 size 0xff00 pages 16
[   81.399873] sof-audio-acpi 80860F28:01: stream_tag 0
[   81.399884] sof-audio-acpi 80860F28:01: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   81.400514] sof-audio-acpi 80860F28:01: error: ipc error for 0x60010000 size 20
[   81.400520] sof-audio-acpi 80860F28:01: error: hw params ipc failed for stream 0
[   81.400525] sof-audio-acpi 80860F28:01: ASoC: 80860F28:01 hw params failed: -12
[   81.400530]  Low Latency: ASoC: hw_params FE failed -12
[   81.407333] sof-audio-acpi 80860F28:01: pcm: close stream 0 dir 1
[   81.505831] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   81.505856] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   82.005966] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[   82.005989] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

**sof-logger**
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM          81248091.875000  81248088.000000      src/lib/alloc.c:273 	error: 8 blocks needed for allocation but only 6 blocks are remaining
    0      1       BUFFER          81248101.718750         9.843750   src/audio/buffer.c:87  	resize error: can't alloc 768 bytes type 97
    0      1          DAI 2.16     81248111.666667         9.947917      src/audio/dai.c:349 	dai_capture_params() error: buffer_set_size() failed to resize buffer. dai_config->periods_sink = 2; period_bytes = 1536; buffer_size = 3072; dd->dma_buffer->alloc_size = 768
    0      1         PIPE          81248124.270833        12.604167 src/audio/pipeline.c:351 	pipeline_params() error: ret = -12, host->comp.id = 12
    0      1          IPC          81248133.854167         9.583333    src/ipc/handler.c:296 	ipc: pipe 2 comp 12 params failed -12
```

**Environment**
linux: https://github.com/thesofproject/linux/commits/integration/soundwire-alpha commit: fa20ceba
sof: https://github.com/thesofproject/sof/commits/sdw-001-drop-stable commit: 596f761b
Topology: file: sof-byt-nocodec.tplg
platform: BYT

Jiangxinx, 2019-09-10:
Update the new reproduce step:
It only can be reproduced before doing any playback test. 

**bisect result**
Caused by #1710

ClarexZhou, 2019-09-10:
@tlauda , seems one regression by #1710, could you please check on this?

Jiangxinx, 2019-09-11:
Can not be reproduced with #1816 .

PR #1818 (2019-09-11): topology: fix wrongly used SCHEDULE_DEADLINE
SCHEDULE_DEADLINE is replaced with SCHEDULE_PERIOD, but there is one
missed in DAI_ADD for W_PIPELINE.

Fixes #1811.
closed 2019-09-11 by jajanusz