thesofproject/sof #1820: [BUG] Huge distortion - Demux tests with ch.mapping

emilchudzik, 2019-09-11:
**Describe the bug**
Distortion occurs on whole recording wave in tests with Demux based on channel mapping.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug) 

**Reproduction Rate**
2% from 50 executions.

**Expected behavior**
clean recording wave without distortion

**Environment**
Windows python tests
Platform WHL-RVP - I2S mode.
Firmware from branch:
abonislawski@8dfdfe1

```
        pipe_cap
    +-------------------------------+
    | +------+   +------+   +-----+ |   +------+
    | | Host |<--| Buff |<--| Dai |<----| SSPx |<--------------------------<
    | +------+   +------+   +-----+ |   +------+                           |
    +-------------------------------+                                      |
                                                  pipe_plb                 |
                                            +---------------------+        |
                                            | +------+   +------+ |     +------+
                                         +--->| Buff |-->| Dai1 |------>| SSPx |
                                         |  | +------+   +------+ |     +------+
         pipe_plb                        |  +---------------------+
    +---------------------------------+  |
    | +------+   +------+   +-------+>---+
    | | Host |-->| Buff |-->| DeMux | |
    | +------+   +------+   +------+>----+
    +---------------------------------+  |        pipe_plb
                                         |  +---------------------+
                                         |  | +------+   +------+ |     +--------+
                                         +--->| Buff |-->| Dai2 |------>| SSPx+1 |
                                            | +------+   +------+ |     +--------+
                                            +---------------------+        |
        pipe_cap                                                           |
    +-------------------------------+                                      |
    | +------+   +------+   +-----+ |   +--------+                         |
    | | Host |<--| Buff |<--| Dai |<----| SSPx+1 |<------------------------<
    | +------+   +------+   +-----+ |   +--------+
    +-------------------------------+
```

**Screenshots or console output**
[image]

**Attachments**
[dma_trace_1.log](https://github.com/thesofproject/sof/files/3601131/dma_trace_1.log)

lgirdwood, 2019-09-11:
@emilchudzik looks like clipping and saturation. Can you try and lower the volume on the CODEC. @singalsu do you agree ?

emilchudzik, 2019-10-28:
@lgirdwood @singalsu this test didn't used the codec. I made loopback on SSP using quirk: 
/* ssc1: LBM */
#define SOF_DAI_INTEL_SSP_QUIRK_LBM		(1 << 6)

There is 1 playback pipeline Host-Buffer-Demux, after it there is 2 playback pipelines where demux redirect audio signal like a mirror. It should be the same signal on both Dai's.
Then on SSP there is a internal loopback.
Reproduce is very rare.

plbossart, 2019-10-28:
you should use a saw-tooth pattern with a fixed increment for each sample, so that we can see what the distorsion really is. From the waveform it's likely a buffer copied from the wrong location or a DMA setup issue.

lgirdwood, 2019-10-28:
@emilchudzik chan0 above looks like R/W pointer sync issues, could be demux to DAI buffer. One other thing to you hear the distorion on the output or just on the capture. This could point to issue on capture DAI buffer.

slawblauciak, 2020-01-28:
@emilchudzik is the issue still relevant?

emilchudzik, 2020-02-03:
This issue doesn't occur on: https://github.com/thesofproject/sof/pull/2212

closed 2020-02-03 by emilchudzik