thesofproject/sof #1830: [BUG][CML][CFL]ipc timed out for 0x80010000 size 144 when doing module unload/reload

dengyangchao, 2019-09-16:
**Describe the bug**
ipc timed out for 0x80010000 size 144 when doing module unload/reload, 
caused by master 8c61213c

**To Reproduce**

1.Disable Pulseaudio via "sudo mv /usr/bin/pulseaudio /usr/bin/pulseaudio.bak"
2.Do system reboot via "sudo reboot"
3.Wait till runtime PM status is suspned
4.run ./sof_bootloop.sh

**frequency**
1/40

**Actual result**
ipc timed out for 0x80010000 size 144

**Expected result**
Module reload should be successfully.

**Environment**
Firmware: 8c61213c https://github.com/thesofproject/sof/commits/master
Kernel: fa20ceb https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware
platform：CML

**dmesg**
```
[ 1437.114084] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1437.114101] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1437.114225] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1437.114244] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1437.114633] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1437.114651] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 1437.617352] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 144
[ 1437.617366] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 1437.617455] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 1437.617465] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 1437.617469] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1437.617476] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 1437.617479] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 1437.617485] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for dmic16k
[ 1437.617489] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -110
[ 1437.618056] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[ 1437.618288] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 1437.618302] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618309] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618334] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618346] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618353] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618376] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618389] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618394] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618417] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618430] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618436] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618458] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618470] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618476] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618497] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618509] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618515] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618537] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618549] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618554] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618576] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618588] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618594] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.618616] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618629] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.618634] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1437.658507] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.658515] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[ 1437.658519] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[ 1437.658660] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power off
[ 1448.261572] sof-audio-pci 0000:00:1f.3: PCI DSP detected
[ 1448.261795] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[ 1448.261801] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[ 1448.261842] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
[ 1448.261843] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
[ 1448.261845] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
[ 1448.261846] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
[ 1448.261850] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
[ 1448.261851] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
[ 1448.261854] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
[ 1448.261855] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
[ 1448.261858] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
[ 1448.261859] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
[ 1448.261866] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
[ 1448.261869] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
[ 1448.261870] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
[ 1448.261932] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[ 1448.261932] sof-audio-pci 0000:00:1f.3: using HDA IRQ 147
[ 1448.261961] sof-audio-pci 0000:00:1f.3: using IPC IRQ 147
[ 1448.261986] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
```
[dmesg.log](https://github.com/thesofproject/sof/files/3614657/dmesg.log)

dengyangchao, 2019-09-16:
issue also can reproduce on CFL-S RVP

dengyangchao, 2019-09-16:
issue observed on CML and CFL-S RVP 
when doing stress test via"aplay -Dhw:0,0 -r48000 -c2 -fs16_le /dev/zero & sudo ./stress-s3.sh 300 1 1"
**recipe**
Firmware: 2bd725df https://github.com/thesofproject/sof/commits/master
Kernel: 6cf9ef05 https://github.com/thesofproject/linux/commits/topic/sof-dev

```
[  107.221925] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  107.222276] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  107.222290] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  107.725884] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 144
[  107.725893] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  107.725981] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  107.725989] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[  107.725992] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  107.725997] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  107.725999] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  107.726003] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for dmic16k
[  107.726006] sof-audio-pci 0000:00:1f.3: error: failed to restore pipeline after resume -110
[  107.726015] PM: dpm_run_callback(): pci_pm_resume+0x0/0xa0 returns -110
[  107.726026] PM: Device 0000:00:1f.3 failed to resume: error -110
[  107.726101] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[  107.726112] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[  107.726116] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[  108.763334] OOM killer enabled.
[  108.763336] Restarting tasks ... done.
[  108.772495] sof-audio-pci 0000:00:1f.3: hda: prepare stream dir 0
[  108.772502] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  108.799814] PM: suspend exit
[  109.277783] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  109.277795] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  109.277959] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  109.277972] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[  109.277975] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  109.277983] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  109.277986] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  109.277998] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[  109.278001] sof-audio-pci 0000:00:1f.3: ASoC: cpu DAI prepare error: -110
[  109.278005]  Analog Playback and Capture: ASoC: backend prepare failed -110
```
[s3_test_3_logger_all.log](https://github.com/thesofproject/sof/files/3614822/s3_test_3_logger_all.log)
[test_3.log](https://github.com/thesofproject/sof/files/3614823/test_3.log)

wenqingfu, 2019-09-16:
> caused by master [8c61213](https://github.com/thesofproject/sof/commit/8c61213c37e6de534b5707ff5a644390549d88ed)

@tlauda

lgirdwood, 2019-09-17:
@dengyangchao is this survivavble when FW is reloaded ?

dengyangchao, 2019-09-18:
> @dengyangchao is this survivavble when FW is reloaded ?

@lgirdwood when issue occur, then use```sof_remove.sh```and```sof_insert.sh```, it back to normal

dengyangchao, 2019-09-19:
run 200 times the issue not reproduce on CFL-S RVP when doing unload/reload

**recipe**
Firmware: 460c41aa https://github.com/thesofproject/sof/commits/master
Kernel: 6fc7eb30 https://github.com/thesofproject/linux/commits/topic/sof-dev

dengyangchao, 2019-09-20:
the issue reproduced on CFL-S RVP when doing module unload/reload
**frequency**
1/34
**recipe**
Firmware: b0bb5a33 https://github.com/thesofproject/sof/commits/master
Kernel: 6fc7eb30 https://github.com/thesofproject/linux/commits/topic/sof-dev

YvonneYang2, 2019-09-26:
Issue can be reproduced on ICL RVP with codec ALC700 in SDW mode when doing module unload/reload

**frequency**
1/85

Test recipe
Firmware: 3ce82a1 https://github.com/thesofproject/sof/commits/ +PR1851 (For HDMI tplg)
Kernel: abc49c2 https://github.com/thesofproject/linux/commits/integration/soundwire-latest
[boot_fail.log](https://github.com/thesofproject/sof/files/3656560/boot_fail.log)

dengyangchao, 2019-09-27:
issue can be reproduced on GLK-chrome when doing module unload/reload

**frequency**
1/73

**test recipe**
Firmware: 7046e849 https://github.com/thesofproject/sof/commits/master
Kernel: 578ff8d7 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware

[boot_fail.log](https://github.com/thesofproject/sof/files/3661097/boot_fail.log)

wwittbrx, 2019-10-01:
I cannot reproduce this issue with https://github.com/thesofproject/sof/pull/1887 applied. Please check it on your side.

dengyangchao, 2019-10-08:
issue can not reproduce on CML, CFL, ICL-RVP HDA and GLK-Chrome, so close the issue

closed 2019-10-08 by dengyangchao