thesofproject/sof #1835: [FEATURE/BUG]Testbench build for debug does not produce debug friendly code

singalsu, 2019-09-17:
**Is your feature request related to a problem? Please describe.**
The cmake build of testbench does not generate an executable without optimization. When run in debugger (gdb + graphical front end) the optimized executable has variables optimized out and code stepping is not linear. The hand editing of makefiles manually to remove code optimization is cumbersome due to complexity of cmake generates makefiles.

**Describe the solution you'd like**
When running command "cmake -DCMAKE_BUILD_TYPE=Debug .." there should be no -O3 optimizations left anywhere in makefiles (testbench, pipeline & components code). Just "-O" might be sufficient but to be safe just leave "-g".

**Describe alternatives you've considered**
No alternates besides hand editing of makefiles.

**Additional context**
N/A

singalsu, 2020-12-03:
This is tricky, need a cmake expert here.

ShriramShastry, 2020-12-03:
grep -rl "\-O3" . | xargs sed -i 's/\-O3/\-O/g'

ShriramShastry, 2020-12-04:
### Tested with below command for all -O3 replacement in all files 
grep -rl "\-O3" . | xargs sed -i 's/\-O3/\-O/g'
### Tested with below command for all -O3 replacement for CMakeLists.txt files
find . -name "CMakeLists.txt" | xargs grep -rl "\-O3" | xargs sed -i 's/\-O3/\-O/g'

lgirdwood, 2020-12-04:
@juimonen can you help here, cmake support conditional logic so you can pass in -DEBUG or similar on the cmd line to select different CFLAGS.

marc-hb, 2022-09-21:
This happens simply because we have `-O3` hardcoded in various `CMakeLists.txt` files, for the testbench this was done in commit 56228b0221dc. The first step is this, try it and see what happens:

```diff
--- a/tools/testbench/CMakeLists.txt
+++ b/tools/testbench/CMakeLists.txt
@@ -30,7 +30,7 @@ target_include_directories(testbench PRIVATE
 )
 endif()
 
-target_compile_options(testbench PRIVATE -g -O3 -Wall -Werror -Wl,-EL -Wmissing-prototypes
+target_compile_options(testbench PRIVATE -g -Wall -Werror -Wl,-EL -Wmissing-prototypes
   -Wimplicit-fallthrough -DCONFIG_LIBRARY -imacros${config_h})
 
 target_link_libraries(testbench PRIVATE -ldl -lm)
```

@ShriramShastry : https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

marc-hb, 2022-09-21:
To see which flags are used without being flooded by the build logs:

```
$ touch tools/testbench/testbench.c 
make -C tools/testbench/build_testbench/ testbench.o VERBOSE=1

```

marc-hb, 2022-09-21:
PS: debugging -O code is madness. Don't do it. Ever.

marc-hb, 2022-09-21:
Also try this:
```
make -C tools/testbench/build_testbench/sof_ep/build/ menuconfig
```

lgirdwood, 2022-09-21:
> @marc-hb are you able to take care of a global `-g` and `-pg` setting for building all code. The latter is for building on host. I'm assuming this would be done in Zephyr Kconfig ?

Sorry, reason I ask as its not propagated through all tools/code and I've no idea if we inherit from Zephyr.

marc-hb, 2022-09-21:
@lgirdwood this feels like significant CMake surgery and I have no idea how long it would take, sorry.
My untested, testbench-specific advice above was just off the top of my head and after looking at it for 5 minutes.

lgirdwood, 2022-09-21:
@marc-hb ok, how about if we make it consistent when we invoke cmake or make e.g. ```make DEBUG=1``` just something quick and easy for the tools developers.

closed 2025-07-04 by github-actions[bot]