thesofproject/sof #1839: [BUG][SDW]aplay 0,0 + aplay 0,2 or aplay 0,3 + arecord 0,1 will cause ipc timed out 6005000 size 12

YvonneYang2, 2019-09-18:
**Describe the bug**
aplay 0,0 + aplay 0,2 or aplay 0,3 + arecord 0,1 will cause ipc timed out 6005000 size 12, other functions are ok .

**To Reproduce**
1.Boot up system 
2.Run ./rt1308_mixer.sh, ./rt711_mixer.sh and ./rt715_mixer.sh
3.open a terminal to run aplay -Dhw0,0 test.wav 
4.Open another terminal to run aplay -Dhw0,2 test.wav

**frequency**
100%

**Expected result**
Aplay normally

**Actual result**
Aplay stopped and dmesg showed  ipc timed out 6005000 size 12

**Environment**
Firmware: bccf4a8a https://github.com/bardliao/linux/tree/topic/for-3-in-1 
Kernel: 54e8eb5 https://github.com/thesofproject/sof/commits/sdw-001-drop-stable
tplg: sof-icl-rt711-rt1308-rt715-hdmi.tplg   rename sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

**dmesg**
```
[   70.110109] plb: soc_pcm_trigger cpu_dai name SDW0 Pin2
[   71.785506] plb: soc_pcm_trigger codec_dai name snd-soc-dummy-dai
[   71.785516] plb: soc_pcm_trigger component name 0000:00:1f.3
[   71.785528] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 2 dir 0 cmd 0
[   71.785834] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[   71.785854] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   71.833463] plb: soc_pcm_trigger codec_dai name snd-soc-dummy-dai
[   71.833472] plb: soc_pcm_trigger component name 0000:00:1f.3
[   71.833483] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 0
[   71.833837] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[   72.289169] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   72.289193] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   72.289354] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   72.289378] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   72.289381] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   72.289401] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   72.289404] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   72.289422]  SDW1-speakers: ASoC: trigger FE failed -110
[   72.289426] sof-audio-pci 0000:00:1f.3: error: trace IO error
[   72.289516] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 0
[   72.289524] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   72.793277] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   72.793301] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   72.793485] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   72.793509] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   72.793512] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   72.793533] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   72.793535] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   72.793638] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   73.297157] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   73.297182] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   73.297380] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   73.297405] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   73.297408] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   73.297426] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   73.297429] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   73.297440]  SDW0-speakers: ASoC: trigger FE failed -110
[   73.297546] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   73.801193] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   73.801216] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   73.801389] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   73.801413] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   73.801417] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   73.801435] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[... 20 lines cut ...]
[   75.009429] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   75.009434] sof-audio-pci 0000:00:1f.3: error: failed to free stream for link 0 dai->id 2
[   75.009438] int-sdw int-sdw.0: intel_free_stream: failed -110
[   75.009510] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 0
```
**error logger**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          IRQ         364903013.802083 364903008.000000 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364907362.656250      4348.854004 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364911710.364583      4347.708496 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364916060.000000      4349.635254 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364920408.333333      4348.333496 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364924756.458333      4348.125000 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364929104.010417      4347.552246 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364933452.239583      4348.229004 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364937799.375000      4347.135254 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364942147.187500      4347.812500 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364946496.406250      4349.218750 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364950844.739583      4348.333496 tel/cavs/interrupt.c:98  	irq_lvl2_handler() error: IRQ storm at level 16 status 00010000
    0      1          IRQ         364955192.656250      4347.916504 tel/cavs/interrupt.c:98  	
```
[aplay02_aplay00_dmesg.log](https://github.com/thesofproject/linux/files/3603886/aplay02_aplay00_dmesg.log)
[aplay02_aplay00_error.log](https://github.com/thesofproject/linux/files/3603887/aplay02_aplay00_error.log)
[aplay02_aplay00.log](https://github.com/thesofproject/linux/files/3603888/aplay02_aplay00.log)

YvonneYang2, 2019-09-19:
Issue can still reproduce with test recipe as below.

Test recipe:
Firmware: 460c41aa https://github.com/thesofproject/sof/commits/master
Kernel: 4ac5440 https://github.com/thesofproject/linux/tree/integration/soundwire-latest
tplg:  sof-icl-rt711-rt1308-rt715-hdmi.tplg rename sof-icl-rt711-rt1308-rt715.tplg  https://github.com/bardliao/sof/blob/fix/soundwire-tplg/tools/topology/
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

[error_logger.log](https://github.com/thesofproject/sof/files/3630076/error_logger.log)
[logger.log](https://github.com/thesofproject/sof/files/3630077/logger.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3630078/dmesg.log)

ClarexZhou, 2019-09-23:
on CML rtc700 SDW RVP, same ipc timed out when aplay + DMIC32 arecord.
**Environment**
Firmware: 0b47129  Master
Kernel: e24f321b on integration/soundwire-latest
Topology: file: same with firmware branch, sof-cml-rt700.m4
platform: CML RVP with onboard ALC700 in SDW mode

YvonneYang2, 2019-09-23:
on ICL rtc700 SDW RVP, same ipc timed out when aplay device 0,0 + DMIC32 arecord.

bardliao, 2019-10-09:
@YvonneYang2 Could you test it with latest FW? It seems that I can't reproduce the issue with latest FW.

YvonneYang2, 2019-10-10:
Check it on icl 3-in-1 with test recipe as below , issue cannot reproduce.

Test recipe:
Firmware: fe68e9b https://github.com/thesofproject/sof/commits/master 
Kernel: 0df05b7 https://github.com/thesofproject/linux/tree/integration/soundwire-latest
tplg:  sof-icl-rt711-rt1308-rt715-hdmi.tplg rename sof-icl-rt711-rt1308-rt715.tplg  path same as FW branch
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

closed 2019-10-10 by YvonneYang2