thesofproject/sof #1843: [BUG]Glitches in beginning of playback via SRC pipe

singalsu, 2019-09-19:
**Describe the bug**
Especially with 48 kHz content playback to Media Playback device the first fraction of a second has discontinuities in output waveform. The effect can be heard when listening carefully.

Witht other sample rates like 32 kHz there's typically one glitch. 48 kHz has several (5).

**To Reproduce**
Playback of 48 kHz content to hw:0,4 device in UP^2 with default topology sof-apl-pcm512x.tplg.

**Reproduction Rate**
Always.

**Expected behavior**
Clean playback start.

**Impact**
It's an annoyance.

**Environment**
1) Branch name and commit hash of 3 repositories: sof (firmware), linux (kernel driver) and soft (tools & topology).
2) Name of the topology file
3) Name of the platform(s) on which the bug is observed.
4) Reproducibility Rate. If you can only reproduce it randomly, it’s useful to report how many times the bug has been reproduced vs. the number of attempts it’s taken to reproduce the bug.

**Screenshots or console output**
[image]
[image]

lgirdwood, 2019-09-30:
@singalsu I assume SRC is reading ahead ? It could just be we need to zero all SRC source buffer before we start.

tlauda, 2019-10-11:
@singalsu On BYT it looks even worse with 48 kHz:

[image]

tlauda, 2019-10-25:
@singalsu Yes, it's clean also on BYT. Closing this one. Please create new bug.

closed 2019-10-25 by tlauda