thesofproject/sof #1845: [FEATURE] Add Cherrytrail to CI

cujomalainey, 2019-09-19:
ChromeOS is upreving cht kernels, we are migrating the platform to SOF in the process. To help smooth integration it would be best to have CHT devices in CI.

plbossart, 2019-09-19:
There are 3 types of CHT devices i can think of:
- Chromebooks. For CI we'd need to use e.g. a Cyan platform with Ubuntu to be able to test the latest kernel.
- Windows devices. I have an Asus T100HA and a headless device, they could be repurposed for CI. Those devices are picky wrt reboot so would need to be tied to a networked power supply to guarantee stable reboots.
- Up board. @singalsu do we have a topology with the 50kHz SRC? I can't recall if this was fully tested.

juimonen, 2019-09-19:
@plbossart we have sof-apl-src-50khz-pcm512x.m4 which was only initially tested, and well it was working then...

lgirdwood, 2019-09-23:
@wenqingfu fyi

cujomalainey, 2019-10-03:
#1847 and #1898 are both regressions that would have likely been caught by CI. Getting boards added once the platform is stabilized would be really helpful.

fredoh9, 2019-10-14:
I got a CHT UP board from @lgirdwood today. Usually we allocate at least two devices for a platform.
Need to discuss with internal team and update here.

plbossart, 2019-10-14:
> I got a CHT UP board from @lgirdwood today. Usually we allocate at least two devices for a platform.
> Need to discuss with internal team and update here.

I was not able to progress with the Up board since the extra pinctrl patches we used back in the 4.14 time no longer apply, and I have no idea how to set-up the pins needed for I2C/I2S as needed. There doesn't seem to be a path to upstream this work, so this board looks like a dead-end for our audio work.

lgirdwood, 2019-10-28:
@plbossart not too concerned about the pinctrl and codec integration here since we share SSP and DMAC with BYT. This is more about validating the clocking and timing differences, i.e. aplay -d 5 actually plays for 5 seconds and not 6.

plbossart, 2019-10-28:
@lgirdwood ok, then we can add an Up board now, it'll work just fine in nocodec mode. I am just a bit frustrated that we can't hear sound because of stupid GPIO setups.

wenqingfu, 2019-10-28:
we will add bsw chromebook instead, won't add UP board to make better use of CI capacity.

fredoh9, 2019-10-29:
BSW_CYN_MAX98090 ia added and being tested for every PR from yesterday night. BSW has identical audio dsp with CHT.

https://sof-ci.01.org/linuxpr/PR1393/build2260/devicetest/

I will close this issue.

closed 2019-10-29 by fredoh9