thesofproject/sof #1847: [BUG] Cherrytrail DMA fails occasionally

cujomalainey, 2019-09-19:
**Describe the bug**
Cherrytrail playback fails, reboots sometimes gets system to a good state which allows playback but occasionally this will appear in the logs

```
    0      2          DMA          32400464.635417         1.979167   src/drivers/dw/dma.c:442    dw_dma_stop(): dma 1 channel 0 stop
    0      1         WAIT          32401809.166667      1344.531250   src/lib/wait.c:96     ewt
    0      1          DMA          32401815.989583         6.822917   src/drivers/dw/dma.c:460    dw_dma_stop() error: dma 1 channel 0
```
or
```
    0      2          SSP          51489200.677083         2.708333 intel/baytrail/ssp.c:572    ssp_trigger()                                                                                                                                                                              
    0      2          SSP          51489203.385417         2.708333 intel/baytrail/ssp.c:521    ssp_start()                                                                                                                                                                                
    0      1          DMA          51490154.375000       950.989563 src/drivers/dw/dma.c:275    dw_dma_start() error: dma 2 channel 1 not ready ena 0x0 status 0x5                                                                                                                         
    0      1         HOST 1.1      51490166.093750        11.718750     src/audio/host.c:796    host_copy() error: dma_copy() failed, ret = 4294967280                                                                                                                                     
    0      1         PIPE          51490174.687500         8.593750 src/audio/pipeline.c:801    pipeline_copy() error: ret = -16, start->comp.id = 3, dir = 1                                                                                                                              
    0      1         PIPE 1.10     51490184.843750        10.156250 src/audio/pipeline.c:977    pipeline_task(): xrun recover failed! pipeline will be stopped!
```

**To Reproduce**
Load cyan with SOF Tot and attempt to run a speaker test

**Reproduction Rate**
4 out of 5 reboots approximately

**Expected behavior**
system is stable

**Impact**
blocker for kerneluprev for bsw

**Environment**
1) Branch name and commit hash
    * SOF 460c41aa661b918f543f86ed3d48bc5042e41fa2 (master) (with https://github.com/thesofproject/sof/pull/1846 and https://github.com/thesofproject/sof/pull/1842)
    * Kernel 6fc7eb30029afb1fad671a72c7171757f154dfd7 (topic/sof-dev)
2) Name of the topology file
    * sof-cht-max98090.tplg 
3) Name of the platform(s) on which the bug is observed
    * Cherrytrail

**Screenshots or console output**
```
localhost ~ # speaker-test -D hw:1,0 -c 2                                                                                                                                                                                                                                                  

speaker-test 1.1.6

Playback device is hw:1,0
Stream parameters are 48000Hz, S16_LE, 2 channels                                                                                                                                                                                                                                          
Using 16 octaves of pink noise                                                                                                                                                                                                                                                             
Rate set to 48000Hz (requested 48000Hz)                                                                                                                                                                                                                                                    
Buffer size range from 96 to 16368                                                                                                                                                                                                                                                         
Period size range from 48 to 4080                                                                                                                                                                                                                                                          
Using max buffer size 16368                                                                                                                                                                                                                                                                
Periods = 4                                                                                                                                                                                                                                                                                
was set period_size = 3312
was set buffer_size = 16368                                                                                                                                                                                                                                                                
 0 - Front Left                                                                                                                                                                                                                                                                            
Write error: -5,Input/output error                                                                                                                                                                                                                                                         
xrun_recovery failed: -5,Input/output error
Transfer failed: Input/output error
```

Relevant dmesg logs
```
[   51.864104] sof-audio-acpi 808622A8:00: pcm: open stream 0 dir 0
[   51.864110] sof-audio-acpi 808622A8:00: period min 192 max 16384 bytes
[   51.864113] sof-audio-acpi 808622A8:00: period count 2 max 16
[   51.864116] sof-audio-acpi 808622A8:00: buffer max 65536 bytes
[   51.864511] sof-audio-acpi 808622A8:00: rate_min: 48000 rate_max: 48000
[   51.864515] sof-audio-acpi 808622A8:00: channels_min: 2 channels_max: 2
[   51.866616] sof-audio-acpi 808622A8:00: rate_min: 48000 rate_max: 48000
[   51.866634] sof-audio-acpi 808622A8:00: channels_min: 2 channels_max: 2
[   51.866965] sof-audio-acpi 808622A8:00: rate_min: 48000 rate_max: 48000
[   51.866969] sof-audio-acpi 808622A8:00: channels_min: 2 channels_max: 2
[   51.866984] sof-audio-acpi 808622A8:00: pcm: hw params stream 0 dir 0
[   51.866992] sof-audio-acpi 808622A8:00: generating page table for 00000000ab5ef320 size 0xffc0 pages 16
[   51.866998] sof-audio-acpi 808622A8:00: stream_tag 0
[   51.867012] sof-audio-acpi 808622A8:00: ipc tx: 0x60010000
[   51.867824] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60010000
[   51.867850] sof-audio-acpi 808622A8:00: pcm: stream dir 0, posn mailbox offset is 1329664
[   51.871161] sof-audio-acpi 808622A8:00: pcm: trigger stream 0 dir 0 cmd 1
[   51.871172] sof-audio-acpi 808622A8:00: ipc tx: 0x60040000
[   51.871357] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60040000
[   52.375426] sof-audio-acpi 808622A8:00: PCM: stream 0 dir 0 DMA position 0 DAI position 0
[   52.879599] sof-audio-acpi 808622A8:00: pcm: trigger stream 0 dir 0 cmd 0
[   52.879625] sof-audio-acpi 808622A8:00: ipc tx: 0x60050000
[   52.880581] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60050000
[   52.880608] sof-audio-acpi 808622A8:00: ipc tx: 0x60030000
[   52.880854] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60030000
[   52.881092] sof-audio-acpi 808622A8:00: pcm: free stream 0 dir 0
[   52.881589] sof-audio-acpi 808622A8:00: pcm: close stream 0 dir 0
```

cujomalainey, 2019-09-20:
Possibly interesting line that might be causing the DMA to crash, does it really have 65k channels?
```
    0      2          DMA         192612458.802083        18.229166 src/drivers/dw/dma.c:187    dw_dma_channel_get(): dma 2 request channel 65535
```

paulstelian97, 2019-09-20:
Looks like some sort of weird issue, like casting -1 to a uint16_t? I noticed the pattern of using negative channel numbers in dma_get() as called from host. Can you show a few lines before that interesting one?

paulstelian97, 2019-09-20:
Found the relevant code for the real issue (line numbers are wrong), the channel stop seems to fail waiting for FIFOs to be empty (function dw_dma_stop, only with CONFIG_DMA_SUSPEND_DRAIN -- which is enabled on CHT and BYT).

Someone who's actually working on that platform?

tlauda, 2019-09-20:
Please try to increase the timeout. It might be that ocassionally FIFO is not emptied on time. Disabling the channel witf FIFO not empty is causing DMA hang and that's why you cannot restart it.

cujomalainey, 2019-09-20:
Doubling the DMA timeout (for baytrail) did not change the behaviour. It seems to consistently fail on master but on 005-cml-drop it flakes and I was able to get audio working a couple times

cujomalainey, 2019-09-20:
@paulstelian97 yes I noticed this morning I forgot to update the ldc between builds which caused incorrect line numbers

cujomalainey, 2019-09-20:
changing all the cherrytrail only dmac configs in `src/platform/baytrail/lib/dma.c` to `#if 0` allows cherrytrail to work. I am going to look closer at the last controllers config

lgirdwood, 2019-09-23:
@cujomalainey that's a good fine and interesting, these values are internal DMAC channel priority configurations. They are intended to allow the DMAC to give bandwidth and FIFO priority to some channels over other ones. We do not use these currently in SOF so it should be safe to comment on CHT platform.

cujomalainey, 2019-10-02:
Continuing DMA failure discussion from #1862

@lgirdwood  I logged the SSP state post start and pre stop

```
    0      1          SSP          21673968.750000  21673968.000000 intel/baytrail/ssp.c:590    SSITR: 0
    0      1          SSP          21673976.770833         8.020833 intel/baytrail/ssp.c:591    SSDR: 0
    0      1          SSP          21673984.791667         8.020833 intel/baytrail/ssp.c:592    SSTO: 0
    0      1          SSP          21673992.812500         8.020833 intel/baytrail/ssp.c:593    SSTSS: 0
    0      1          SSP          21674000.833333         8.020833 intel/baytrail/ssp.c:594    SFIFOL: 30000
    0      1          SSP         321949477.447917 300275488.000000 intel/baytrail/ssp.c:551    ssp_start(): post-write dump
    0      1          SSP         321949484.270833         6.822917 intel/baytrail/ssp.c:552    SSCR0: c009bf
    0      1          SSP         321949492.291667         8.020833 intel/baytrail/ssp.c:553    SSCR1: 1200cc0
    0      1          SSP         321949500.312500         8.020833 intel/baytrail/ssp.c:554    SSCR2: 801
    0      1          SSP         321949508.385417         8.072917 intel/baytrail/ssp.c:555    SSCR3: 30607
    0      1          SSP         321949516.510417         8.125000 intel/baytrail/ssp.c:556    SSCR4: 1400
    0      1          SSP         321949524.531250         8.020833 intel/baytrail/ssp.c:557    SSCR5: 26
    0      1          SSP         321949532.552083         8.020833 intel/baytrail/ssp.c:558    SSPSP: 14008c
    0      1          SSP         321949540.625000         8.072917 intel/baytrail/ssp.c:559    SFIFOTT: 30003
    0      1          SSP         321949548.645833         8.020833 intel/baytrail/ssp.c:560    SSTSA: 3
    0      1          SSP         321949556.770833         8.125000 intel/baytrail/ssp.c:561    SSRSA: 3
    0      1          SSP         321949564.895833         8.125000 intel/baytrail/ssp.c:562    SSSR: f024
    0      1          SSP         321949572.916667         8.020833 intel/baytrail/ssp.c:563    SSITR: 0
    0      1          SSP         321949580.937500         8.020833 intel/baytrail/ssp.c:564    SSDR: 0
    0      1          SSP         321949588.958333         8.020833 intel/baytrail/ssp.c:565    SSTO: 0
    0      1          SSP         321949596.979167         8.020833 intel/baytrail/ssp.c:566    SSTSS: 0
    0      1          SSP         321949605.000000         8.020833 intel/baytrail/ssp.c:567    SFIFOL: 1f0000
    0      1          SSP          21320256.666667              nan intel/baytrail/ssp.c:557    SSCR5: 26
    0      1          SSP          21320264.687500         8.020833 intel/baytrail/ssp.c:558    SSPSP: 14008c
    0      1          SSP          21320272.916667         8.229167 intel/baytrail/ssp.c:559    SFIFOTT: 30003
    0      1          SSP          21320281.197917         8.281250 intel/baytrail/ssp.c:560    SSTSA: 3
    0      1          SSP          21320289.427083         8.229167 intel/baytrail/ssp.c:561    SSRSA: 3
    0      1          SSP          21320297.708333         8.281250 intel/baytrail/ssp.c:562    SSSR: 20f034
    0      1          SSP          21320305.989583         8.281250 intel/baytrail/ssp.c:563    SSITR: 0
    0      1          SSP          21320314.218750         8.229167 intel/baytrail/ssp.c:564    SSDR: 0
    0      1          SSP          21320322.500000         8.281250 intel/baytrail/ssp.c:565    SSTO: 0
    0      1          SSP          21320330.729167         8.229167 intel/baytrail/ssp.c:566    SSTSS: 80000001
    0      1          SSP          21320339.010417         8.281250 intel/baytrail/ssp.c:567    SFIFOL: 0
    0      1          SSP          21673873.697917    353534.687500 intel/baytrail/ssp.c:578    ssp_stop(): pre-write dump
    0      1          SSP          21673880.520833         6.822917 intel/baytrail/ssp.c:579    SSCR0: c009bf
    0      1          SSP          21673888.541667         8.020833 intel/baytrail/ssp.c:580    SSCR1: 1100cc0
    0      1          SSP          21673896.562500         8.020833 intel/baytrail/ssp.c:581    SSCR2: 801
    0      1          SSP          21673904.583333         8.020833 intel/baytrail/ssp.c:582    SSCR3: 30607
    0      1          SSP          21673912.604167         8.020833 intel/baytrail/ssp.c:583    SSCR4: 1400
    0      1          SSP          21673920.625000         8.020833 intel/baytrail/ssp.c:584    SSCR5: 26
    0      1          SSP          21673928.645833         8.020833 intel/baytrail/ssp.c:585    SSPSP: 14008c
    0      1          SSP          21673936.666667         8.020833 intel/baytrail/ssp.c:586    SFIFOTT: 30003
    0      1          SSP          21673944.687500         8.020833 intel/baytrail/ssp.c:587    SSTSA: 3
    0      1          SSP          21673952.708333         8.020833 intel/baytrail/ssp.c:588    SSRSA: 3
```
See attached for full dump of initial states from set config
[bad_after_set_config.txt](https://github.com/thesofproject/sof/files/3683770/bad_after_set_config.txt)
[good_after_set_config.txt](https://github.com/thesofproject/sof/files/3683771/good_after_set_config.txt)
[good_before_set_config.txt](https://github.com/thesofproject/sof/files/3683772/good_before_set_config.txt)
[bad_before_set_config.txt](https://github.com/thesofproject/sof/files/3683773/bad_before_set_config.txt)

I tried writing the default back to SSSR before set_config with no change in behaviour. I am trying to understand how to determine the DMA alignment from the doc to see if that is the issue. The document is a bit daunting so if can give me some pointers that would be great.

cujomalainey, 2019-10-04:
fun little tidbit, a suspend/resume cycle (with suspend enabled on byt/cht) appears to fix this bug

lgirdwood, 2019-10-07:
@cujomalainey one other thing to check is that DMA HW LLI mode is not used on CHT, iirc only SW LLI mode should be used on BYT, CHT and BSW (worth checking sw-dma.c config here). This can cause corrupted audio.
@mengdonglin can someone check all the SSP bits as I'm travelling for next two weeks.

lgirdwood, 2019-11-12:
@cujomalainey I'm guessing thi sis resolved now ?

cujomalainey, 2019-11-12:
I got sidetracked by some other issues, cht is back on my plate this week, will look into it.

lgirdwood, 2019-11-12:
@cujomalainey ok, btw @keyonjie has fixed some issues wrt suspend on CHT so these should be merging soon. I am suspcicious that suspend is fixing it, it does sound like HW gettng stuck and needing the old "switch it OFF and ON again" fix.

cujomalainey, 2019-11-22:
just tried ToT and still having issues, but I am getting a full firmware oops now which should help me debug tomorrow

cujomalainey, 2019-11-22:
Looks like the oops was triggered by the new agent that tracks clock drift.

```
    0      1           SA         262579275.156250     56554.167969      src/lib/agent.c:55     validate(), ll drift detected, delta = 20812
    0      1           SA         262758275.468750    179000.312500      src/lib/agent.c:55     validate(), ll drift detected, delta = 20818
    0      1           SA         262994275.833333    236000.359375      src/lib/agent.c:55     validate(), ll drift detected, delta = 20824
    0      1           SA         263051248.697917     56972.863281      src/lib/agent.c:55     validate(), ll drift detected, delta = 20303
    0      1           SA         263287249.218750    236000.515625      src/lib/agent.c:55     validate(), ll drift detected, delta = 20313
    0      1           SA         263523253.854167    236004.640625      src/lib/agent.c:55     validate(), ll drift detected, delta = 20403
    0      1           SA         263759249.531250    235995.671875      src/lib/agent.c:55     validate(), ll drift detected, delta = 20320
    0      1           SA         263995259.062500    236009.531250      src/lib/agent.c:55     validate(), ll drift detected, delta = 20502
    0      1           SA         264231255.104167    235996.046875      src/lib/agent.c:55     validate(), ll drift detected, delta = 20426
    0      1           SA         264467264.270833    236009.171875      src/lib/agent.c:55     validate(), ll drift detected, delta = 20603
    0      1           SA         264703259.895833    235995.625000      src/lib/agent.c:55     validate(), ll drift detected, delta = 20519
    0      1           SA         264939269.479167    236009.578125      src/lib/agent.c:55     validate(), ll drift detected, delta = 20703
    0      1           SA         265175265.520833    235996.046875      src/lib/agent.c:55     validate(), ll drift detected, delta = 20627
    0      1           SA         265411274.687500    236009.171875      src/lib/agent.c:55     validate(), ll drift detected, delta = 20803
    0      1           SA         265647270.364583    235995.671875      src/lib/agent.c:55     validate(), ll drift detected, delta = 20719
```

which happens on the first playback but after that, all playback/records fail on the trigger IPC call when they try to start

lgirdwood, 2019-11-22:
@cujomalainey default SSP master clock is different on CHT/BSW comapred to BYT.  25MHz vs 19.2MHz IIRC, it seems that we have an incorrect clock setting in CHT which is failing on scheduling/agent calculations etc @ranj063 @keyonjie do you guys have this HW to check all the clock macros.

plbossart, 2019-11-22:
> @cujomalainey default SSP master clock is different on CHT/BSW comapred to BYT. 25MHz vs 19.2MHz IIRC, it seems that we have an incorrect clock setting in CHT which is failing on scheduling/agent calculations etc @ranj063 @keyonjie do you guys have this HW to check all the clock macros.

we only use 19.2 in all cases. 25 MHz is not used for SSP, ever.

ranj063, 2019-11-22:
@lgirdwood yes we've got the BSW cyan chromebook. It is already part of our CI now

cujomalainey, 2019-11-22:
Do you have a snipit I can apply to correct this shift in the mean time to adjust the clocking so I am unblocked on my debugging?

lgirdwood, 2019-11-22:
@plbossart my understanding was that the ext timer either used 25 or 19.2, but could it be incorrectly set on CHT/BSW ? The only other difference I can see is
```c
#if CONFIG_BAYTRAIL
static struct freq_table platform_cpu_freq[] = {
	{ 25000000, 25000, 0x0 },
	{ 25000000, 25000, 0x1 },
	{ 50000000, 50000, 0x2 },
	{ 50000000, 50000, 0x3 }, /* default */
	{ 100000000, 100000, 0x4 },
	{ 200000000, 200000, 0x5 },
	{ 267000000, 267000, 0x6 },
	{ 343000000, 343000, 0x7 },
};
#elif CONFIG_CHERRYTRAIL
static struct freq_table platform_cpu_freq[] = {
	{ 19200000, 19200, 0x0 },
	{ 19200000, 19200, 0x1 },
	{ 38400000, 38400, 0x2 },
	{ 50000000, 50000, 0x3 }, /* default */
	{ 100000000, 100000, 0x4 },
	{ 200000000, 200000, 0x5 },
	{ 267000000, 267000, 0x6 },
	{ 343000000, 343000, 0x7 },
};
#endif
```
@cujomalainey what happens if you swap the array elems for SSP clock in clk.c (I'm wondering if the clocks are inverted). 
```c
static struct freq_table platform_ssp_freq[] = {
	{ 19200000, 19200, PMC_SET_SSP_19M2 }, /* default */
	{ 25000000, 25000, PMC_SET_SSP_25M },
};
```

cujomalainey, 2019-11-22:
so get this, that message was only appearing when boot beep was allowed to run. When i flip the `platform_ssp_freq` table then the message appears when boot beep is **NOT** allowed to play.

keyonjie, 2019-11-22:
@lgirdwood @cujomalainey it looks to me that here the error happens on dw dma part, it is somewhat similar to https://github.com/thesofproject/sof/issues/2107

cujomalainey, 2019-11-22:
@keyonjie yes the initial part of this bug does look similar, but we deemed it a hardware issue and bypassed it in #1861. I don't think the issue we are facing now is related.

cujomalainey, 2019-11-22:
@lgirdwood I am wondering if maybe the SSP clock isn't being updated

Looking at the set_freq code between the SSP and CPU, the CPU sets some register bits to the desired clk freq then sends an IPC to trigger the update on the PMC (not sure what the PMC is.) Where as the SSP appears to blindly pass in the SSP frequency index as the IPC parameter. Is this correct?

https://github.com/thesofproject/sof/blob/master/src/platform/baytrail/lib/clk.c#L85

Unfortunately no other platforms have a set_freq for the SSP to go off as a reference. I tried add a call to set_freq as part of the clk init but it did not affect anything which made me suspect this.

cujomalainey, 2019-11-22:
Apologies that was a red herring, I thought it was sending an enum. I incorrectly traced it, I see the commands now in the header.

lgirdwood, 2019-11-23:
@cujomalainey PMC is a companion micro controller than manages some PM domains and clocks. I suspect boot beep is setting the clocking and we are changing it later. I'm pretty sure that PC beep and core boot have a PMC driver so they could be setting clocks via SHIM or PMC. 

Can you dump the SHIM contents with and without PC beep enabled ? There is a lot of clock config in there and it's possible we have some inconsistency.

cujomalainey, 2019-11-25:
Diff of the files below.

[good_shim.txt](https://github.com/thesofproject/sof/files/3889265/good_shim.txt)
[bad_shim.txt](https://github.com/thesofproject/sof/files/3889266/bad_shim.txt)

```
diff --git a/bad/shim b/good/shim
index 5487a2b..86ae5f0 100644
--- a/bad/shim
+++ b/good/shim
@@ -9,8 +9,8 @@
 0000080    0007    0000    0000    0000    0e0e    0000    0000    0000
 0000090    0000    0000    0000    0000    0000    0000    0000    0000
 *
-00000c0    e3e5    1ceb    0000    8000    c44e    1ceb    0000    0000
+00000c0    3f0c    6dc8    0000    8000    275e    6dc8    0000    0000
 00000d0    0000    0000    0000    0000    0000    0000    0000    0000
 00000e0    0000    0000    0000    0000    0001    0000    0001    8000
-00000f0    0001    0000    0001    8000    0c35    0000    0180    6000
+00000f0    0001    0000    0001    8000    0001    0000    0001    8000
 0000100
```

cujomalainey, 2019-11-26:
So I forced the SHIM values back to the good for SHIM_SSP2_DIVL and SHIM_SSP2_DIVH in the SSP probe but I read back `4b401` rather what I wrote which was `80000001` in `SHIM_SSP2_DIVH` so not sure what the issue is there.

lgirdwood, 2019-11-26:
@cujomalainey I think the SSP M/N dividers need an atomic 64bit update OR may need to be disabled prior to updates OR may need a write sequence. I do see an update bit (b61) which would also imply that 32bit writes are supported. i.e write bits 0 - 31, then 32 - 63 and set updte bit.

cujomalainey, 2019-11-26:
I'll try the ordering. Otherwise there is a bug in `shim_read/write` as that is what I was using to set and check.

lgirdwood, 2019-11-26:
@cujomalainey thanks, I'm not 100% sure as we never used these bits on BYT, they are on the other platforms but IO is 32bit instead of 64bit.

cujomalainey, 2019-11-26:
Ah yes, I definitely made that mistake, also I'm remembering there is a 90% chance I forgot to put the 0x to make it hex

cujomalainey, 2019-11-27:
**SUCCESS!!!**
Adding the following writes fixes the bug. Audio playback works and the agent is happy.
```
shim_write(SHIM_SSP2_DIVL, 1);
shim_write(SHIM_SSP2_DIVL + 2, 0x80010000);
shim_write(SHIM_SSP2_DIVH, 0x80000001);
shim_write(SHIM_SSP2_DIVH + 2, 0x18000);
```
Now the question of where to integrate it in SOF? Also I traced the respective writes in depthcharge which are causing the issue. 

https://chromium.googlesource.com/chromiumos/platform/depthcharge/+/refs/heads/master/src/drivers/bus/i2s/braswell/braswell.c#607

lgirdwood, 2019-11-27:
@cujomalainey great :) There is no harm in adding these writes to platform.c init() in order to reset the M/N dividers to a sane state expected by FW. 
@plbossart I guess this same M/N disable logic would be needed for BDW too ?

cujomalainey, 2019-11-27:
If needed I can run a test on my Samus for BDW

cujomalainey, 2019-11-28:
Scratch that, there appears to be dai_link issues that are confusing me greatly on Samus with SOF

```
[   31.854128] bdw-rt5677 bdw-rt5677: info: override BE DAI link Codec DSP                                                          
[   31.854133] bdw-rt5677 bdw-rt5677: info: override BE DAI link Codec                                                                                                                                                                                                                      
[   31.854158] bdw-rt5677 bdw-rt5677: ASoC: CPU DAI snd-soc-dummy-dai not registered 
```
So I can't test until I wrap my head around why that is happening. I can file another bug if we want to bother fixing samus. I think it would be nice to officially support it for testing purposes.

lgirdwood, 2019-11-28:
@cujomalainey yeah, can you file bug on that. I've just had a look at bdw-rt5677.c and can see it's mixing DPCM and non DPCM DAI links which is probably rarely tested as all the others are DPCM.

plbossart, 2019-12-02:
> @cujomalainey great :) There is no harm in adding these writes to platform.c init() in order to reset the M/N dividers for all SSP ports to a sane state expected by FW.
> @plbossart I guess this same M/N disable logic would be needed for BDW too ?

I don't think those are M/N registers, we don't use them on Baytrail. they look like standard dividers to me?

closed 2019-12-16 by jajanusz

youling257, 2019-12-16:
with "platform: cht: reset ssp clk M/N dividers on boot" patch, i see these warning.
[   10.293720] sof-audio-acpi 808622A8:00: error: sink SSP0.IN not found
[   10.294200] sof-audio-acpi 808622A8:00: ASoC: no sink widget found for SSP0.IN
[   10.294594] sof-audio-acpi 808622A8:00: ASoC: Failed to add route modem_in -> direct -> SSP0.IN
[   10.295097] sof-audio-acpi 808622A8:00: error: source SSP0.OUT not found
[   10.308311] sof-audio-acpi 808622A8:00: ASoC: no source widget found for SSP0.OUT
[   10.313123] sof-audio-acpi 808622A8:00: ASoC: Failed to add route SSP0.OUT -> direct -> modem_out
I don't want to see.

ranj063, 2019-12-16:
@youling257 you'll need  https://github.com/thesofproject/sof/pull/2171/commits/edd8d70c58980c45d3d0b1e856d18f08f65605cc to fix those errors

PR #2148 (2019-12-16): platform: cht: reset ssp clk M/N dividers on boot
The firmware may have modified these values to execute features such as
a boot beep. Therefore we cannot assume they are set to their defaults.
Lets write the defaults just to be safe.

Fixes #1847
commit 9f3714f6 (2019-12-16): platform: cht: reset ssp clk M/N dividers on boot

The firmware may have modified these values to execute features such as
a boot beep. Therefore we cannot assume they are set to their defaults.
Lets write the defaults just to be safe.

Fixes #1847