thesofproject/sof #1853: [BUG] 50kHz SRC hw_params fail on APL/CHT

plbossart, 2019-09-20:
**Describe the bug**

hw params fail when using the 50kHz SRC on APL with the sof-apl-src-50khz-pcm512x.tplg
The same error happens when using this topology on CHT.

**To Reproduce**
Use Up2 board, connect HiFiberry board, add PCM512x support with ACPI scripts, copy sof-apl-src-50khz-pcm512x.tplg to target and symlink it as sof-apl-pcm512x.tplg

**Reproduction Rate**
100%

**Expected behavior**
clean sound coming out...

**Impact**
showstopper to add CHT to CI farm

**Environment**

firmware b0bb5a33c8ae1f8c31a267b265975828bc05d638
kernel 

1) Branch name and commit hash of 3 repositories: sof (firmware), linux (kernel driver) and soft (tools & topology).
2) Name of the topology file
3) Name of the platform(s) on which the bug is observed.
4) Reproducibility Rate. If you can only reproduce it randomly, it’s useful to report how many times the bug has been reproduced vs. the number of attempts it’s taken to reproduce the bug.

**Screenshots or console output**

````
   0      2          DMA         110014462.031250        10.520833 intel/cavs/hda-dma.c:429 	hda-dmac: 5 channel 0 -> get
    0      2          DMA         110014467.604167         5.572917 intel/cavs/hda-dma.c:609 	hda-dmac: 5 channel 0 -> config
    0      2          SRC         110014475.572917         7.968750  src/audio/src/src.c:570 	src_params()
    0      2          SRC         110014480.312500         4.739583  src/audio/src/src.c:597 	src_params(), source_rate = 48000, sink_rate = 50000
    0      2          SRC         110014484.479167         4.166667  src/audio/src/src.c:599 	src_params(), params->channels = 2, dev->frames = 50
    0      2          SRC         110014489.791667         5.312500  src/audio/src/src.c:609 	src_params(), blk_in = 0, blk_out = 0
    0      1          MEM         110014498.385417         8.593750      src/lib/alloc.c:273 	error: 8 blocks needed for allocation but only 3 blocks are remaining
    0      2          DAI 1.5     110014519.114583        20.729166      src/audio/dai.c:385 	dai_params()
    0      2          DAI 1.5     110014524.739583         5.625000      src/audio/dai.c:254 	dai_playback_params() dest_dev = 12 stream_id = 0 src_width = 4 dest_width = 4
    0      2          DAI 1.5     110014529.375000         4.635417      src/audio/dai.c:280 	dai_playback_params() fifo 8A10
    0      2         PIPE 1.6     110014537.864583         8.489583 src/audio/pipeline.c:399 	pipeline_prepare()
    0      2         HOST 1.0     110014542.135417         4.270833     src/audio/host.c:702 	host_prepare()
    0      2          SRC         110014547.968750         5.833333  src/audio/src/src.c:799 	src_prepare()
    0      1         COMP         110014553.229167         5.260417 rc/audio/component.c:120 	comp_set_sink_buffer() error: sink buffer size is not sufficient
    0      1          SRC         110014558.906250         5.677083  src/audio/src/src.c:858 	src_prepare() error: comp_set_sink_buffer() failed
    0      1         PIPE         110014564.791667         5.885417 src/audio/pipeline.c:406 	pipeline_prepare() error: ret = -22,dev->comp.id = 0
    0      1          IPC         110014569.739583         4.947917    src/ipc/handler.c:305 	ipc: pipe 1 comp 0 prepare failed -22
````

````
[   18.955827] sof-audio-pci 0000:00:0e.0: pcm: open stream 0 dir 0
[   18.955829] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
[   18.955830] sof-audio-pci 0000:00:0e.0: period count 2 max 16
[   18.955832] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
[   18.956516] sof-audio-pci 0000:00:0e.0: rate_min: 50000 rate_max: 50000
[   18.956519] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[   18.956528] sof-audio-pci 0000:00:0e.0: rate_min: 50000 rate_max: 50000
[   18.956529] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[   18.956535] sof-audio-pci 0000:00:0e.0: rate_min: 50000 rate_max: 50000
[   18.956537] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[   18.956542] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 0 dir 0
[   18.956547] sof-audio-pci 0000:00:0e.0: generating page table for 00000000e0e1b5bb size 0x4b00 pages 5
[   18.956555] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x140000 successful
[   18.956572] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   18.956575] sof-audio-pci 0000:00:0e.0: period_bytes:0x12c0
[   18.956576] sof-audio-pci 0000:00:0e.0: periods:4
[   18.956585] sof-audio-pci 0000:00:0e.0: stream_tag 1
[   18.956592] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   18.956825] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   18.956831] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
[   18.956834] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -22
[   18.956836]  Port5: ASoC: hw_params FE failed -22
````

plbossart, 2019-09-20:
@singalsu @juimonen can you guys take a look please?

tlauda, 2019-09-20:
@plbossart @singalsu @juimonen Looks like there is not enough heap space for the buffer allocation. Probably SRC requires larger buffer and hence the problem.

juimonen, 2019-09-20:
@tlauda this topology is quite small and samplerate is only 50khz... I wonder how we can run out of memory? @plbossart do you see the created buffer sizes in fw logs? Can you attach the full logs?

juimonen, 2019-09-20:
so in the topology I see only 3 buffers sized 1200, 1200 and 800 bytes, we should have enough mem?

plbossart, 2019-09-20:
@juimonen it's probably easier to get answers to your questions if you run the tests yourself... I have no added value here, I just reported that the functionality is broken.

singalsu, 2019-09-20:
@plbossart I was busy with other things but fortunately next week looks much better now so I will be able to work with this. Thanks for testing and reporting the bug!

It could be just that the buffers resize code fails now. With previous 50 kHz run we used hacked topology with buffer sizes tweaks that might have avoided to trigger the issue.

plbossart, 2019-09-20:
No worries @singalsu. I added the topology if you want to try on the Up board. I assumed the issues where the same but the SRC and memory sizes are different so there might be a different root cause for the hw_params fail.

lgirdwood, 2019-09-30:
@singalsu CHT has less free memory than BYT (due to extra SSP and DMACs being allocated). See @cujomalainey PRs for disabling DMAC2.

singalsu, 2019-10-01:
@lgirdwood Thanks, good point, I didn't know. Also I think there's a bit excess in topologies with SRC for buffers that could be reduced.

lgirdwood, 2019-10-01:
@singalsu good find, it's likely CHT topology has drifted from BYT topology. They should be aligned.

plbossart, 2019-10-01:
We don't have a topology with 50kHz SRC for BYT, do we? I adapted the one from APL for these tests. And even on APL it still doesn't work despite the much larger internal memory.

lgirdwood, 2019-10-01:
@plbossart we probably dont, @singalsu does XCC make any footrpint difference for SRC i.e. will it squeeze into CHT. Have you also edited kconfig to remove any unsued components ?

juimonen, 2019-10-01:
So the memory for this topology is minimal. probably one of the smallest footprints we have... let me check this tomorrow, there's something else going wrong.

cujomalainey, 2019-10-01:
@lgirdwood, the other cht topologies likely need the additional fixes I have been applying to the maxim one as well

juimonen, 2019-10-02:
The problem is that 50khz topology is using pipe-src-playback pipeline (not pipe-src-volume-playback). So in 50kHz topology SRC is connected directly to DAI and SRC is using 3 periods and DAI 2 periods. Because the last buffer is "dma connected" it is not allowed to do resize (this is the current code in dai) -> thus fail. If you have this kind of connection in the middle of the pipeline (like in pipe-src-volume) the resize handles the difference in period count, it just sees the total buffer size and that it is not "dma connected".

So 2 possible fixes: 
1) change SRC to use 2 periods to comply with the DAI
2) change 50 khz topology to use src-volume-pipe

Also we should think should the last component before DAI be able to change the size of "dma_connected" buffers?

tlauda, 2019-10-02:
@juimonen SRC is not allowed to perform the resize of such buffer, but DAI should be able to do it. I think the problem is that DAI in topology is set to just 2 periods, thus it calculates wrong buffer size and make wrong resize.

juimonen, 2019-10-02:
@tlauda ok. I'm not sure which one is better..?. src to 2 or dai to 3 periods... anyway good that the dai patch is there. I think the src 3 periods is some old leftover so it should work with 2 I guess.

tlauda, 2019-10-02:
@juimonen For APL my patch is setting every DAI to 3, but maybe for CHT 2 would be better.

plbossart, 2019-10-02:
I can understand the DAI using 3 periods, since it needs to deal with the preloading stuff.

I don't understand why we would have different DAI settings on different platforms?
I just don't see why a processing element would require 3 periods?

Too many periods, too little time...

tlauda, 2019-10-03:
@plbossart It's because we're using the last buffer in pipeline directly as the DMA buffer (which is a mistake in my opinion). We've changed scheduling for cAVS platforms to schedule all pipelines on one DMA irq, so due to the specifics of DMA IP we need 3 periods for it to work correctly (timer scheduling will also require such change). I really think we should separate DMA buffers from pipeline buffers, so the specific HW could fulfill its own working requirements internally. Also it would eliminate the need for preload, because we would just transfer zeroes at the beginning.

juimonen, 2019-10-04:
made this to move us forward: https://github.com/thesofproject/sof/pull/1901

@tlauda I tested your commit and changed the dai periods in sof-apl-src-50khz pipe: I could only get noise out... So should I perhaps somehow change the SSP params also?

Anyway we can graduate to that, when I/we discover what is the issue.

tlauda, 2019-10-04:
@juimonen It looks like you played different bit depth and had no component to perform pcm conversion. With pipelines without volume we should probably somehow restrict pcm formats available on host. Is there a way to do that from m4 level?

juimonen, 2019-10-04:
@tlauda hmm ok, I guess we don't have conversions in any other component than volume. Should we add this kind of feature into host etc?

We have the PCM_CAPABILITIES macro used for example like following: PCM_CAPABILITIES(Passthrough Playback PCM_ID, `S32_LE,S24_LE,S16_LE', PCM_MIN_RATE, PCM_MAX_RATE, 2, PIPELINE_CHANNELS, 2, 16, 192, 16384, 65536, 65536)

So I think you should be able to restrict for example to S32_LE.

tlauda, 2019-10-04:
@juimonen There's already task for extracting it from volume #1610.

closed 2019-10-17 by tlauda