thesofproject/sof #1860: [FEATURE] Add DMA trace run-time enable/disable feature

keyonjie, 2019-09-23:
**Is your feature request related to a problem? Please describe.**
At the moment, DMA trace will be enabled and started by default, we can't pause or disable it after it is started. This will lead to issue that it is impossible to fulfil the feature when the system is entering some special state (e.g. low power d0i3 state where DMA trace is required to be stopped).

**Describe the solution you'd like**
implement DMA trace disable/free interface, introduce new IPC messages to respond these requirement from host.

mrajwa, 2019-09-25:
> This will lead to issue that it is impossible to fulfil the feature when the system
I don't understand what is the issue here, are you trying to say logs during d0i3 should not be send or what?

lgirdwood, 2019-09-30:
@mrajwa @keyonjie I'm assuming this is a request for an new IPC command to disable the DMA trace ?

akloniex, 2019-10-24:
dma-trace API extended in PR https://github.com/thesofproject/sof/pull/1995

tlauda, 2019-10-30:
@keyonjie Does the current API extension is enough for you, so this issue can be closed?

lgirdwood, 2019-11-05:
@keyonjie ping

paulstelian97, 2019-12-20:
@keyonjie Does this still apply?

closed 2019-12-23 by lgirdwood

keyonjie, 2019-12-24:
@paulstelian97 @lgirdwood yes, this is closed via PR#1995