thesofproject/sof #1869: [BUG] DSP panic and ipc timeout in wov test on GLK Chrome

aiChaoSONG, 2019-09-25:
**Describe the bug**
once run arecord on wov pipeline, there will be an "Unable to install hw params" message in the terminal, check kernel and sof logs, you'll see the issue.

**To Reproduce**
run `arecord -Dhw:0,8 -M -N -r 16000 -c2 -f s16_le -vvv -d 5  wov.wav`

**Reproduction Rate**
100%

**Environment**
1) Branch
    * Kernel:  topic/sof-dev {e03f8829}
    * SOF: master {d42c0195}
2) Name of the topology file
    * Topology: {sof-glk-da7219-kwd.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {GLK Chrome}
4) OS: Ubuntu1804

```
[  155.898924] sof-audio-pci 0000:00:0e.0: error : DSP panic!
[  155.898928] sof-audio-pci 0000:00:0e.0: panic: dsp_oops_offset 788480 offset 788480
[  155.898932] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  155.898978] sof-audio-pci 0000:00:0e.0: error: runtime exception
[  155.898980] sof-audio-pci 0000:00:0e.0: error: trace point 00004000
[  155.898982] sof-audio-pci 0000:00:0e.0: error: panic at :0
[  155.898984] sof-audio-pci 0000:00:0e.0: error: DSP Firmware Oops
[  155.898986] sof-audio-pci 0000:00:0e.0: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  155.898988] sof-audio-pci 0000:00:0e.0: EXCCAUSE 0x0000000d EXCVADDR 0x00000004 PS       0x00060b25 SAR     0x00000000
[  155.898990] sof-audio-pci 0000:00:0e.0: EPC1     0xbe01ec76 EPC2     0xbe02b9ce EPC3     0x00000000 EPC4    0x00000000
[  155.898992] sof-audio-pci 0000:00:0e.0: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  155.899001] sof-audio-pci 0000:00:0e.0: EPS2     0x00060f20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  155.899003] sof-audio-pci 0000:00:0e.0: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  155.899005] sof-audio-pci 0000:00:0e.0: stack dump from 0xbe061360
[  155.899014] sof-audio-pci 0000:00:0e.0: 0xbe061360: be01ec76 00000000 00000000 be0306a9
[  155.899020] sof-audio-pci 0000:00:0e.0: 0xbe061364: 00000000 00000000 be053680 00000000
[  155.899025] sof-audio-pci 0000:00:0e.0: 0xbe061368: 182a7b00 4e808d0d b68a7028 ffff9a0b
[  155.899028] sof-audio-pci 0000:00:0e.0: 0xbe06136c: ba21f600 ffff9a0b b575d800 ffff9a0b
[  155.899030] sof-audio-pci 0000:00:0e.0: 0xbe061370: b575d800 ffff9a0b 91cbb9d2 ffffffff
[  155.899032] sof-audio-pci 0000:00:0e.0: 0xbe061374: c037fbe0 ffffae08 b601b400 ffff9a0b
[  155.899034] sof-audio-pci 0000:00:0e.0: 0xbe061378: b6371340 ffff9a0b 00000000 00000000
[  155.899036] sof-audio-pci 0000:00:0e.0: 0xbe06137c: 00000000 00000000 00000000 00000000
[  155.899038] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  156.401009] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60010000 size 108
[  156.401018] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  156.401063] sof-audio-pci 0000:00:0e.0: error: runtime exception
[  156.401065] sof-audio-pci 0000:00:0e.0: error: trace point 00004000
[  156.401066] sof-audio-pci 0000:00:0e.0: error: panic at :0
[  156.401069] sof-audio-pci 0000:00:0e.0: error: DSP Firmware Oops
[  156.401070] sof-audio-pci 0000:00:0e.0: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  156.401073] sof-audio-pci 0000:00:0e.0: EXCCAUSE 0x0000000d EXCVADDR 0x00000004 PS       0x00060b25 SAR     0x00000000
[  156.401075] sof-audio-pci 0000:00:0e.0: EPC1     0xbe01ec76 EPC2     0xbe02b9ce EPC3     0x00000000 EPC4    0x00000000
[  156.401076] sof-audio-pci 0000:00:0e.0: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  156.401078] sof-audio-pci 0000:00:0e.0: EPS2     0x00060f20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  156.401080] sof-audio-pci 0000:00:0e.0: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  156.401082] sof-audio-pci 0000:00:0e.0: stack dump from 0xbe061360
[  156.401086] sof-audio-pci 0000:00:0e.0: 0xbe061360: be01ec76 00000000 00000000 be0306a9
[  156.401088] sof-audio-pci 0000:00:0e.0: 0xbe061364: 00000000 00000000 be053680 00000000
[  156.401090] sof-audio-pci 0000:00:0e.0: 0xbe061368: fc42bc00 a7066b64 b68a7028 ffff9a0b
[  156.401092] sof-audio-pci 0000:00:0e.0: 0xbe06136c: c1def9e8 ffffae08 00000000 00000000
[  156.401094] sof-audio-pci 0000:00:0e.0: 0xbe061370: b700a0e8 ffff9a0b b700a0f0 ffff9a0b
[  156.401097] sof-audio-pci 0000:00:0e.0: 0xbe061374: 93325800 ffff9a0b ba0b9800 ffff9a0b
[  156.401099] sof-audio-pci 0000:00:0e.0: 0xbe061378: b620c800 ffff9a0b ba0b9a00 ffff9a0b
[  156.401101] sof-audio-pci 0000:00:0e.0: 0xbe06137c: 00000000 00000000 00000000 00000001
[  156.401108] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[  156.401109] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  156.401114] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  156.401116] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[  156.401118] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 2
[  156.401121] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -110
[  156.401127]  DMIC16k: ASoC: hw_params FE failed -110
[  156.401137] sof-audio-pci 0000:00:0e.0: pcm: free stream 8 dir 1
[  156.413332] sof-audio-pci 0000:00:0e.0: pcm: close stream 8 dir 1
[  158.440527] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
[  158.445265] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  158.953019] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x40010000 size 76
[  158.953027] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[  158.953071] sof-audio-pci 0000:00:0e.0: error: runtime exception
[  158.953073] sof-audio-pci 0000:00:0e.0: error: trace point 00004000
[  158.953075] sof-audio-pci 0000:00:0e.0: error: panic at :0
[  158.953077] sof-audio-pci 0000:00:0e.0: error: DSP Firmware Oops
[  158.953079] sof-audio-pci 0000:00:0e.0: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  158.953081] sof-audio-pci 0000:00:0e.0: EXCCAUSE 0x0000000d EXCVADDR 0x00000004 PS       0x00060b25 SAR     0x00000000
[  158.953084] sof-audio-pci 0000:00:0e.0: EPC1     0xbe01ec76 EPC2     0xbe02b9ce EPC3     0x00000000 EPC4    0x00000000
[  158.953086] sof-audio-pci 0000:00:0e.0: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  158.953087] sof-audio-pci 0000:00:0e.0: EPS2     0x00060f20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  158.953089] sof-audio-pci 0000:00:0e.0: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  158.953091] sof-audio-pci 0000:00:0e.0: stack dump from 0xbe061360
[  158.953095] sof-audio-pci 0000:00:0e.0: 0xbe061360: be01ec76 00000000 00000000 be0306a9
[  158.953097] sof-audio-pci 0000:00:0e.0: 0xbe061364: 00000000 00000000 be053680 00000000
[  158.953099] sof-audio-pci 0000:00:0e.0: 0xbe061368: d289ea00 6bc8c5a2 b68a7028 ffff9a0b
[  158.953101] sof-audio-pci 0000:00:0e.0: 0xbe06136c: c02c3d10 ffffae08 00000000 ffffae08
[  158.953104] sof-audio-pci 0000:00:0e.0: 0xbe061370: b700a0e8 ffff9a0b b700a0f0 ffff9a0b
[  158.953106] sof-audio-pci 0000:00:0e.0: 0xbe061374: b68a7200 ffff9a0b 00000001 00000000
[  158.953108] sof-audio-pci 0000:00:0e.0: 0xbe061378: 40010000 00000000 00000000 00000000
[  158.953110] sof-audio-pci 0000:00:0e.0: 0xbe06137c: 00000000 00000000 d289ea00 6bc8c5a2
[  158.953116] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  158.953117] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  158.953121] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  158.953124] sof-audio-pci 0000:00:0e.0: ctx_save ipc error -110, proceeding with suspend
[  158.953140] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[  158.953143] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[  158.953146] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[  158.953168] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000010 at  00000044
```

[dmesg_glk.log](https://github.com/thesofproject/sof/files/3650967/dmesg_glk.log)
[trace_log_glk.log](https://github.com/thesofproject/sof/files/3650968/trace_log_glk.log)
[err_log_glk.log](https://github.com/thesofproject/sof/files/3650969/err_log_glk.log)

aiChaoSONG, 2019-09-25:
This issue can also be observed on CML Chrome with [wovapp](https://github.com/aiChaoSONG/wovapp) in S24_LE mode.
run `./wovapp -Dhw:0,8 -d5 -t -s -fs24_le -p 400 -N`
*Notice: this app works fine in s16_le mode*

YvonneYang2, 2019-09-25:
Check on WHL-I2S, same behavior as CML Chrome when run wovapp in s24_le mode. 
This app also woks fine in s16_le mode .

aiChaoSONG, 2019-10-15:
I bisect commit by commit, sof commit a35ec91 cause this issue. As the commit is from 40 days ago, please look at it and fix this issue asap. @mrajwa

lgirdwood, 2019-11-04:
@aiChaoSONG can you confirm these WoV params are supported ? Otherwise this result would be expected. @keyonjie care to comment on what rates/formats are supported here.

keyonjie, 2019-11-04:
@lgirdwood I think the error happens because either 'kpb->sel_sink' or 'kpb->host_sink' is NULL. @mrajwa can you help check what happens here?

mrajwa, 2019-11-05:
@aiChaoSONG, please paste whole firmware error log. Your bisect candidate is likely wrong too.

tlauda, 2020-03-12:
Ancient bug. Everything is working now.

closed 2020-03-12 by tlauda