thesofproject/sof #1877: [BUG] interrupt arecord after wov triggered causes ipc timeout and dsp panic

aiChaoSONG, 2019-09-26:
**Describe the bug**
interrupt wov pipeline when it is triggered will cause ipc timeout and dsp panic.

**To Reproduce**
1. run `arecord -Dhw:0,8 -M -N -r 16000 -c2 -fs16_le -vvv -d5 wov.wav`
2. make some sound to trigger recording.
3. once triggered, press `CTRL + C` to interrupt arecord.

**Reproduction Rate**
100%

**Expected behavior**
arecord exit successfully, no error occur

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev {`e03f8829`}
    * SOF: master {`3ce82a15`}
2) Name of the topology file
    * Topology: {sof-cml-rt5682-max98357a.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML Chrome}

```
[   64.157602] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   64.157820] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60030000 size 12
[   64.157832]  DMIC16k: ASoC: trigger FE failed -16
[   64.157901] sof-audio-pci 0000:00:1f.3: pcm: free stream 8 dir 1
[   64.157909] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   64.157998] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   64.158009] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   64.158020] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   64.158309] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[   64.158314] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   64.661425] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   64.661446] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   64.661707] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[   64.661735] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   64.661740] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   64.661757] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   64.661762] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   64.661930] sof-audio-pci 0000:00:1f.3: pcm: close stream 8 dir 1
[   64.662059] sof-audio-pci 0000:00:1f.3: received event 4 for widget DETECT9.0
[   64.662064] sof-audio-pci 0000:00:1f.3: received event 8 for widget DETECT9.0
[   64.662072] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   65.165364] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   65.165377] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   65.165539] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[   65.165553] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   65.165555] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   65.165565] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   65.165567] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   65.165570] sof-audio-pci 0000:00:1f.3: error: failed to trigger DETECT9.0
[   65.165572] sof-audio-pci 0000:00:1f.3: error: failed to trigger widget DETECT9.0
[   65.165576] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   65.669424] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   65.669445] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   65.669709] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[   65.669738] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   65.669743] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   65.669761] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   65.669766] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   65.669773] sof-audio-pci 0000:00:1f.3: error: failed to trigger DETECT9.0
[   65.669779] sof-audio-pci 0000:00:1f.3: error: failed to trigger widget DETECT9.0
[   65.669786] sof-audio-pci 0000:00:1f.3: ASoC: POST_PMD: DETECT9.0 event failed: -110
[   67.814333] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[   67.834019] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   68.341426] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   68.341446] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   68.341706] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[   68.341729] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   68.341734] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   68.341751] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   68.341757] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[   68.341807] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[   68.341821] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   68.341828] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[   68.341920] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
```
[dmesg.log](https://github.com/thesofproject/sof/files/3656452/dmesg.log)
[trace_log.log](https://github.com/thesofproject/sof/files/3656453/trace_log.log)
[err_log.log](https://github.com/thesofproject/sof/files/3656454/err_log.log)

lgirdwood, 2019-09-30:
@mrajwa @keyonjie fyi - I'm assuming you are still working on this.

wwittbrx, 2019-10-07:
This issue cannot be reproduced with d1f4320. Please check it on your side so we can close it.

aiChaoSONG, 2019-10-08:
The behavior is normal, but I still see some error messages:
**dmesg**
```
[  168.172885] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60030000 size 12
[  168.172896]  DMIC16k: ASoC: trigger FE failed -16
```
**sof-logger**
```
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          MEM         348195876.562500 348195872.000000      src/lib/alloc.c:273 	error: 171 blocks needed for allocation but only 170 blocks are remaining
    0      1         PIPE         352125060.572917   3929184.000000 src/audio/pipeline.c:760 	pipeline_reset() error: ret = -16, host->comp.id = 40
```

tlauda, 2019-10-30:
@aiChaoSONG It's false positive error. The original issue is gone.

closed 2019-10-30 by tlauda