thesofproject/sof #1885: [BUG] [CNL] Firmware load fails on 5.4 mainline kernel

tokyovigilante, 2019-09-27:
**Describe the bug**
HW - Lenovo X1 Yoga G4 (i7-8665U with Cannon Lake Audio), Fedora 31 beta.

I've built a 5.4/mainline kernel (commit da05b5ea12c1e50b2988a63470d6b69434796f8b) with the 5.4 ALSA/sound fixes merged, and SOF from master (commit 913644b1a837fdcb21839584310063b7eb9e4720). I get the kernel log:
```
[   35.470052] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   35.470057] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   35.470198] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   35.474226] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   35.483498] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   35.483499] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   35.520225] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   35.520227] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   38.596599] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   38.596629] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[   38.596650] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   38.596652] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   38.596653] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
```
I realise this may be an unsupported configuration, however the 5.3 SOF code and firmware provided by Lenovo (somewhat unofficially) at https://forums.lenovo.com/t5/Ubuntu/Guide-X1-Carbon-7th-Generation-Ubuntu-compatability/m-p/4503771#M2849 does work, although with very low volume from the rear speakers (Laptop has two woofers and two tweeters) which I am hoping will improve with newer firmware.

**To Reproduce**
Install mainline kernel and SOF firmware/topolgy from head. Attempt to boot kernel and use sound.
 
**Reproduction Rate**
100%

**Expected behavior**
Successful firmware load with working sound output.

**Impact**
Annoyance unless it's movie night, then showstopper.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: mainline da05b5ea12c1e50b2988a63470d6b69434796f8b 
    * SOF: 913644b1a837fdcb21839584310063b7eb9e4720
2) Name of the topology file
    * Topology:  Unknown but all built topology is in /lib/firmware/intel/sof-tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CNL

plbossart, 2019-09-27:
@tokyovigilante did you mean that you compiled the firmware yourself with GCC and using the community key? If yes, I don't think this can work. You'd have to use the 1.3 firmware signed with the Intel production key.

tokyovigilante, 2019-09-27:
Thanks for the reply. That's correct, I just followed the build instructions. Is that signing process available to end users?

plbossart, 2019-09-27:
Unfortunately no. The OEM decides which keys are used. You can only sign your own firmware on 'open' platforms such as Up2 boards or Chromebooks based on GeminiLake or newer.

Also the issue is unlikely to be due to the firmware. The firmware only provides stereo to the HDaudio codec, and I would *guess* the split between tweeter/woofer is handled in the amplifier stages - outside of the chipset and it may be a codec configuration issue.

One way to test this hypothesis would be to use the legacy driver - by blacklisting snd-sof-pci and testing with the snd-hda-intel driver if the sound levels are acceptable. Disclaimer: playing with your hardware and sound levels may result in damaged speakers, use at your own risk, etc. Also the legacy driver will not support the capture over DMIC, so this is only to check the codec behavior.

tokyovigilante, 2019-09-28:
Thanks for the clarification. That is a little disappointing, as it does detract from the open nature of the project if I'm unable to use it outside an OEM-provided build. Lenovo has provided one for the 5.3 kernel to be fair though.

However on rebooting to the current 5.3 release kernel provided by Fedora (Linux firebert 5.3.1-300.fc31.x86_64 #1 SMP Tue Sep 24 21:00:05 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux) without replacing my custom built firmware, the sound driver and firmware seem to be loaded fine and the sound does work as below:
```
[   24.906671] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   24.906677] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   24.906831] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   24.912522] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   24.917077] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   24.917079] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   24.944652] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   24.944653] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   24.952787] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[   25.027456] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-5dd9a
[   25.027457] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:8:0
[   25.027691] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   25.036126] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp1
[   25.036127] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp2
[   25.036127] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link iDisp3
[   25.036128] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Analog Playback and Capture
[   25.036129] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link Digital Playback and Capture
[   25.036129] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic01
[   25.036130] skl_hda_dsp_generic skl_hda_dsp_generic: info: override FE DAI link dmic16k
[   25.037131] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:8:0
[   25.037134] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   25.038150] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   25.038153] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   25.038960] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   25.038961] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   25.039826] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   25.039828] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   25.042369] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   25.042370] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   25.042372] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   25.042373] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   25.042374] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   25.042375] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   25.043629] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   25.043631] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   25.044454] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   25.044455] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   25.045479] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   25.055732] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   25.059532] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 5:0
[   25.059635] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0
[   25.059740] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 7:0
[   25.066489] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   25.066489] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[   25.066490] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   25.066491] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[   25.066491] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[   25.066492] snd_hda_codec_realtek ehdaudio0D0:      Mic=0x19
[   25.103780] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[   25.103789] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[   25.103794] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[   25.103802] skl_hda_dsp_generic skl_hda_dsp_generic: Analog Codec DAI <-> Analog CPU DAI mapping ok
[   25.104325] skl_hda_dsp_generic skl_hda_dsp_generic: Digital Codec DAI <-> Digital CPU DAI mapping ok
[... cut ...]
[   48.563090] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 7:0
[   49.076023] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:0
[   49.077323] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:0
[   49.077485] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 6:0
[   49.077652] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 7:0
```
I'll try your suggestions regarding the legacy HDA driver and possibly chase up the findings with lenovo.

tokyovigilante, 2019-10-08:
On further review it seems you are correct and only signed firmware is able to be loaded on my ThinkPad. I have reverted to the signed 1.3 firmware and topology, but this is not compatible with the current sof-dev kernel branch (with the recent 5.4 rc1 merge) as follows:

```
[   34.110594] sof-audio-pci 0000:00:1f.3: PCI DSP detected
[   34.179098] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   34.179100] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   34.179160] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
[   34.179161] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
[   34.179164] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
[   34.179165] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
[   34.179168] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
[   34.179169] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
[   34.179172] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
[   34.179173] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
[   34.179176] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
[   34.179176] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
[   34.179184] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
[   34.179187] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
[   34.179188] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
[   34.179261] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   34.179262] sof-audio-pci 0000:00:1f.3: using HDA IRQ 173
[   34.179304] sof-audio-pci 0000:00:1f.3: using IPC IRQ 173
[   34.179334] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   34.179335] sof-audio-pci 0000:00:1f.3: PP capability, will probe DSP later.
[   34.179337] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[   34.179340] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   34.183356] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   34.183358] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[   34.183365] sof-audio-pci 0000:00:1f.3: No ASoC machine driver found - using nocodec
[   34.183383] sof-audio-pci 0000:00:1f.3: loading firmware
[   34.186872] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[   34.186876] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   34.186883] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   34.186885] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[   34.186886] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[   34.186916] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   34.186929] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   34.186931] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[   34.186932] sof-audio-pci 0000:00:1f.3: periods:1
[   34.186959] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[   34.186972] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[   34.186976] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[   34.186980] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   34.187652] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[   34.187656] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[   34.187689] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   34.187690] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   34.254387] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[   34.267285] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
[   34.267906] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[   34.267908] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[   34.272305] sof-audio-pci 0000:00:1f.3: ipc rx: 0x70000000
[   34.272307] sof-audio-pci 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
[   34.272415] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-5dd9a
[   34.272417] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:11:0
[   34.272638] sof-audio-pci 0000:00:1f.3: found ext header type 1 size 0xb8
[   34.272763] sof-audio-pci 0000:00:1f.3:  mailbox upstream 0x81000 - size 0x1000
[   34.272764] sof-audio-pci 0000:00:1f.3:  mailbox downstream 0xa0000 - size 0x2000
[   34.272765] sof-audio-pci 0000:00:1f.3:  stream region 0xc1000 - size 0x1000
[   34.272769] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x70000000
[   34.272797] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   34.272802] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   34.272806] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[   34.272814] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[   34.272817] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[   34.272819] sof-audio-pci 0000:00:1f.3: SOF firmware trace disabled
[   34.272886] sof-nocodec sof-nocodec: info: override BE DAI link NoCodec-0
[   34.272887] sof-nocodec sof-nocodec: info: override BE DAI link NoCodec-1
[   34.272901] sof-nocodec sof-nocodec: info: override BE DAI link NoCodec-2
[   34.272902] sof-nocodec sof-nocodec: info: override BE DAI link NoCodec-3
[... 530 lines cut ...]
[   38.339452] sof-audio-pci 0000:00:1f.3: periods:1
[... cut ...]
[   38.435461] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[   38.435464] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000
[   38.435583] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000
```
Neither ALSA nor Pulseaudio detect the hardware. I assume this is due to incompatibilities with the SOF firmware v1.3 and the 5.4 kernel changes.

Going forward, would it be possible to host signed builds of RC versions (or better nightlies) of the firmware for the various desktop/laptop platforms?

plbossart, 2019-10-08:
yes we have plans to provide signed releases. I am not sure we can do signed nightly builds due to logistics, @lgirdwood would need to comment.

The problem you have is a bug on our side, we should be backwards compatible with the 1.3 firmware and topology even when updating the kernel. @ranj063 @kv2019i does the dmesg trace above ring a bell?

@xiulipan @wenqingfu as discussed we need to have a kernel CI test against 1.3 firmware and topology to catch problems with non-backwards compatible issues before they are reported by users.

tokyovigilante, 2019-10-08:
Thanks again for the reply, that's good to know releases will be ongoing.

I should mention the above dmesg trace is after symlinking from `sof-hda-generic.tplg` to `sof-cnl-nocodec.tplg`. Lenovo provided only the `sof-hda-generic.tplg` and without the symlink (but with the 1.3 firmware and the 5.4 kernel i get:

<detail>
<summary>
dmesg
</summary>

```
[   58.868443] sof-audio-pci 0000:00:1f.3: PCI DSP detected
[   58.868716] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   58.868718] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   58.868767] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
[   58.868768] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
[   58.868771] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
[   58.868772] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
[   58.868775] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
[   58.868776] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
[   58.868779] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
[   58.868780] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
[   58.868783] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
[   58.868783] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
[   58.868800] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
[   58.868803] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
[   58.868804] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
[   58.868874] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   58.868875] sof-audio-pci 0000:00:1f.3: using HDA IRQ 173
[   58.868917] sof-audio-pci 0000:00:1f.3: using IPC IRQ 173
[   58.868948] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   58.868949] sof-audio-pci 0000:00:1f.3: PP capability, will probe DSP later.
[   58.868952] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[   58.868956] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   58.873012] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   58.873014] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[   58.873021] sof-audio-pci 0000:00:1f.3: No ASoC machine driver found - using nocodec
[   58.873090] sof-audio-pci 0000:00:1f.3: loading firmware
[   58.874624] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[   58.874627] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   58.874634] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   58.874636] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[   58.874637] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[   58.874666] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   58.874678] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   58.874680] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[   58.874680] sof-audio-pci 0000:00:1f.3: periods:1
[   58.874702] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[   58.874708] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[   58.874709] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[   58.874713] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   58.875681] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[   58.875684] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[   58.915452] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   58.915454] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   58.940795] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[   58.954928] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
[   58.955443] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[   58.955453] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[   58.957948] sof-audio-pci 0000:00:1f.3: ipc rx: 0x70000000
[   58.957950] sof-audio-pci 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
[   58.958231] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-5dd9a
[   58.958232] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:11:0
[   58.958708] sof-audio-pci 0000:00:1f.3: found ext header type 1 size 0xb8
[   58.958808] sof-audio-pci 0000:00:1f.3:  mailbox upstream 0x81000 - size 0x1000
[   58.958809] sof-audio-pci 0000:00:1f.3:  mailbox downstream 0xa0000 - size 0x2000
[   58.958810] sof-audio-pci 0000:00:1f.3:  stream region 0xc1000 - size 0x1000
[   58.958821] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x70000000
[   58.959237] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   58.959257] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   58.959261] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[... 41 lines cut ...]
[   62.531168] sof-audio-pci 0000:00:1f.3: periods:1
[   62.531262] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[   62.531291] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[   62.531295] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[   62.531303] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   62.531807] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[... cut ...]
[   62.627691] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   62.627702] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   62.627706] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[   62.627714] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[   62.627723] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[   62.627726] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000
[   62.627921] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000
```

</detail>

plbossart, 2019-10-08:
I think you have a config issue for your kernel, most likely you didn't select HDAudio support.

please make sure you have this enabled.
CONFIG_SND_SOC_SOF_HDA_LINK=y
CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y

tokyovigilante, 2019-10-09:
I reviewed my config and those opions were indeed not present, however I could not find them even after running `make oldconfig` with the standard Fedora config copied over, nor with `make olddefconfig`. However they seemed to build after merging the relevent fragments from `sof/kconfig`

However I am still unable to load the 1.3 firmware/topology:

<details>
<summary>dmesg:</summary>

```
[   33.434152] sof-audio-pci 0000:00:1f.3: PCI DSP detected
[   33.434794] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   33.435657] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   33.435912] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0xc00
[   33.435914] sof-audio-pci 0000:00:1f.3: found ML capability at 0xc00
[   33.435921] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x800
[   33.435922] sof-audio-pci 0000:00:1f.3: found DSP capability at 0x800
[   33.435927] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x500
[   33.435927] sof-audio-pci 0000:00:1f.3: found GTS capability at 0x500
[   33.435938] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x1f00
[   33.435939] sof-audio-pci 0000:00:1f.3: found DRSM capability at 0x1f00
[   33.435949] sof-audio-pci 0000:00:1f.3: checking for capabilities at offset 0x700
[   33.435950] sof-audio-pci 0000:00:1f.3: found SPIB capability at 0x700
[   33.435956] sof-audio-pci 0000:00:1f.3: DMA mask is 64 bit
[   33.435959] sof-audio-pci 0000:00:1f.3: hda global caps = 0x1009701
[   33.435961] sof-audio-pci 0000:00:1f.3: detected 9 playback and 7 capture streams
[   33.438058] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   33.438060] sof-audio-pci 0000:00:1f.3: using HDA IRQ 173
[   33.438094] sof-audio-pci 0000:00:1f.3: using IPC IRQ 173
[   33.438168] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   33.438169] sof-audio-pci 0000:00:1f.3: PP capability, will probe DSP later.
[   33.438178] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[   33.438188] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   33.443254] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   33.443263] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[   33.443286] sof-audio-pci 0000:00:1f.3: No ASoC machine driver found - using nocodec
[   33.443353] sof-audio-pci 0000:00:1f.3: loading firmware
[   33.452958] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[   33.452966] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   33.452987] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   33.452997] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[   33.452998] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[   33.453035] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   33.453096] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   33.453106] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[   33.453107] sof-audio-pci 0000:00:1f.3: periods:1
[   33.453218] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[   33.453281] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[   33.453291] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[   33.453305] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   33.454196] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[   33.454449] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010e0e successful
[   33.454549] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   33.454551] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   33.518258] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[   33.530609] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
[   33.530629] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[   33.530637] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
[   33.535460] sof-audio-pci 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
[   33.535461] sof-audio-pci 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
[   33.535755] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-5dd9a
[   33.535757] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:7:0 Kernel ABI 3:11:0
[   33.536232] sof-audio-pci 0000:00:1f.3: found ext header type 1 size 0xb8
[   33.536347] sof-audio-pci 0000:00:1f.3:  mailbox upstream 0x81000 - size 0x1000
[   33.536348] sof-audio-pci 0000:00:1f.3:  mailbox downstream 0xa0000 - size 0x2000
[   33.536349] sof-audio-pci 0000:00:1f.3:  stream region 0xc1000 - size 0x1000
[   33.536354] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x70000000: FW_READY
[   33.536359] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   33.536383] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   33.536393] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000002 at  00000048
[   33.536425] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000004 at  00000044
[   33.536435] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000044
[... 509 lines cut ...]
[   33.559334] sof-audio-pci 0000:00:1f.3: error: failed to load DSP topology -22
[   33.559335] sof-audio-pci 0000:00:1f.3: ASoC: failed to probe component -22
[   33.559359] sof-nocodec sof-nocodec: ASoC: failed to instantiate card -22
[... cut ...]
[   36.548004] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[   36.548054] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   36.548071] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[   36.548082] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[   36.548290] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   36.548300] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
```

</details>

tokyovigilante, 2019-10-09:
That's with the symlink removed, the only 1.3 topology file Lenovo is currently providing is `sof-hda-generic.tplg`, so I assume those messages relate to the nocodec topology being absent.

xiulipan, 2019-10-10:
@plbossart @mengdonglin 
We will try to add a matrix of test for daily test for CI:
FW: release 1.3/1.4/master
Kernel: sof-dev/linux/audio for-next
Any other release version need to be tracked?

plbossart, 2019-10-10:
@xiulipan it'd be good if you can test integration/soundwire-latest with the SoundWire options enabled on older platforms w/o SoundWire, just to check if things work without any impact from the added SoundWire initializations/parsing. 
Just today I found a case I didn't manage on Up2 w/ hda-generic-idisp so it'd be quite good before going upstream. We should use the latest firmware there. It's a different problem from this issue #1885 really.

PaulVanSchayck, 2019-10-20:
> I think you have a config issue for your kernel, most likely you didn't select HDAudio support.
> 
> please make sure you have this enabled.
> CONFIG_SND_SOC_SOF_HDA_LINK=y
> CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y

I'm hitting a similar looking problem on Ubuntu 19.10 (and also on a Lenovo X1 7th gen) using the Ubuntu kernel 5.3.0-18-generic. When checking its config it reports this:
```
$ grep CONFIG_SND_SOC_SOF_HDA /boot/config-5.3.0-18-generic 
CONFIG_SND_SOC_SOF_HDA_COMMON=m
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=m
```
I'm also getting this error where it loads the nocodec machine driver:
```
[   10.067961] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   10.067965] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040380
[   10.068296] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   10.073342] sof-audio-pci 0000:00:1f.3: No ASoC machine driver found - using nocodec
[   10.081756] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
```
Would this be an issue with this particular kernel config, and we need to talk to the Ubuntu devs to enable the other options? Or is there something else going on?

Sorry for semi-hijacking this issue. But I thought them to be closely related to each other.

plbossart, 2019-10-21:
@PaulVanSchayck can you attach the config? I wonder why nocodec is even selected.

PaulVanSchayck, 2019-10-21:
@plbossart Here you go. Once again, this is the stock Ubuntu kernel for 19.10, built by Canonical.

[config-5.3.0-18-generic.txt](https://github.com/thesofproject/sof/files/3750535/config-5.3.0-18-generic.txt)

plbossart, 2019-10-21:
ok, no idea why Ubuntu enabled 'nocodec'.

````
	  This adds support for a dummy/nocodec machine driver fallback
	  option if no known codec is detected. This is typically only
	  enabled for developers or devices where the sound card is
	  controlled externally
````

These two options should be removed really:
CONFIG_SND_SOC_SOF_NOCODEC=m
CONFIG_SND_SOC_SOF_NOCODEC_SUPPORT=y

That said, this doesn't explain why an HDaudio card is not detected. I'll try with the Ubuntu kernel from git://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/eoan with this config.

plbossart, 2019-10-21:
ok, this is definitively an Ubuntu issue.
They defined NOCODEC support, which is mutually exclusive with HDAudio support! when you remove the NOCODEC support, then a new option shows up for HDAudio support.

The config should be modified with the following diffs 

````diff
6656c6656
< # CONFIG_SND_SOC_INTEL_SKYLAKE is not set
---
> CONFIG_SND_SOC_INTEL_SKYLAKE=m
6661,6664c6661,6664
< # CONFIG_SND_SOC_INTEL_CNL is not set
< # CONFIG_SND_SOC_INTEL_CFL is not set
< # CONFIG_SND_SOC_INTEL_CML_H is not set
< # CONFIG_SND_SOC_INTEL_CML_LP is not set
---
> CONFIG_SND_SOC_INTEL_CNL=m
> CONFIG_SND_SOC_INTEL_CFL=m
> CONFIG_SND_SOC_INTEL_CML_H=m
> CONFIG_SND_SOC_INTEL_CML_LP=m
6667c6667
< # CONFIG_SND_SOC_INTEL_SKYLAKE_HDAUDIO_CODEC is not set
---
> CONFIG_SND_SOC_INTEL_SKYLAKE_HDAUDIO_CODEC=y
6702c6702,6703
< # CONFIG_SND_SOC_SOF_NOCODEC_SUPPORT is not set
---
> CONFIG_SND_SOC_SOF_NOCODEC=m
> CONFIG_SND_SOC_SOF_NOCODEC_SUPPORT=y
6706d6706
< CONFIG_SND_SOC_SOF_PROBE_WORK_QUEUE=y
6734,6735d6733
< CONFIG_SND_SOC_SOF_HDA_LINK=y
< CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y
6737d6734
< CONFIG_SND_SOC_SOF_HDA=m
````

full file here: 
[config-5.3.0-18-fixed.txt](https://github.com/thesofproject/sof/files/3750806/config-5.3.0-18-fixed.txt)

tokyovigilante, 2019-10-21:
It turns out this was the issue for me too sorry, I had the HD audio codec enabled but also the nocodec option, disabling this allows the 1.3 firmware/topology to work with kernel 5.4 and my existing pulseaudio config/ALSA UCM. Sorry about that!

PaulVanSchayck, 2019-10-21:
Thanks for the help and double checks Pierre.

After searching the Ubuntu Kernel tracker, turned out this was already self-reported by the Ubuntu developers a couple of days a go in [this bug report on launchpad](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1848490).

lgirdwood, 2019-11-04:
@plbossart can we close this now with your latest commits being merged ?

tokyovigilante, 2019-11-04:
Firmware load working well for me now with the kernel configs mentioned in the thread.

plbossart, 2019-11-04:
Closing. Thanks @tokyovigilante and @PaulVanSchayck for your help, we appreciate feedback from the community.

closed 2019-11-04 by plbossart