thesofproject/sof #1886: [BUG][BYT][Capture]  Input/output error during pause-release.

Jiangxinx, 2019-09-29:
**Describe the bug**
Since #1578 had been closed, but this issue still can be reproduced on byt nocodec platform,  the reproduction rate is very low (about 1/20).  And sof can continue work fine after this issue occurs,  I also can not observe obvious error trace from dmesg and sof-logger .

**To Reproduce**
1.Arecord via "arecord -Dhw:0,0 -r48000 -c2 -fs16_le 1.wav -vv -i"
2.Pause arecord
3.Resume asrecord

**Reproduction Rate**
1/20

**Environment**
Firmware: 8528d932 https://github.com/thesofproject/sof/commits/master
Kernel: f1c5cf5e https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-byt-nocodec.tplg
platform： BYT

**Screenshots or console output**
```
ubuntu@cd-byt-mb-nocodec-04:~$ arecord -Dhw:0,0 -fs16_LE -c2 -r 48000 1.wav -vv -i
Recording WAVE '1.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
Hardware PCM card 0 'sof-nocodec' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16320
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16320
  silence_threshold: 0
  silence_size : 0
  boundary     : 9187343239835811840
  appl_ptr     : 0
  hw_ptr       : 0
               arecord: pcm_read:2103: read error: Input/output error
```

**dmesg**
```
[ 3875.038983] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 3875.048830] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 1 cmd 4
[ 3875.048838] sof-audio-acpi 80860F28:01: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[ 3875.048932] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[ 3875.048992] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 0 DAI position 228480
[ 3875.132477] sof-audio-acpi 80860F28:01: ipc rx: 0x600a000c: GLB_STREAM_MSG: POSITION
[ 3875.132513] sof-audio-acpi 80860F28:01: posn : host 0x3fc0 dai 0xe7180 wall 0x185638
[ 3875.132542] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a000c: GLB_STREAM_MSG: POSITION
[ 3875.132592] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 4080 DAI position 236640
[ 3875.133848] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 4080 DAI position 236640
[ 3875.217497] sof-audio-acpi 80860F28:01: ipc rx: 0x600a000c: GLB_STREAM_MSG: POSITION
[ 3875.217532] sof-audio-acpi 80860F28:01: posn : host 0x7f80 dai 0xef100 wall 0x313d3a
[ 3875.217561] sof-audio-acpi 80860F28:01: ipc rx done: 0x600a000c: GLB_STREAM_MSG: POSITION
[ 3875.217611] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 8160 DAI position 244800
[ 3875.218997] sof-audio-acpi 80860F28:01: PCM: stream 0 dir 1 DMA position 8160 DAI position 244800
[ 3875.219018] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 1 cmd 3
[ 3875.219033] sof-audio-acpi 80860F28:01: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 3875.219160] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[ 3875.220779] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 1 cmd 4
[... 96 lines cut ...]
    0      2       VOLUME         3710945734.947917         2.083333 udio/volume/volume.c:488   volume_ctrl_get_cmd(), channel = 1, value = 65536
    0      2          IPC         3710945827.343750        92.395836    src/ipc/handler.c:840   ipc: hdr 0x50020000 tx (108) > rx (92)
    0      2          IPC         3710945829.322917         1.979167    src/ipc/handler.c:842   ipc: comp 3 -> cmd 0
    0      2       VOLUME         3710945831.718750         2.395833 udio/volume/volume.c:511   volume_cmd()
    0      2       VOLUME         3710945833.906250         2.187500 udio/volume/volume.c:481   volume_ctrl_get_cmd(), SOF_CTRL_CMD_VOLUME / SOF_CTRL_CMD_SWITCH, cdata->comp_id = 3
    0      2       VOLUME         3710945835.989583         2.083333 udio/volume/volume.c:488   volume_ctrl_get_cmd(), channel = 0, value = 65536
    0      2       VOLUME         3710945838.072917         2.083333 udio/volume/volume.c:488   volume_ctrl_get_cmd(), channel = 1, value = 65536
    0      2          IPC         3710945932.031250        93.958336    src/ipc/handler.c:840   ipc: hdr 0x50020000 tx (108) > rx (92)
    0      2          IPC         3710945934.062500         2.031250    src/ipc/handler.c:842   ipc: comp 2 -> cmd 0
    0      2       VOLUME         3710945936.354167         2.291667 udio/volume/volume.c:511   volume_cmd()
    0      2       VOLUME         3710945938.541667         2.187500 udio/volume/volume.c:481   volume_ctrl_get_cmd(), SOF_CTRL_CMD_VOLUME / SOF_CTRL_CMD_SWITCH, cdata->comp_id = 2
    0      2       VOLUME         3710945940.625000         2.083333 udio/volume/volume.c:488   volume_ctrl_get_cmd(), channel = 0, value = 65536
    0      2       VOLUME         3710945942.708333         2.083333 udio/volume/volume.c:488   volume_ctrl_get_cmd(), channel = 1, value = 65536
    0      2          IPC         3832891735.104167 121945792.000000    src/ipc/handler.c:224   ipc: comp 12 -> params
    0      2          DMA         3832891738.541667         3.437500 src/drivers/dw/dma.c:180   dw_dma_channel_get(): dma 0 request channel 0
    0      2          DMA         3832891918.229167       179.687500 src/drivers/dw/dma.c:238   dw_dma_channel_put(): dma 0 channel 0 put
    0      2         PIPE 2.17    3832891930.572917        12.343750 src/audio/pipeline.c:340   pipeline_params()
    0      2         HOST 2.12    3832891933.177083         2.604167     src/audio/host.c:563   host_params()
    0      2          DMA         3832891951.354167        18.177084 src/drivers/dw/dma.c:180   dw_dma_channel_get(): dma 0 request channel 65535
    0      2       VOLUME         3832891959.218750         7.864583 udio/volume/volume.c:244   volume_params()
    0      2          DAI 2.16    3832891962.343750         3.125000      src/audio/dai.c:387   dai_params()
    0      2          DAI 2.16    3832891964.947917         2.604167      src/audio/dai.c:334   dai_capture_params() src_dev = 4 stream_id = 0 src_width = 4 dest_width = 4
    0      2          DAI 2.16    3832891967.239583         2.291667      src/audio/dai.c:360   dai_capture_params() fifo FF2A2010
    0      2         PIPE 2.17    3832891972.708333         5.468750 src/audio/pipeline.c:399   pipeline_prepare()
```

tlauda, 2019-09-30:
@Jiangxinx It is very similar to the sporadic fail with alsabat. No errors in FW, but ALSA for some reason detects the XRUN and stops pipeline. @juimonen You think https://github.com/thesofproject/linux/pull/1219 can potentially fix it?

juimonen, 2019-09-30:
@tlauda it could help... worth testing anyway. I guess in the thesofproject/linux#1219 general agreement was that having pm_downtime 0 reveals something that is the root cause for this, and as the case is quite hairy (and I have other stuff to do) I didn't have time to actually figure out what is the problem.

tlauda, 2019-10-02:
@Jiangxinx Should go to kernel repository.

closed 2019-10-02 by tlauda