thesofproject/sof #1899: [FEATURE] Kaby Lake support

chrysophylax, 2019-10-03:
**Is your feature request related to a problem? Please describe.**
Kaby Lake devices are currently not supported by SOF. There are many of us who would appreciate having audio out through our audio jack & speakers.
Quoted from the documentation

> while the Sky Lake and Kaby Lake platforms are also based on the cAVS 1.5 architecture, they are not supported at this time due to differences in boot flow and memory architecture.

**Describe the solution you'd like**
A roadmap to support for Kaby Lake devices, e.g. i5-8250u or some concrete description of how the platforms differ despite being based on the cAVS 1.5 architecture. It would permit individuals to contribute labour to this project.

**Describe alternatives you've considered**
Ripping out blobs from a ChromeOS recovery image as per https://github.com/GalliumOS/galliumos-distro/issues/379#issuecomment-403285839

**Additional context**
00:1f.3 Multimedia audio controller: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
	Flags: bus master, fast devsel, latency 64, IRQ 127
	Memory at d112c000 (64-bit, non-prefetchable) [size=16K]
	Memory at d1110000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_soc_skl

**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

lgirdwood, 2020-01-06:
@chrysophylax thanks, we have been internally looking at how to make SKL/KBL platforms more FW developer friendly but we are blocked by immutable signing keys which means only Intel can sign code for these platforms (making external community collaboration practically impossible). On APL onwards the signing keys are changeable and can be updated by OEM, etc. Sorry about this. I will update this FEATURE is anything changes though.

Fwiw, I think the fastest way atm to use SOF would be to use on a newer platform (e.g. Google Chromebooks use a public key so you can sign the code yourself in Chrome developer mode).

plbossart, 2020-01-06:
@lgirdwood see https://bugzilla.kernel.org/show_bug.cgi?id=201251, there are quite a few folks who can't get the DMICs to work on KBL platforms. Even if we only provide an Intel signed version of the SOF firmware, that would still be very helpful.

lgirdwood, 2020-01-06:
@plbossart I agree, it would be helpful to provide this signed version for working DMIC, I didn't realise the DMIC had multiple issues (kernel, ACPI, topology) on KBL.
@lbetlej @mmaka1 what would it take to validate and sign base level functionality on KBL ? We already have the SOF memory map and build support for KBL but it needs validated and signed. I have some kernel patches to support the boot for changes, but I cant test due to code signing.

chrysophylax, 2020-01-06:
It's true, even an Intel-signed only version would be very helpful for a big pool of users. I had issues with almost everything on my Intel i5-8250u until I ripped out binary blobs from my chromebook image (see https://github.com/GalliumOS/galliumos-distro/issues/536) and jammed it into my Linux install. It'd be much nicer to do this in a non-hacky way.

plbossart, 2020-01-06:
> It's true, even an Intel-signed only version would be very helpful for a big pool of users. I had issues with almost everything on my Intel i5-8250u until I ripped out binary blobs from my chromebook image (see [GalliumOS/galliumos-distro#536](https://github.com/GalliumOS/galliumos-distro/issues/536)) and jammed it into my Linux install. It'd be much nicer to do this in a non-hacky way.

You are lucky in a way, the configuration for topology/SSPs would work on a Chromebook, but for a generic HDAudio solution those binary blobs don't even exist, the combination of HDaudio+DMIC was not enabled with the KBL driver. So it's not really about doing things in a a less hacky-way but really enable those devices.

cujomalainey, 2020-01-06:
This would be excellent to have for when it comes time for KBL kernel uprev. Will make my life much easier when it comes to enabling the new kernel. It would help with debugging having the insight into the firmware. Is there anything that I can help do to enable this?

dgreid, 2020-01-07:
Of the two options, it would be best if we had some kind of stub that allowed Chrome OS to sign firmware.
However, having an Intel-signed SoF blob for kabby lake will make our lives much easier when it's time to move KBL off of 4.4 kernels. If that's an easier request, I'm all for going that route.

plbossart, 2020-01-07:
@lgirdwood to make Chrome/KBL off of 4.4 kernels, we'd need to enable the SSP clock control from the machine drivers, they use clk_prepare_enable/disable for the MCLK and BCLK

https://github.com/thesofproject/linux/blob/52ec20e276a6b49f74f77b18e04cf1b1359b75ae/sound/soc/intel/boards/kbl_rt5663_rt5514_max98927.c#L102

That's an added functionality that's not needed for regular HDAudio+DMIC cases

lgirdwood, 2020-01-07:
@plbossart ok, thanks - Btw I've pushed my kernel patches here
https://github.com/thesofproject/linux/tree/lrg/topic/kbl-fw-loader

These have only been tested on early KBL HW, this needs completed by someone with mature KBL development HW (and can sign code).

So in order to have working SOF on KBL with DMIC it looks like we would need (off the top of my head):- 

1) Completion of driver codeloader.
2) Machine SSP clock control.
3) Validation of FW on KBL.
4) SOF binaries signed for KBL.
5) KBL HW in CI
6) 

Anything else ? @lbetlej can you think of anything else here?

I don't see any great technical issues with 1 - 4 (but I could be easily wrong), it looks more like availability of mature KBL development HW for developers and CI would be enough to test and validate.

cujomalainey, 2020-01-07:
> Of the two options, it would be best if we had some kind of stub that allowed Chrome OS to sign firmware.
> However, having an Intel-signed SoF blob for kabby lake will make our lives much easier when it's time to move KBL off of 4.4 kernels. If that's an easier request, I'm all for going that route.

I will make a bug for this so we can track this independently

plbossart, 2020-01-08:
I think we should split the work. The more urgent is to make sure there is a solution for plain vanilla HDAudio+DMIC, so that would be 

a) Completion of driver codeloader.
b) firmware generation and signature
c) firmware/driver cross-validation on HDaudio+DMIC platforms
d) firmware/driver in CI

and then for Chrome
e) Machine SSP clock control.
f) smart-amp integration capabilities (used on some versions of KBL)

We'd also need the smart-amp processing used on some versions of KBL (can't recall which one).

FantasyCookie17, 2020-01-12:
I could test software made for solving this issue, as I have an AKALI360 as well. However I don't think I'm good enough at programming to seriously contribute to the development apart from this. Does the `alsa-firmware` package available in most distros contain the SOF?

lgirdwood, 2020-01-13:
@FantasyCookie17 AKALI360 should be also be a good platform for testing thanks. FW binaries will be heading for linux-firmware package, we are still sorting out mechanics of this.

FantasyCookie17, 2020-02-19:
Is there anywhere I can track progress on this? Or should I just look at which kind of pull requests and commits in general get accepted?

plbossart, 2020-02-19:
> Is there anywhere I can track progress on this? Or should I just look at which kind of pull requests and commits in general get accepted? (apart from looking at #2256)

it's rather easy to track progress: there's been no progress since January...

chrysophylax, 2020-02-20:
Is there something the less-technical community can do to help? test stubs or such? I have a AKALI360 I'm willing to do software surgery on for the cause.

plbossart, 2020-02-20:
> Is there something the less-technical community can do to help? test stubs or such? I have a AKALI360 I'm willing to do software surgery on for the cause.

unfortunately the bulk of the work is very low-level boot flows and the firmware would again need to be signed by Intel for devices designed for Windows. It's going to happen but we've several Q1 topics already.

chrysophylax, 2020-02-20:
OK, thanks for the reassuring words that it's not forgotten :-)

wmertens, 2020-05-25:
Commenting to help find other users find this issue:

If you have an Acer Swift 3 with an 8th gen Intel CPU and you want your Intel Digital Microphone Array (DMIC) under Linux, then you need to subscribe to this issue.

3ddd, 2020-05-25:
what type of controller do you have on Acer? This one?
lspci -nn | grep Audio
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)

wmertens, 2020-05-25:
yes correct
```
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)
```

The realtek codec is ALC265

3ddd, 2020-05-25:
would that Request affect my x360 1040 G5 with Conexant Codec?
Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)
	Subsystem: Hewlett-Packard Company Sunrise Point-LP HD Audio [103c:8470]

!!HDA-Intel Codec information
!!---------------------------
--startcollapse--

Codec: Conexant Generic
Address: 0
AFG Function Id: 0x1 (unsol 1)
Vendor Id: 0x14f120d0
Subsystem Id: 0x103c8470
Revision Id: 0x100001

daric93, 2020-08-08:
I have found this issue after unsuccessfully trying to get microphone working on my Lenovo Yoga C930.
Audio card is identical to the reported above:
```
$ lspci -nn | grep Audio
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)
```

lgirdwood, 2020-09-22:
After some investigation it looks like running SOF on SKL or KBL wont be possible at this time due to an immutable audio DSP signing key (which is also used by all OSes for audio authentication) on the platform.  Closing this.

closed 2020-09-22 by lgirdwood

satmandu, 2020-09-23:
There was talk earlier this year by @plbossart about the possibility of working to get intel to sign firmware for the devices with Windows firmware.  Is there any chance of that happening?

(Some devices like Kaby Lake Chromebooks, such as nocturne, should have signed firmware already for use with ChromeOS's cras. Is it possible to get signed firmware for these which would work with sof?)

(And would it be useful to leave an issue open for tracking this until that happens?)

plbossart, 2020-09-24:
> There was talk earlier this year by @plbossart about the possibility of working to get intel to sign firmware for the devices with Windows firmware. Is there any chance of that happening?

it appears that's not going to happen for SKL/KBL. Not my decision.

FantasyCookie17, 2020-09-24:
What's the rationale for that? Did Intel refuse to do it?

sykhro, 2021-02-09:
> After some investigation it looks like running SOF on SKL or KBL wont be possible at this time due to an immutable audio DSP signing key (which is also used by all OSes for audio authentication) on the platform. Closing this.

What does this mean, exactly? That key can't be reused to sign new firmware?

chrysophylax, 2021-02-09:
> What does this mean, exactly? That key can't be reused to sign new firmware?

Give it up peeps, Intel has decided to leave us behind. I'll be selling my Intel hardware and moving to AMD this year.

plbossart, 2021-02-09:
See https://gist.github.com/crojewsk/4e6382bfb0dbfaaf60513174211f29cb for Kabylake support.

chrysophylax, 2021-02-09:
Thank you for the workaround!

kreaninw, 2021-02-09:
> Thank you for the workaround!

Is it working?

TilmanRpk, 2021-05-14:
> > Thank you for the workaround!
> 
> Is it working?

YES

rotech, 2021-08-30:
The solution posted by plbossart only works for Realtek codecs.  Any possibility of support for other codecs?  I have a Conexant CX11970 (0x14f120d0).

elzii, 2021-12-11:
> > > Thank you for the workaround!
> > 
> > 
> > Is it working?
> 
> YES

Would you mind detailing the steps you took? Trying to do this on a Pixelbook Eve 2017.

plbossart, 2021-12-13:
> > > > Thank you for the workaround!
> > > 
> > > 
> > > Is it working?
> > 
> > 
> > YES
> 
> Would you mind detailing the steps you took? Trying to do this on a Pixelbook Eve 2017.

See comment above https://github.com/thesofproject/sof/issues/1899#issuecomment-776070707

ZeroQLi, 2023-12-17:
@rotech did you ever figure it out

I am also stuck in the exact same boat as you (kaby-lake with Conexant CX8200 audio codec)

Has there been 0 progress in kaby lake support?

plbossart, 2023-12-18:
This is an AVS driver issue, not SOF. @crojewsk that one is for you.

elzii, 2023-12-25:
Confirmed to be working on Pixelbook EVE 2017 - both i5 and i7 models.