thesofproject/sof #1900: [BUG] docker: sof_qemu fails to build

lgirdwood, 2019-10-03:
**Describe the bug**
Follow getting started and install docker.
run 
```
cd ~/work/sof/sof/scripts/docker_build/sof_qemu
./docker-build.sh
```
Then 
```
Setting up build-essential (12.4ubuntu1) ...
Setting up libcurl3-gnutls:amd64 (7.58.0-2ubuntu3.8) ...
Setting up gpg-wks-client (2.2.4-1ubuntu1.2) ...
Setting up pkg-config (0.29.1-0ubuntu2) ...
Setting up git (1:2.17.1-1ubuntu0.4) ...
Setting up gnupg (2.2.4-1ubuntu1.2) ...
Setting up libglib2.0-dev:amd64 (2.56.4-0ubuntu0.18.04.4) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for ca-certificates (20180409) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Removing intermediate container 1be97a5d160f
 ---> b1cb6bdc2ced
Step 9/14 : RUN useradd --create-home -d /home/sof -u $UID -G sudo sof && echo "sof:test0000" | chpasswd && adduser sof sudo
 ---> Running in c9845200810f
useradd: UID 0 is not unique
The command '/bin/sh -c useradd --create-home -d /home/sof -u $UID -G sudo sof && echo "sof:test0000" | chpasswd && adduser sof sudo' returned a non-zero code: 4
```

xiulipan, 2019-10-08:
@lgirdwood Are you working with ROOT?
the UID should be the user that create the qemu image.

I have some auto build scripts for the qemu. Let get some CI for the qemu also.

lgirdwood, 2019-10-28:
@xiulipan it turns out it needs the group permissions to take effect alongside the non root guid. All working now.

closed 2019-10-28 by lgirdwood