thesofproject/sof #1903: Unclear usage of DMA_CB_TYPE_IRQ in SSI-SPI

paulstelian97, 2019-10-04:
I have noticed after the recent refactor of interrupt handling that almost all usages of DMA_CB_TYPE_IRQ have been removed, but there still is a callback registered for this in SSI-SPI. However I don't see any other usage (including potential call site for this callback) of this macro.

Is this callback type even doing anything right now? Is it alright that it is still there, or to ignore it completely? Does everything work properly in this situation?

Is `spi_completion_work` in the driver called correctly after the refactor?

tlauda, 2019-10-04:
@paulstelian97 Aligned SSI-SPI driver: #1904.

closed 2019-10-04 by tlauda

paulstelian97, 2019-10-04:
Thanks, was unclear for the Dummy DMA if I needed to care about that callback anymore.