thesofproject/sof #1913: [BUG] remove void* arithmetic

kv2019i, 2019-10-08:
**Describe the bug**
SOF codebase has multiple instances of doing arithmetic with void pointer. This is illegal both in ANSI C and C++, although allowed by many compilers.

There's no good reason to have such code in the codebase.

**To Reproduce**
Pass -pedantic-errors or -Werror-pointer-arith to GCC build.

**Reproduction Rate**
100%

**Expected behavior**
No illegal C/C++ constructs used.

**Impact**
When building SOF with Zephyr as the RTOS, this causes issues as void*-arithmetic is forbidden in Zephyr build system.

**Environment**
Latest sof/master.

FYI @finikorg

lgirdwood, 2019-10-28:
@kv2019i do you have an ETA for fix and also GCC flag updates ?

kv2019i, 2019-10-29:
@lgirdwood wrote:
> @kv2019i do you have an ETA for fix and also GCC flag updates ?

No, nobody has signed up to this task yet. This is not on the priority list (at least yet).

paulstelian97, 2019-10-29:
On this side of cosmetic fixes, I think I could get to this one. For the moment, I'm waiting for the asserts PR to be merged and am planning on fixing #1999 first. After those I think I could add this one in my backlog.

paulstelian97, 2019-10-29:
On it, added a flag, draft PR coming up today, full PR possibly also today.

PR #2023 (2019-11-04): sof: build: Disallow void pointer arithmetic
Void pointer arithmetic isn't supported in Zephyr. This pull request removes it.

~Draft because I only added the flag so I get compile errors, without actually removing anything. I believe I should also add the flag on host?~ (no longer draft, host is also covered)

Fixes: #1913
closed 2019-11-04 by jajanusz