thesofproject/sof #1915: [BUG][CML][sof0.7] KWD feature is broken

kamal-ap, 2019-10-09:
**Describe the bug**
SOF release 0.7 
URL: https://github.com/thesofproject/linux/commits/topic/keyword_detect
Commit: 6664b426664b42  Tag: release-sof-cml-kwd-20190809

SOF FW source | URL: https://github.com/thesofproject/sof/commits/cml-007-drop-stable       Commit: 0207c7546fe23607249279162c7742fad0e93475

Chromium backported kernel  : https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/1821883/3

Logs:
[ 4259.051523] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[ 4259.051538] sof-audio-pci 0000:00:1f.3: error: failed to trigger DETECT9.0
[ 4259.051546] sof-audio-pci 0000:00:1f.3: error: failed to trigger widget DETECT9.0
[ 4261.904868] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 4262.008622] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 4425.879145] PDLOG 2019/09/26 05:17:05.779 P0 Disconnected

**To Reproduce**
1) Record using 'arecord -Dhw:0,8 -M -N -c 2 -f S24_LE -r 16000 tmp.wav  -vvv'
2) clap hands and check its detected

**Reproduction Rate**
10/10

**Actual Behaviour**
Claps are not detected

**Expected behavior**
Keyword 'clap sound' should be detected

tlauda, 2019-10-09:
@kamal-ap Could you attach full dmesg and also FW logs from logger app?

wwittbrx, 2019-10-11:
Hi @kamal-ap I have checked your recipe on Ubuntu CML platform and I have reproduced the 'failed to trigger DETECT9.0' error. However, this error happens on pipeline shutdown when we free the components (after GLB_STREAM_MSG: TRIG STOP). It does not have any effect on KD detection and your recorded file should look fine and arecord should exit successfully.

To put it simply the error happens after we did everything we wanted to do and does not affect core KD functionality. The pipeline is rebuilt for next arecord so there should be no side effects, think of this error message it as a warning. 

I encourage you to share more logs if my answer does not reflect your findings. The easiest way to check if we trigger detection is to use sof-logger, which should print notify_host() and notify_kpb() lines when detection happens.

keyonjie, 2019-10-11:
@kamal-ap which platform were you running on? It looks there are difference b/w rt5682 and rt1011.

sathyap-chrome, 2019-10-15:
@keyonjie  This is run on Helios : this uses RT5682 for Headset and RT1011 for SPK AMP.

sathyap-chrome, 2019-10-16:
We got report that this issue is specific to 1 platform and its mostly due to low DMIC gain.
We can close this bug and we can track the specific issue and get back if we need sof team support.

closed 2019-10-17 by kamal-ap