thesofproject/sof #1924: [BUG][TGL] multi-pipeline failed catch 'DSP panic' in dmesg

Bin-QA, 2019-10-12:
**Describe the bug**
run different pipeline failed for TGL

**To Reproduce**
1. In terminal to run "aplay -D hw:0,0 -f dat 48California_Dreaming.wav"
2. In other terminal to run "aplay -D hw:0,4 -f dat 48California_Dreaming.wav"

**Reproduction Rate**
100%

**Expected behavior**
Can run multi-pipeline on TGL

**Impact**
Check ABI

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology)
    * SOF: {TGL v0.2}
2) Name of the platform(s) on which the bug is observed.
    * Platform: {TGL}

**Screenshots or console output**
[  565.126849] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
[  565.126869] sof-audio-pci 0000:00:1f.3: nothing to do in IPC IRQ thread
[  565.127226] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x24001e successful
[  565.127236] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  565.127487] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  565.127503] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[  565.127907] sof-audio-pci 0000:00:1f.3: **error : DSP panic!**
[  565.127914] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[  565.127932] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  565.128379] sof-audio-pci 0000:00:1f.3: error: runtime exception

[aplay.abi.log](https://github.com/thesofproject/sof/files/3720373/aplay.abi.log)
[aplay.dmesg.log](https://github.com/thesofproject/sof/files/3720374/aplay.dmesg.log)
[debug-sof.tar.gz](https://github.com/thesofproject/sof/files/3720375/debug-sof.tar.gz)
[md5sum.txt](https://github.com/thesofproject/sof/files/3720376/md5sum.txt)

Bin-QA, 2019-10-14:
Verify patch:  https://patch-diff.githubusercontent.com/raw/thesofproject/linux/pull/1037.patch
Couldn't catch "POSTION" in dmesg keyword
This Bug still exist

jajanusz, 2019-10-14:
@Bin-QA Repro recipe is incomplete - What kernel did you use?

Bin-QA, 2019-10-14:
@jajanusz internel test kernel

jajanusz, 2019-10-14:
@Bin-QA We don't work on BUGs with kernel that we don't have, if you want us to reproduce error, please give commit hash of kernel that you see error with.

slawblauciak, 2019-10-14:
Please check on a timer-driven pipeline, with SSP or SDW.
To make a pipeline timer-driven in the topology, the 11th param of PIPELINE_PCM_ADD has to be set to SCHEDULE_TIME_DOMAIN_TIMER.

xiulipan, 2019-10-14:
@jajanusz Can you try to dump the exception with sof-coredump-to-gdb.sh?
The exception is attached.

Bin-QA, 2019-10-15:
Add kernel configure for Kernel

[config.txt](https://github.com/thesofproject/sof/files/3728588/config.txt)

closed 2019-10-17 by tlauda

keqiaozhang, 2019-10-17:
@jajanusz 
This issue is gone after applying https://github.com/thesofproject/sof/pull/1955 and https://github.com/thesofproject/sof/pull/1956.

PR #1955 (2019-10-17): tigerlake: increase system runtime heap size
Update memory.h to have similar capabilities to other platforms,
that were updated when dynamic DMA channels were introduced.
Previous values were not enough for some topologies.

Fixes #1924