thesofproject/sof #1925: [BUG][SDW]arecord+aplay will cause I/O error.

YvonneYang2, 2019-10-12:
**Describe the bug**
aplay 0,0/2/3 + arecord 0,4--> ctrl+z and fg on arecord 0,4 --> I/O error
arecord 0,1 and arecord 0,4 --> pause/release --> I/O error
arecord 0,1 + aplay 0,0/2/3 aplay --> I/O error
Seems commit f648e8a (https://github.com/thesofproject/linux/pull/1070/commits) has been removed in soundwire-latest branch
error log is same as https://github.com/thesofproject/sof/issues/1578

**To Reproduce**
1.open terminal A to run command "time aplay -Dhw:0,0 -f dat wav/48California_Dreaming.wav -vv -i"
2.open another terminal B to run command "arecord -Dhw:0,4 -r 48000 -c2 -f s32_le test.wav -vv -i"
3.Ctrl+Z to background arecord in terminal B, then fg to foreground.

**frequency**
100%

**Expected result**
Aplay and arecord work normally.

**Actual result**
aplay and arecord stop, and shows I/O error.

**Environment**
Firmware: c7d0149 https://github.com/thesofproject/sof/commits/master 
Kernel: e7fb5a5 https://github.com/thesofproject/linux/tree/integration/soundwire-latest
tplg:  sof-icl-rt711-rt1308-rt715-hdmi.tplg rename sof-icl-rt711-rt1308-rt715.tplg  path same as FW branch
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

<details>
 <summary>error logger</summary>

````
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 5.28    318102259.270833 318102272.000000      src/audio/dai.c:593 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         318102269.062500         9.791667 of/audio/component.h:759 	comp_overrun() error: dev->comp.id = 28, sink->free = 768, copy_bytes = 0
    0      1         PIPE         318102334.427083        65.364586 src/audio/pipeline.c:854 	pipeline_copy() error: ret = -61, start->comp.id = 28, dir = 0
    0      1         PIPE 5.29    318102346.093750        11.666667 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP         319120822.656250   1018476.562500 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
````
</details>

[fg_error_logger.log](https://github.com/thesofproject/sof/files/3720487/fg_error_logger.log)
[fg_logger.log](https://github.com/thesofproject/sof/files/3720488/fg_logger.log)
[fg_dmesg.log](https://github.com/thesofproject/sof/files/3720489/fg_dmesg.log)

plbossart, 2019-10-12:
No, the patch ASoC: SOF: Intel: hda: Disable DMI L1 entry during capture is absolutely in integration/soundwire-latest. You must have a bad configuration or a different bug appeared. Please recheck, thanks.

bardliao, 2019-10-14:
@YvonneYang2 I can't reproduce the issue with the same kernel and FW as yours.

YvonneYang2, 2019-10-14:
Issue caused by commit 71d9ee4(master branch)

@tlauda  Could you have a look? Thanks a lot.

tlauda, 2019-10-14:
Please provide full FW logs. Current log file is empty.

YvonneYang2, 2019-10-14:
Aplay 0,0 + arecord 0,4 --> ctrl+z and fg on arecord 0,4 --> I/O error

All logs is as below:
[logger_error.log](https://github.com/thesofproject/sof/files/3723196/logger_error.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3723197/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3723198/logger.log)

tlauda, 2019-10-14:
@YvonneYang2 Normal log is still empty. I don't want to be rude, but could you actually check the content before upload?

YvonneYang2, 2019-10-14:
Aplay 0,0 + arecord 0,4 --> ctrl+z and fg on arecord 0,4 --> I/O error

Sorry, I cannot catch the sof error log and normal log at the same time because of the sof-logger bug. So upload the all logs I can catch.

[all_logger.log](https://github.com/thesofproject/sof/files/3723459/all_logger.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3723460/dmesg.log)

plbossart, 2019-10-14:
Confirmed on my side even without all the Ctrl-z.

<details> <summary> terminal A: </summary>

````
root@jf-icl-rvp-hda-8:~# speaker-test -Dhw:0,0 -c2 -r48000

speaker-test 1.1.3

Playback device is hw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 1 - Front Right
Time per period = 5.592194
 0 - Front Left
 1 - Front Right
Time per period = 5.934904
 0 - Front Left
 1 - Front Right
Time per period = 5.999813
 0 - Front Left
 1 - Front Right
Time per period = 5.866118
 0 - Front Left
 1 - Front Right
Time per period = 5.999929
 0 - Front Left
 1 - Front Right
Write error: -5,Input/output error
xrun_recovery failed: -5,Input/output error
Transfer failed: Input/output error
root@jf-icl-rvp-hda-8:~# 
````
</details>

<details> <summary> terminal B: </summary>

````
root@jf-icl-rvp-hda-8:~# arecord -Dhw:0,4 -c2 -r48000 -fdat
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
RIFF$�WAVEfmt ���data�arecord: pcm_read:2103: read error: Input/output error
root@jf-icl-rvp-hda-8:~# 
````
</summary>

[dmesg_arecord.log](https://github.com/thesofproject/sof/files/3726136/dmesg_arecord.log)

[arecord_logger.txt](https://github.com/thesofproject/sof/files/3726137/arecord_logger.txt)

plbossart, 2019-10-14:
I double-checked that aplay/arecord work separately, it's when either is started while the other is already streaming that we see IPC issues

<details> <summary> console log: <\summary>

````
root@jf-icl-rvp-hda-8:~# dmesg -C
root@jf-icl-rvp-hda-8:~# modprobe snd-sof-pci
root@jf-icl-rvp-hda-8:~# speaker-test -Dhw:0,0 -c2 -r48000

speaker-test 1.1.3

Playback device is hw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 1 - Front Right
Time per period = 5.592017
 0 - Front Left
 1 - Front Right
^CTime per period = 3.673565
root@jf-icl-rvp-hda-8:~# 
root@jf-icl-rvp-hda-8:~# 
root@jf-icl-rvp-hda-8:~# 
root@jf-icl-rvp-hda-8:~# arecord -Dhw:0,4 -r48000 -c2 -f dat 
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
RIFF$�WAVEfmt ���data�^CAborted by signal Interrupt...
root@jf-icl-rvp-hda-8:~# speaker-test -Dhw:0,0 -c2 -r48000

speaker-test 1.1.3

Playback device is hw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 1 - Front Right
Time per period = 5.592182
 0 - Front Left
^C 1 - Front Right
Time per period = 0.079813
root@jf-icl-rvp-hda-8:~# arecord -Dhw:0,4 -r48000 -c2 -f dat 
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
RIFF$�WAVEfmt ���data�arecord: pcm_read:2103: read error: Input/output error
root@jf-icl-rvp-hda-8:~# dmesg > dmesg_arecord.log

````
</details>

[dmesg_arecord.log](https://github.com/thesofproject/sof/files/3726206/dmesg_arecord.log)

[icl-logger.txt](https://github.com/thesofproject/sof/files/3726210/icl-logger.txt)

plbossart, 2019-10-14:
Solved for me by https://github.com/thesofproject/sof/pull/1935

@YvonneYang2 please double-check.

tlauda, 2019-10-15:
@YvonneYang2 Could you check if master + #1935 + #1946 + https://github.com/thesofproject/linux/pull/1277 fixes this and other recent SDW issues?

YvonneYang2, 2019-10-16:
Only test it with master + #1935 +  https://github.com/thesofproject/linux/pull/1277 since has conflicts when merge #1946. Issue cannot reproduce on icl 3-in1 rvp.

YvonneYang2, 2019-10-21:
Test it with latest master (#1935 and #1946 have been merged) and integration/soundwire-latest (29debf8) + PR1277, issue hasn't been all fixed.  
Aplay 0,0 +Arecord 0,1 --> background and foreground on arecord 0,1 --> I/O error

Logs:
[logger.log](https://github.com/thesofproject/sof/files/3749945/logger.log)
[dmesg.log](https://github.com/thesofproject/sof/files/3749946/dmesg.log)

YvonneYang2, 2019-10-23:
Check it with test recipe as below (linux#1277 has been merged) , below scenario can still reproduce the issue.

Aplay 0,0 +Arecord 0,1 --> background and foreground on aplay 0,1 --> I/O error

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 7c85d69
FW: https://github.com/thesofproject/sof/commits/master  commit: 8b059f6
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

logs:
[dmesg.log](https://github.com/thesofproject/sof/files/3761276/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3761277/logger.log)

slawblauciak, 2019-10-23:
When doing fg I'm noticing a bunch of "[ 1408.960476] rt700 sdw:0:25d:700:0:0: Bus clash error detected" errors happening in the kernel. What is that?

plbossart, 2019-10-23:
Y

> Reproduced on ICL RVP with RT700
> 
> When doing fg I'm noticing a bunch of
> "[ 1408.960476] rt700 sdw:0:25d:700:0:0: Bus clash error detected" errors happening in the kernel.
> What is that?
> 
> Are you aware of this, @plbossart ?

Yes, I don't know what those are. Bus clash and parity errors are reported by Realtek devices, this means that either they drive a zero while the master drives a one, or that the parity they see is different from what the master observes.

This could come from a bad electrical connection or connectors that make the signal flaky. or a hardware bug on the Realtek side.

I believe those errors are only transients, and since we use the auto-retransmit on errors it's likely that the commands actually succeed when we retry. I haven't experimented this theory that could be proven by playing with the Cadence IP error-threshold.

bardliao, 2019-10-24:
> Y
>
>
> Yes, I don't know what those are. Bus clash and parity errors are reported by Realtek devices, this means that either they drive a zero while the master drives a one, or that the parity they see [...]

Actually, I saw this when I removed the power supply of the 3-in-1 test card by accident.

slawblauciak, 2019-10-24:
@plbossart I just can't help but wonder whether those could cause some extra latency in data transfers, enough for an underrun to occur.

YvonneYang2, 2019-10-24:
Issue still can reproduce on CML SDW Laptop with ALC711+ ALC1308+ ALC715 .

kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 15f27d8
FW: https://github.com/thesofproject/sof/commits/master  commit: f277189
tplg: Same with FW branch, sof-cml-rt711-rt1308-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

dengyangchao, 2019-10-24:
issue also can reproduce on TGL RVP with codec ACL711, ALC1308 in SDW mode

kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 15f27d8
FW: https://github.com/thesofproject/sof/commits/master commit: f277189
TPLG: Same with TGL FW branch commit:  sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg

[dmesg.log](https://github.com/thesofproject/sof/files/3766493/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3766494/logger.log)
[logger_error.log.txt](https://github.com/thesofproject/sof/files/3766496/logger_error.log.txt)

plbossart, 2019-10-24:
> @plbossart I just can't help but wonder whether those could cause some extra latency in data transfers, enough for an underrun to occur.
> 
> Frankly, given the same exact situation on our side, I doubt it's a problem with electrical connections, unless it's just a coincidence that we're getting the same kind of behavior.

the master does not detect errors, that's the odd part. If the master complained yes I would be very worried.

In general the bus clash and parity result in a command NAK, they can't really impact audio. what will impact audio is if the slave loses sync, and this point we redo the enumeration and there will be an underrun.

slawblauciak, 2019-10-25:
@plbossart me and @tlauda noticed something rather peculiar happening in the capture DMA copies that are happening in FW.

Normally, as the capture stream goes in this particular case, we copy 384 bytes each 1ms.
The very moment we send CTRL+Z to aplay, the playback stream is stopped, and we receive new DAI config on the playback pipeline.
Regardless, this isn't what raises the most questions. That would be the very next copy on the capture pipeline, requesting the copy of a lot of data (e.g. 800, or sometimes over 1100 bytes), even tho only a millisecond has passed. What's happening in the kernel/driver that we suddenly copy so much data after only a millisecond?

plbossart, 2019-10-25:
@slawblauciak the kernel is not even aware of your 1ms transfers, and the DMA pace is completely controlled by the DSP. The host driver is not involved in requesting more or fewer data, it's only notified when a period elapsed (interrupt set with the BDL).

So I am not sure what the description of "requesting the copy of a lot of data" means. There are no requests to the firmware at all.

tlauda, 2019-10-25:
> @slawblauciak the kernel is not even aware of your 1ms transfers, and the DMA pace is completely controlled by the DSP. The host driver is not involved in requesting more or fewer data, it's only [...]

@plbossart We are talking about DW-DMA connected to ALH, so this is not entirely true. DMA pace is controlled by the interface to which it transfers the data. We've confgured DMA's link list to handle 384 bytes per ms. When playback stream is being stopped (after we received DAI config IPC), the already running capture stream is getting crazy fast and ALH suddenly decides to produce 3 times more data then it supposed to. Are you sure that during stopping SDW streams you are not touching some config registers, that could potentially cause this kind of scenario?

plbossart, 2019-10-25:
the kernel does not handle the DW_DMA so I am not sure what you are asking me.

tlauda, 2019-10-25:
@plbossart But handles ALH, so I'm asking what ALH register you are writing during STOP sequence.

plbossart, 2019-10-25:
@tlauda we don't update ALH registers on stop.
see code here https://github.com/thesofproject/linux/blob/1c3de3d2054456132a04c9de4a73ec6dbf491657/drivers/soundwire/intel.c#L545

we configure the values once per PDI. the Stop/start has no impact on those values.

plbossart, 2019-10-25:
Also what I find very surprising is this issue is reported to exist even when the capture and playback are handled on different links, which are managed completely separately by the kernel

plbossart, 2019-10-25:
@YvonneYang2 I can only reproduce the IO error when I capture and playback on the same link to RT711

<details> <summary> playback </summary>

````
root@jf-icl-rvp-hda-8:~# speaker-test -Dhw:0,0 -c2 -r48000

speaker-test 1.1.3

Playback device is hw:0,0
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 96 to 16368
Period size range from 48 to 4080
Using max buffer size 16368
Periods = 4
was set period_size = 3312
was set buffer_size = 16368
 0 - Front Left
 1 - Front Right
Time per period = 5.592206
 0 - Front Left
 1 - Front Right
Time per period = 5.934937
 0 - Front Left
 1 - Front Right
^Z
[1]+  Stopped                 speaker-test -Dhw:0,0 -c2 -r48000
root@jf-icl-rvp-hda-8:~# fg
speaker-test -Dhw:0,0 -c2 -r48000
Write error: -32,Broken pipe
Write error: -5,Input/output error
xrun_recovery failed: -5,Input/output error
Transfer failed: Input/output error
root@jf-icl-rvp-hda-8:~# 
````

</details>

<details> <summary> capture </summary>

````
root@jf-icl-rvp-hda-8:~# arecord -Dhw:0,1 -c2 -r48000 -f dat
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
RIFF$�WAVEfmt ���data�arecord: pcm_read:2103: read error: Input/output error

````
</details>

When I capture from RT715 (hw:0,4) I don't see any issues. can you please clarify the test conditions?

YvonneYang2, 2019-10-28:
Yes, capture from RT715 won't reproduce the issue. Issue can only be reproduced in one scenario as below: 

Test steps:
1. Terminal A:  aplay -Dhw:0,0 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i
2. Terminal B: arecord -Dhw:0,1 -fs16_le -c2 -r 48000 -vv -i tmp.wav
3.  In Terminal A:  switch aplay to background via "ctrl + z" , then switch foreground via "fg"

Impact: 
I/O error occurred after switching aplay to foreground from background. 

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 661d261 + PR1359
FW: https://github.com/thesofproject/sof/commits/master  commit: 78b022b
tplg: https://github.com/bardliao/sof/commits/soundwire-tplg (commit: 0408153), sof-cml-rt711-rt1308-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715 

Logs:
[dmesg.log](https://github.com/thesofproject/sof/files/3777415/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3777416/logger.log)

plbossart, 2019-10-28:
@YvonneYang2 I can only reproduce the issue if the foreground/background change happens for the first stream that gets played. E.g. in your sequence above, if I put the terminal B in background, nothing happens. and if I reverse the sequence, it's the opposite, the arecord is the only one to trigger the problem.

In other words, there is a state established when the first stream starts playing, and stopping it has an impact on the second.

plbossart, 2019-10-29:
@bardliao I pushed some changes to my branch fix/sdw-bus-clash (which also has tentative fixes for other issues like memory leaks and broken state machines)

see attached log 
[dmesg3.log](https://github.com/thesofproject/sof/files/3781421/dmesg3.log)

it seems the rt711 is programmed with the same values for the sink and source ports, or the values don't seem to make sense.
Can you try and add all dpN registers to the function sdw_dump_slave_dpN() while I get some sleep, it'd help us confirm that indeed the two streams operate with the same parameters, which is a clear bug. Thanks!

bardliao, 2019-10-29:
> @bardliao I pushed some changes to my branch fix/sdw-bus-clash (which also has tentative fixes for other issues like memory leaks and broken state machines)

@plbossart Sorry, I can't find the fix/sdw-bus-clash branch. Did you push it?

> see attached log
> [dmesg3.log](https://github.com/thesofproject/sof/files/3781421/dmesg3.log)
>
> it seems the rt711 is programmed with the same values for the sink and source ports, or the values don't [...]

bardliao, 2019-10-29:
@plbossart I wrote a sdw_dump_slave_dpN() function by myself and add it in the end of rt711_pcm_hw_params(). The difference of dpN registers are 0x2, 0x24 and 0x34 which are DPn_PortCtrl and DPn_OffsetCtrl1.
Please see the following for detail.

<details> <summary> dp3 dump </summary>

```
300 00000000
301 00000000
302 00000020
303 0000001f
304 00000000
305 00000003
320 00000000
321 ffffffc3
322 000000c7
323 00000000
324 00000001
325 00000000
326 00000013
327 00000000
328 ffffffc3
330 00000000
331 ffffffc3
332 000000c7
333 00000000
334 00000001
335 00000000
336 00000013
337 00000000
338 ffffffc3
```
</details>

<details> <summary> dp4 dump </summary>

```
400 00000000
401 00000000
402 00000000
403 0000001f
404 00000000
405 00000003
420 00000000
421 ffffffc3
422 000000c7
423 00000000
424 00000041
425 00000000
426 00000013
427 00000000
428 ffffffc3
430 00000000
431 ffffffc3
432 000000c7
433 00000000
434 00000041
435 00000000
436 00000013
437 00000000
438 ffffffc3
```
</details>

<details> <summary> sdw_dump_slave_dpN  </summary>

```
diff --git a/sound/soc/codecs/rt711.c b/sound/soc/codecs/rt711.c
index 96f9fe21d9a4..e02200370399 100644
--- a/sound/soc/codecs/rt711.c
+++ b/sound/soc/codecs/rt711.c
@@ -918,6 +918,28 @@ static void rt711_shutdown(struct snd_pcm_substream *substream,
 	kfree(stream);
 }
 
+static void sdw_dump_slave_dpN(struct sdw_slave *slave, int port)
+{
+	int addr, val, i;
+
+	pr_err("bard: read port %d start\n", port);
+	for (i = 0; i <= 0x5; i++) {
+		addr = (port << 8) | i;
+		val = sdw_read(slave, addr);
+		pr_err("%02x %08x\n", addr, val);
+	}
+	for (i = 0x20; i <= 0x28; i++) {
+		addr = (port << 8) | i;
+		val = sdw_read(slave, addr);
+		pr_err("%02x %08x\n", addr, val);
+	}
+	for (i = 0x30; i <= 0x38; i++) {
+		addr = (port << 8) | i;
+		val = sdw_read(slave, addr);
+		pr_err("%02x %08x\n", addr, val);
+	}
+}
+
 static int rt711_pcm_hw_params(struct snd_pcm_substream *substream,
 				struct snd_pcm_hw_params *params,
 				struct snd_soc_dai *dai)
@@ -1004,6 +1026,7 @@ static int rt711_pcm_hw_params(struct snd_pcm_substream *substream,
 	regmap_write(rt711->regmap, RT711_ADC1_FORMAT_H, val);
 	regmap_write(rt711->regmap, RT711_ADC2_FORMAT_H, val);
 
+	sdw_dump_slave_dpN(rt711->slave, port);
 	return retval;
 }
```
</details>

plbossart, 2019-10-29:
@bardliao branch pushed now, not sure what happened

plbossart, 2019-10-29:
Not having any luck on this one. The registers seem correct on both master and slave ports, so not sure what could cause the bus clashes. These bus clashes are definitively not an electrical issue but a bad programming, but I just don't see it yet. 

The only think I see as suspicious is this trace, as if the port 2 (playback) and port 3(capture) did not use the same bank.

````
[   50.677703] intel-sdw sdw-master-0: plb: cdns_port_enable: port 3 bank 1 action 3
[   50.678304] intel-sdw sdw-master-0: plb: cdns_port_enable: port 2 bank 0 action 0
````

[dmesg4.log](https://github.com/thesofproject/sof/files/3785790/dmesg4.log)

plbossart, 2019-10-29:
@tlauda @ranj063 this might be a red-herring but I wonder if we have the start sequence wrong for SoundWire: the traces show that we first do a bank switch, which starts the transfers on the bus, then we send a TRIG_START IPC to the firmware, which will likely start the DMAs. You'd really want to have all the DMAs set-up and pipelines preloaded, then the bank switch happens.

ranj063, 2019-10-29:
@plbossart Does the "bank switch" happen in the START callback for the SDW BE DAI?

I recently switch the trigger order to SND_SOC_DPCM_TRIGGER_POST. So with this change, the BE DAI start trigger gets called first followed by the FW DAI trigger (which sets up the DMA). Maybe this should be different for SDW?

plbossart, 2019-10-29:
@ranj063 I tried but it's likely a separate issue - no improvement. We will have to revisit this for multi-link playback, the bank switch is required to guarantee we play samples in phase.

plbossart, 2019-10-29:
@tlauda when the playback bombs out, I see a large burst of traces, which repeat the same thing

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      2       BUFFER          48567712.395833  48567712.000000   src/audio/buffer.c:129 	comp_update_buffer_produce(), no bytes to produce, source->comp.id = 1, source->comp.type = 5, sink->comp.id = 4, sink->comp.type = 2
    0      2       BUFFER          48567716.718750         4.322917   src/audio/buffer.c:184 	comp_update_buffer_consume(), no bytes to consume, source->comp.id = 0, source->comp.type = 1, sink->comp.id = 1, sink->comp.type = 5
    0      2          DMA          48567722.447917         5.729167 src/drivers/dw/dma.c:1050	dw_dma_free_data_size() size is 0!
````

No idea what this means, is there a way to disable this flood and print this error only once so that we can look at better traces?

plbossart, 2019-10-29:
@tlauda here are some traces with the errors above commented out to see something

[t4.log](https://github.com/thesofproject/sof/files/3786189/t4.log)
[dmesg6.log](https://github.com/thesofproject/sof/files/3786191/dmesg6.log)

plbossart, 2019-10-29:
Better trace with the DAI ID in alh_trigger
It looks like we are not able to recover from the underflow on playback, and end-up stopping capture as well. What's odd is that on the kernel side we do send a trig_start for playback after the prepare but it doesnt' show in the fimware trace.

[t5.log](https://github.com/thesofproject/sof/files/3786278/t5.log)

plbossart, 2019-10-29:
diffs below if this helps reproduce this case:

````diff
diff --git a/src/audio/buffer.c b/src/audio/buffer.c
index 51525496..1d8c6d67 100644
--- a/src/audio/buffer.c
+++ b/src/audio/buffer.c
@@ -121,12 +121,14 @@ void comp_update_buffer_produce(struct comp_buffer *buffer, uint32_t bytes)
 
 	/* return if no bytes */
 	if (!bytes) {
+#if 0
 		trace_buffer("comp_update_buffer_produce(), "
 			     "no bytes to produce, source->comp.id = %u, "
 			     "source->comp.type = %u, sink->comp.id = %u, "
 			     "sink->comp.type = %u", buffer->source->comp.id,
 			     buffer->source->comp.type, buffer->sink->comp.id,
 			     buffer->sink->comp.type);
+#endif
 		return;
 	}
 
@@ -176,12 +178,14 @@ void comp_update_buffer_consume(struct comp_buffer *buffer, uint32_t bytes)
 
 	/* return if no bytes */
 	if (!bytes) {
+#if 0
 		trace_buffer("comp_update_buffer_consume(), "
 			     "no bytes to consume, source->comp.id = %u, "
 			     "source->comp.type = %u, sink->comp.id = %u, "
 			     "sink->comp.type = %u", buffer->source->comp.id,
 			     buffer->source->comp.type, buffer->sink->comp.id,
 			     buffer->sink->comp.type);
+#endif
 		return;
 	}
 
diff --git a/src/audio/host.c b/src/audio/host.c
index d8551ed6..a53d755e 100644
--- a/src/audio/host.c
+++ b/src/audio/host.c
@@ -760,7 +760,7 @@ static int host_copy(struct comp_dev *dev)
 
 	copy_bytes = host_buffer_get_copy_bytes(dev);
 	if (!copy_bytes) {
-		trace_host_with_ids(dev, "host_copy(): no bytes to copy");
+		//trace_host_with_ids(dev, "host_copy(): no bytes to copy");
 		return ret;
 	}
 
diff --git a/src/drivers/dw/dma.c b/src/drivers/dw/dma.c
index 8d52e762..4d72832e 100644
--- a/src/drivers/dw/dma.c
+++ b/src/drivers/dw/dma.c
@@ -1046,8 +1046,10 @@ static int dw_dma_free_data_size(struct dma_chan_data *channel)
 	if (size < 0)
 		size += dw_chan->ptr_data.buffer_bytes;
 
+#if 0
 	if (!size)
 		trace_dwdma("dw_dma_free_data_size() size is 0!");
+#endif
 
 	return size;
 }
diff --git a/src/drivers/intel/cavs/alh.c b/src/drivers/intel/cavs/alh.c
index ea127ccc..ec7cadbd 100644
--- a/src/drivers/intel/cavs/alh.c
+++ b/src/drivers/intel/cavs/alh.c
@@ -22,15 +22,15 @@
 
 static int alh_trigger(struct dai *dai, int cmd, int direction)
 {
-	trace_alh("alh_trigger() cmd %d", cmd);
+	trace_alh("alh_trigger() dai %d cmd %d", dai->index, cmd);
 
 	return 0;
 }
 
 static int alh_set_config(struct dai *dai, struct sof_ipc_dai_config *config)
 {
-	trace_alh("alh_set_config() config->format = 0x%4x",
-		  config->format);
+	trace_alh("alh_set_config() dai %d config->format = 0x%4x",
+		  dai->index, config->format);
 
 	return 0;
 }
````

tlauda, 2019-10-30:
@plbossart This is exactly what @slawblauciak told you here https://github.com/thesofproject/sof/issues/1925#issuecomment-546317781. The only mistake we've made was the assumption, that the capture overrun happens on playback stop, but in fact it happens on start, or at least on something you're doing on host when starting the playback stream again.
To sum up what is happening here:
1. Start playback.
2. Start capture.
3. Make playback go to background.
4. Make playback go to foreground:
    a. Kernel doing some things to prepare and start stream. <-- **Our capture DW-DMA suddenly gets full buffer of data (3 ms) instead of 1 ms and suddenly stops, since it isn't configured to handle such a big period.**
    b. Kernel sends START ipc to FW. <-- We are not getting this.

plbossart, 2019-10-30:
@tlauda regardless of what happens in 4.a, any idea as to why the 4.b START_IPC is not received. We have logs on the kernel side that do show it's being sent, so not sure how to debug further.

tlauda, 2019-10-30:
@plbossart I've looked into FW logs one more time and I was mistaken. We do receive params ipc and START ipc for playback stream after reporting overrun on capture. However both logs I have end up after that, so I don't know what happens next.

plbossart, 2019-10-30:
@tlauda nothing happens after the underflow, both streams are stopped and that's game over. You can see the sequence in the dmesg6.log, it's standard trigger-stop, hw_free, close for both playback and capture. I don't know why we don't see more traces from firmware, maybe we have a buffer that's not full. is there a debugfs way to requesting the trace buffer be flushed?

tlauda, 2019-10-31:
@plbossart If nothing interesting happens after underrun, then let's focus on rootcausing the actual problem.

plbossart, 2019-10-31:
@tlauda can you think of additional traces that might help? 

also what I find difficult is that the dmesg and sof-logger timestamps are completely different, didn't we have an IPC to align the timestamps and make sure the time deltas are the same on both sides

slawblauciak, 2019-11-06:
@plbossart @bardliao are there any findings that might point what's actually wrong here?

plbossart, 2019-11-06:
@slawblauciak I haven't had time to drill down further. I still think it's a kernel bug with the wrong data port settings applied, that's the only rational explanation for the bus clash
I plan to try with fixed settings, e.g. initialize all ports on startup and just play with enable/disable to confirm my theory.
It's a difficult one to debug indeed.

plbossart, 2019-11-07:
@slawblauciak @tlauda @bardliao I have traced and double checked all ports, alternate banks and can't find a bad configuration or smoking gun.

At this point my hypotheses as to what the root cause might be:

a) a misunderstanding on what happens in the hw_free case. the configuration I end-up sending is this 
````c
	memset(&config, 0, size);
	config.hdr.size = size;
	config.hdr.cmd = SOF_IPC_GLB_DAI_MSG | SOF_IPC_DAI_CONFIG;
	config.type = SOF_DAI_INTEL_ALH;
	config.dai_index = (link_id << 8) | d->id;
	config.alh.stream_id = 0xFFFF; /* invalid value on purpose */
````
When I look at the firmware code, it's unclear to me if this value is used for anything? alh_get_handshake() and alh_get_fifo don't really perform any reset and for the get_fifo case there's not even a range check. Likewise in dai.c it's not clear to me that the value 0xFFFF is actually used 

````c
		/* As with HDA, the DMA channel is assigned in runtime,
		 * not during topology parsing.
		 */
		channel = config->alh.stream_id;
		dd->stream_id = config->alh.stream_id;
````

b) Currently we release this ALH stream only when there is a hw_free. But in the case of the underflow, the hw_free is not called, so the ALH configuration is not reset. This could maybe be an issue for your DMA management?

c) a problem with ALH/SHIM registers. Currently the host sets up the values on resume, and we never modify the settings. Maybe we have to disable the ALH mappings on TRIGGER_STOP? I didn't see such a requirement but that could very well be a miss in the programming sequence.

Comments welcome.

plbossart, 2019-11-07:
Latest trace attached for reference. No trace of problems with port configurations.

[dmesg7.log](https://github.com/thesofproject/sof/files/3817416/dmesg7.log)

plbossart, 2019-11-08:
I got it. This was option c).

the following change fixes this specific issue

````c
#if 0
	intel_pdi_shim_configure(sdw, sdw->pdi);
	intel_pdi_alh_configure(sdw, sdw->pdi);
	sdw_cdns_config_stream(cdns, ch, dir, sdw->pdi);

	/* Inform DSP about PDI stream number */
	ret = intel_params_stream(sdw, substream, dai, sdw->hw_params,
				  sdw->instance,
				  sdw->pdi->intel_alh_id);
	if (ret)
		goto err;
#endif
````

This was added to make sure that on resume, where all context was lost, we reconfigure ALH/SHIM/firmware.

But when prepare is called because of an underflow, with a running stream, then this violates one of the explicit requirements in the cAVS Programming reference.

eg for STRMxCFG.CHAN and STRMxCFG.DMAT: The value should be programmed before enabling the Audio Link Hub stream transfer by setting PTEN or GPDMA / DSP Link DMA enable.

So what we have to do is revisit my brilliant simplification that we didn't need a state variable to figure out if we are resuming or just handing with underflows. Gah,

plbossart, 2019-11-09:
@bardliao can you try the branch fix/sdw_dai_prepare on my GitHub, it's a POC on how we could keep track of the suspend information. there is no .suspend for the dai ops, but there is one for the dai_driver, so that might work for us.

I couldn't make it work on the laptop, looks like deep sleep is not supported, and I have no time left to test on RVP, hopefully you can take a look early next week. Thanks!