thesofproject/sof #1926: [BUG][SDW]Switch source in sound setting, sound output stopped but paplay still running.

YvonneYang2, 2019-10-12:
**Describe the bug**
Switch source in sound setting, sound output stopped but paplay still running. 

**To Reproduce**
1.paplay via command "paplay test.wav -vvv "
2.open sound settings and switch output source between speaker and headphone.

**frequency**
100%

**Expected result**
Palay work normally.

**Actual result**
Paplay has sound output before switch source, then sound stopped but paplay still running when switch source.

**Environment**
Firmware: c7d0149 https://github.com/thesofproject/sof/commits/master 
Kernel: e7fb5a5 https://github.com/thesofproject/linux/tree/integration/soundwire-latest
tplg:  sof-icl-rt711-rt1308-rt715-hdmi.tplg rename sof-icl-rt711-rt1308-rt715.tplg  path same as FW branch
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

<details>
 <summary>error logger</summary>

````
	CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 3.16    299449417.239583 299449408.000000      src/audio/dai.c:588 	dai_check_for_xrun() error: underrun due to no data available
    0      1         COMP         299449427.031250         9.791667 of/audio/component.h:743 	comp_underrun() error: dev->comp.id = 16, source->avail = 768, copy_bytes = 0
    0      1         PIPE         299449518.489583        91.458336 src/audio/pipeline.c:833 	pipeline_copy() error: ret = -61
    0      1         PIPE 3.17    299449528.541667        10.052083 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1          DAI 2.10    324766764.062500  25317236.000000      src/audio/dai.c:593 	dai_check_for_xrun() error: overrun due to no data available
    0      1         COMP         324766773.802083         9.739583 of/audio/component.h:759 	comp_overrun() error: dev->comp.id = 10, sink->free = 768, copy_bytes = 0
    0      1         PIPE         324766900.260417       126.458336 src/audio/pipeline.c:854 	pipeline_copy() error: ret = -61, start->comp.id = 10, dir = 0
    0      1         PIPE 2.11    324766912.135417        11.875000 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP         360414329.687500  35647416.000000 rc/audio/component.c:192 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET

````
</details>

[switch_source_insoundsetting_no_sound.log](https://github.com/thesofproject/linux/files/3720526/switch_source_insoundsetting_no_sound.log)
[switch_source_insoundsetting_no_sound_dmesg.log](https://github.com/thesofproject/linux/files/3720527/switch_source_insoundsetting_no_sound_dmesg.log)
[switch_source_insoundsetting_no_sound_error.log](https://github.com/thesofproject/linux/files/3720528/switch_source_insoundsetting_no_sound_error.log)

YvonneYang2, 2019-10-14:
Issue is caused by commit 71d9ee4(master branch)

@tlauda Could you have a look? Thanks.

plbossart, 2019-10-14:
@YvonneYang2 can you clarify on which outputs you are playing for headset and speakers?
Traditionally we use the same output at the firmware level, but with this topology we are talking about different PCM and DAIs for headset and speakers.

Also, before you start playing with switching, can you clarify if independently you are able to play on headset and speakers, one at a time?

YvonneYang2, 2019-10-16:
Paplay from headset is used for RT711.(aplay 0,0)
Paplay from speakerd is used for RT1308 (aplay0,2)  

Yes,  paplay via speaker or headset can work independently. Issue only occurred when paplay then switch source between speaker and headset in sound setting.

juimonen, 2019-10-16:
So probably pulseaudio "legacy" alsa paths don't work with this setup. We need to switch to ucm for this platform and see if that can work better...

YvonneYang2, 2019-10-16:
Only test it with master + #1935 + thesofproject/linux#1277 since has conflicts when merge #1946. Issue cannot reproduce on icl 3-in1 rvp.

YvonneYang2, 2019-10-17:
Test it with latest integration/soundwire-latest(commit :0ebe10b) and master branch(commit: 0d54df1d),   #1395 and #1946 have merged while thesofproject/linux#1277 has not yet. But issue not reproduced. Close the bug.

RanderWang, 2019-10-22:
I can reproduce it.

RanderWang, 2019-10-22:
I will debug it on my CML

YvonneYang2, 2019-10-23:
Issue reproduced with test recipe as below. 

Steps:
1. Paplay via headset/HDMI/DP 
2. Check amixer cget numid=37 and 38: value is on.
3. Wait for 1 minute, switch source to speaker in sound setting.

After step3, speaker cannot output sound while paplay shows still running. Check the amixer cget  cget numid=37 and 38 again , the value turned to off.

Logs:
[dmesg.log](https://github.com/thesofproject/sof/files/3761699/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3761700/logger.log)

YvonneYang2, 2019-10-24:
Issue cannot reproduce on CML laptop. 

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 15f27d8 + PR1359
FW: https://github.com/thesofproject/sof/commits/master  commit: f277189
tplg: Same with FW branch, sof-cml-rt711-rt1308-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

RanderWang, 2019-10-25:
so, it is fixed. Could you help to close it ?  thanks!

YvonneYang2, 2019-10-25:
Issue can still be reproduced with test recipe as below(PR1359 not merged), so reopen the issue. 

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 1c3de3d 
FW: https://github.com/thesofproject/sof/commits/master commit: 89ef0af 
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode  

Will close the issue once PR1359 is merged into Soundwire-latest branch.