thesofproject/sof #1927: [BUG][SDW] Parecord can't capture audio normally if change input volume from sound setting

ClarexZhou, 2019-10-12:
**Describe the bug**
Parecord can't capture audio normally if change input volume from sound setting

**To Reproduce**
1. Capture audio via "parecord 1.wav -vvv"
2. Open sound setting, select input source
3. Change input source volume 

**Reproduction Rate**
100%

**Expected behavior**
Parecord  volume should be changed and should capture audio successfully.

**Impact**
Parecord can't capture audio normally if change input volume from sound setting

**Environment**
Firmware: c7d0149 https://github.com/thesofproject/sof/commits/master 
Kernel: e7fb5a5 https://github.com/thesofproject/linux/tree/integration/soundwire-latest
tplg:  sof-icl-rt711-rt1308-rt715-hdmi.tplg rename sof-icl-rt711-rt1308-rt715.tplg  path same as FW branch
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode 

Error logger:
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 1.4      87147399.270833  87147400.000000      src/audio/dai.c:588 	dai_check_for_xrun() error: underrun due to no data available
    0      1         COMP          87147409.166667         9.895833 of/audio/component.h:743 	comp_underrun() error: dev->comp.id = 4, source->avail = 768, copy_bytes = 0
    0      1         PIPE          87147500.625000        91.458336 src/audio/pipeline.c:833 	pipeline_copy() error: ret = -61
    0      1         PIPE 1.5      87147510.572917         9.947917 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!
```
[parecord.txt](https://github.com/thesofproject/sof/files/3720541/parecord.txt)
[parecord_logger.txt](https://github.com/thesofproject/sof/files/3720543/parecord_logger.txt)

plbossart, 2019-10-12:
@ClarexZhou please update your bug description with a better wording than "can't capture audio normally". Does it stop playing, if the volume not right, is the audio distorted? The description is not clear enough as is, we can't figure out what 'normal' means

ClarexZhou, 2019-10-14:
> @ClarexZhou please update your bug description with a better wording than "can't capture audio normally". Does it stop playing, if the volume not right, is the audio distorted? The description is not clear enough as is, we can't figure out what 'normal' means

Audio isn't captured after volume is changed. Always repeat the latest word before changing volume.
eg.  Speaker "test one two three"-> then change volume->speak others-> always "three" is repeated to play after volume changed.
Attach the recorded audio file.
[1.zip](https://github.com/thesofproject/sof/files/3722695/1.zip)

YvonneYang2, 2019-10-14:
Issue is caused by commit 71d9ee4(master branch)

@tlauda Could you have a look? Thanks.

YvonneYang2, 2019-10-16:
Only test it with master + #1935 + thesofproject/linux#1277 since has conflicts when merge #1946. Issue cannot reproduce on icl 3-in1 rvp.

YvonneYang2, 2019-10-30:
Issue cannot reproduce with test recipe as below, so close the bug.

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 9f1bd5b
FW: https://github.com/thesofproject/sof/commits/master  commit: 0e8cf66e
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

closed 2019-10-30 by YvonneYang2