thesofproject/sof #1928: [BUG][SDW] IPC timed out for 0x60050000: GLB_STREAM_MSG: TRIG_STOP occurred during multipipline stress test.

YvonneYang2, 2019-10-14:
**Describe the bug**
IPC timed out for 0x60050000: GLB_STREAM_MSG: TRIG_STOP occurred during multipipline stress test.

**To Reproduce**
Run command "./multiplex_pipeline-test.sh pcm0p pcm4p pcm1c pcm2c pcm3c"

**Reproduction Rate**
1/54

**Expected behavior**
Multipipline works normally.

**Impact**
Dmesg shows 0x60050000: GLB_STREAM_MSG: TRIG_STOP.

**Environment**
Firmware: e755494 Master
Kernel: d2b55e0 on integration/soundwire-latest
Topology: file: same with firmware branch, sof-cml-rt700.m4
platform: CML RVP with onboard ALC700 in SDW mode
Other: Runtime PM status is enabled (without debug=1 options)

<details>
 <summary>Dmesg</summary>

````
[ 1791.900263] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.900266] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=1
[ 1791.902220] rt700 sdw:1:25d:700:0:0: rt700_clock_config complete, clk_freq=6000000
[ 1791.904254] rt700 sdw:1:25d:700:0:0: rt700_clock_config complete, clk_freq=6000000
[ 1791.906705] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 0 cmd 1
[ 1791.906729] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x14001e successful
[ 1791.906730] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 2 dir 1 cmd 1
[ 1791.906731] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.906754] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 1
[ 1791.906761] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2034001e successful
[ 1791.906950] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.906953] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.907148] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.907312] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2044001e successful
[ 1791.907314] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.907553] rt700 sdw:1:25d:700:0:0: rt700_clock_config complete, clk_freq=6000000
[ 1791.907577] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
[ 1791.909164] rt700 sdw:1:25d:700:0:0: rt700_clock_config complete, clk_freq=6000000
[ 1791.912206] rt700 sdw:1:25d:700:0:0: rt700_clock_config complete, clk_freq=6000000
[ 1791.914182] rt700 sdw:1:25d:700:0:0: rt700_clock_config complete, clk_freq=6000000
[ 1792.411109] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 2 dir 1 cmd 0
[ 1792.411116] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 1792.914890] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[ 1792.914893] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[ 1792.914899] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 1792.914970] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 1792.914976] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc000018f rirb 00
[ 1792.914977] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 1792.914982] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 1792.914983] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 1792.915069] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915070] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915070] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915071] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915072] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915073] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915073] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915074] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915075] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915076] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915076] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 1792.915078] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20340000 successful
[ 1792.915084]  DMIC01: ASoC: trigger FE failed -110
[ 1792.915100] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 1
[... 135 lines cut ...]
    0      1         DMIC         2245794844.218750        12.083333 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794855.677083        11.458333 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794867.760417        12.083333 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794879.218750        11.458333 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794891.354167        12.135417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794902.864583        11.510417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794914.947917        12.083333 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794926.406250        11.458333 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794938.541667        12.135417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794950.052083        11.510417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794962.187500        12.135417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794973.697917        11.510417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794985.833333        12.135417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245794997.343750        11.510417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245795009.479167        12.135417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245795020.989583        11.510417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245795033.125000        12.135417 rs/intel/cavs/dmic.c:1503	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         2245795044.635417

````
</details>

[multipipeline_test_54_logger.txt](https://github.com/thesofproject/sof/files/3723173/multipipeline_test_54_logger.txt)
[multipipeline_test_54_logger_error.txt](https://github.com/thesofproject/sof/files/3723174/multipipeline_test_54_logger_error.txt)
[multipipeline_test_54_dmesg.txt](https://github.com/thesofproject/sof/files/3723175/multipipeline_test_54_dmesg.txt)

ClarexZhou, 2019-10-14:
Error was 100% reproduced previously, seen #1839, might be improved by PR #1887

slawblauciak, 2019-10-15:
@singalsu hey, mind taking a look at this? In the error log of FW I can see a whole lot of traces indicating, that the DMIC FIFO is full. Any hints what this might point at?

tlauda, 2019-10-17:
@ClarexZhou @YvonneYang2 Can it still be reproduced with master + linux#1277?

YvonneYang2, 2019-10-18:
ipc timed out for 0x80010000 size 120 occurred at multipipline stress test 70 times with latest kernel(commit:  d88ee7d) + linux PR #1227 and master(commit: 81cc31b). 

Run with command:  sudo ./multiplex_pipeline-test.sh pcm0p pcm2p pcm5c pcm1c pcm4c"
dmesg:
```
[  778.678834] sof-audio-pci 0000:00:1f.3: In hda_link_pcm_trigger cmd=0
[  778.678839] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  778.683355] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 1 dir 1 cmd 0
[  778.683448] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 4 dir 1 cmd 0
[  779.185782] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  779.185791] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  779.185799] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  779.186136] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  779.186154] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000006 intlctl 0xc0000007 rirb 00
[  779.186157] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[  779.186174] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  779.186177] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  779.186383] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186389] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186392] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186395] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186397] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186400] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186402] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186428] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for iDisp1
[  779.186436]  iDisp1: ASoC: trigger BE failed -110
[  779.186441]  HDMI1: ASoC: trigger FE cmd: 0 failed: -110
[  779.186461] sof-audio-pci 0000:00:1f.3: pcm: free stream 5 dir 0
[  779.186491] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  779.186548] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  779.186730] sof-audio-pci 0000:00:1f.3: error: trace IO error
```
[multipipeline_test_70_error_logger.log](https://github.com/thesofproject/sof/files/3743168/multipipeline_test_70_error_logger.log)
[multipipeline_test_70_logger_error.log](https://github.com/thesofproject/sof/files/3743169/multipipeline_test_70_logger_error.log)
[multipipeline_test_70_dmesg.log](https://github.com/thesofproject/sof/files/3743170/multipipeline_test_70_dmesg.log)

slawblauciak, 2019-10-21:
I tried to reproduce it on ICL RVP (FW master 0234df5, linux 29debf8 + PR #1277).
I get completely different results however:

[dmesg.log](https://github.com/thesofproject/sof/files/3751204/dmesg.log)
[test.log](https://github.com/thesofproject/sof/files/3751205/test.log)

I will try again using the same exact recipe, and then later on CML RVP when I get one.

slawblauciak, 2019-10-22:
I can't seem to be able to find the right kernel commit with the d88ee7d hash, can you please provide the full hash?

YvonneYang2, 2019-10-22:
@slawblauciak  
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest 
commit: d88ee7dc5c42c2f3a645fe6e23d9d754e9ea5842

slawblauciak, 2019-10-22:
Hmm, I can't fetch that commit, it might no longer exist on the remote branch. Somebody might've force-pushed or something.

sinahuang, 2019-10-24:
issue still reproduce when run " ./multiplex_pipeline-test.sh pcm0p pcm2p pcm6p pcm4c"
failed at 267th time. 
[multipipeline_test_267_logger_error.log](https://github.com/thesofproject/sof/files/3765929/multipipeline_test_267_logger_error.log)
[multipipeline_test_267_all_logger.log](https://github.com/thesofproject/sof/files/3765930/multipipeline_test_267_all_logger.log)
[multipipeline_test_267_dmesg.log](https://github.com/thesofproject/sof/files/3765931/multipipeline_test_267_dmesg.log)

besides during test, dmesg start show error info "error: iteration 0 of Core En/ROM load failed: -110"   at 113th time.

[multipipeline_test_113_dmesg.log](https://github.com/thesofproject/sof/files/3765933/multipipeline_test_113_dmesg.log)
[multipipeline_test_113_logger_error.log](https://github.com/thesofproject/sof/files/3765934/multipipeline_test_113_logger_error.log)

kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 15f27d8
FW: https://github.com/thesofproject/sof/commits/master  commit: f277189
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

ClarexZhou, 2019-10-24:
@slawblauciak  Previous reproduced kernel already contains PR#1277. kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 15f27d8

YvonneYang2, 2019-10-30:
IO tranfer timed out occurred then showed ipc timed out for 0x80010000 size 120 when ran multipipline stress test. Failed at 117th time.

Run with command: sudo ./multiplex_pipeline-test.sh pcm0p pcm2p pcm4c"

<details>
 <summary>IO transfer timed out </summary>

````
[ 5105.486953] rt711 sdw:0:25d:711:0: [rt711_sdw_write] 7321 83a1 <= b000
[ 5105.486964]  SDW0-Playback: ASoC: BE SDW0-Playback event 1 dir 0
[ 5105.486972]  SDW3-Capture: ASoC: hw_params BE SDW3-Capture
[ 5105.486986]  SDW0-Playback: ASoC: trigger BE SDW0-Playback cmd 1
[ 5105.486987] intel-sdw sdw-master-0: intel_trigger: SDW0 Pin2: start
[ 5105.486988] sdw_enable_stream: SDW0 Pin2-Playback: start
[ 5107.507036] intel-sdw sdw-master-0: IO transfer timed out
[ 5107.507080] intel-sdw sdw-master-3: IO transfer timed out
[ 5109.555045] intel-sdw sdw-master-3: IO transfer timed out
[ 5109.555049] intel-sdw sdw-master-0: IO transfer timed out
[ 5109.555069] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 7520 85a0 7420 84a0 <= 0039 a000
[ 5111.603034] intel-sdw sdw-master-0: IO transfer timed out
[ 5111.603059] intel-sdw sdw-master-3: IO transfer timed out
[ 5113.651043] intel-sdw sdw-master-3: IO transfer timed out
[ 5113.651066] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 7207 8287 <= 0031
[ 5113.651084] intel-sdw sdw-master-0: IO transfer timed out
[ 5115.699010] intel-sdw sdw-master-0: IO transfer timed out
[ 5115.699011] intel-sdw sdw-master-3: IO transfer timed out
[ 5117.747000] intel-sdw sdw-master-3: IO transfer timed out
[ 5117.747001] intel-sdw sdw-master-0: IO transfer timed out
[ 5117.747008] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 7208 8288 <= 0031
[ 5119.795063] intel-sdw sdw-master-3: IO transfer timed out
[ 5119.795067] intel-sdw sdw-master-0: IO transfer timed out
[ 5121.843007] intel-sdw sdw-master-3: IO transfer timed out
[ 5121.843008] intel-sdw sdw-master-0: IO transfer timed out
[ 5121.843016] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 7209 8289 <= 0031
[ 5123.891013] intel-sdw sdw-master-0: IO transfer timed out
[ 5123.891014] intel-sdw sdw-master-3: IO transfer timed out
[ 5123.891024] rt711 sdw:0:25d:711:0: [rt711_sdw_write] 00e0 <= 0001
[ 5125.939065] intel-sdw sdw-master-3: IO transfer timed out
[ 5125.939070] intel-sdw sdw-master-0: IO transfer timed out
[ 5125.939087] rt711 sdw:0:25d:711:0: [rt711_sdw_write] 00f0 <= 0001
[ 5125.939095] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 7227 82a7 <= 0031
[ 5125.939100] rt711 sdw:0:25d:711:0: rt711_clock_config complete, clk_freq=6000000
[ 5125.939112] rt715 sdw:3:25d:715:0: Update DAI routes for rt715-aif2 capture
[ 5125.939121] rt715 sdw:3:25d:715:0: Connecting DAI route rt715 DP4TX -> rt715 DP4 Capture
[ 5125.939133] intel-sdw sdw-master-3: intel_hw_params: SDW3 Pin2: start

````
</details>

<details>
 <summary>IPC timed out</summary>

````
[ 5126.448082] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5126.448152] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5126.448155] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5126.448350] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5126.448368] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5126.955043] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[ 5126.955054] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[ 5126.955066] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 5126.955325] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 5126.955337] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000000 intlctl 0xc0000001 rirb 00
[ 5126.955342] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[ 5126.955353] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[ 5126.955358] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 5126.955366] sof-audio-pci 0000:00:1f.3: error: failed to set DAI hw_params for link 1 dai->id 2 ALH 23
[ 5126.955375] intel-sdw sdw-master-1: ASoC: cpu DAI prepare error: -110
[ 5126.955381]  SDW1-Playback: ASoC: backend prepare failed -110
[ 5126.955489]  DMIC: ASoC: found 2 audio capture paths
[ 5126.955498] sdw_rt711_rt1308_rt715 sdw_rt711_rt1308_rt715: ASoC: find BE for widget ALH0x302.IN
[ 5126.955503] sdw_rt711_rt1308_rt715 sdw_rt711_rt1308_rt715: ASoC: try BE (not set)
[ 5126.955507] sdw_rt711_rt1308_rt715 sdw_rt711_rt1308_rt715: ASoC: try BE ALH3.IN
[ 5126.955510] sdw_rt711_rt1308_rt715 sdw_rt711_rt1308_rt715: ASoC: try BE (not set)
[ 5126.955513] sdw_rt711_rt1308_rt715 sdw_rt711_rt1308_rt715: ASoC: try BE ALH0x302.IN
[ 5126.955520]  DMIC: connected new DPCM capture path DMIC <- SDW3-Capture

````
</details>

kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 9f1bd5b + PR1359
FW: https://github.com/thesofproject/sof/commits/master  commit: 2ca6a4a
tplg: https://github.com/bardliao/sof/commits/soundwire-tplg (commit: 0408153), sof-cml-rt711-rt1308-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715 

Logs:
[multipipeline_test_117_error_logger.log](https://github.com/thesofproject/sof/files/3788400/multipipeline_test_117_error_logger.log)
[multipipeline_test_117_logger_error.log](https://github.com/thesofproject/sof/files/3788401/multipipeline_test_117_logger_error.log)
[multipipeline_test_117_dmesg.log](https://github.com/thesofproject/sof/files/3788402/multipipeline_test_117_dmesg.log)

Script:
[multiplex_pipeline-test.sh.txt](https://github.com/thesofproject/sof/files/3788408/multiplex_pipeline-test.sh.txt)

tlauda, 2019-10-30:
@plbossart @ranj063 I think you still have some kind of problem with receiving an IPC from DSP. It's IPC timeout, but you clearly can see that DSP has generated reply:
```
[ 5126.955337] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000000 intlctl 0xc0000001 rirb 00
[ 5126.955342] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[ 5126.955353] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
```

slawblauciak, 2019-10-30:
The IO transfer timeouts are the most concerning I'd say, as they point to a problem in the communication with the codec.

plbossart, 2019-10-30:
> @plbossart @ranj063 I think you still have some kind of problem with receiving an IPC from DSP. It's IPC timeout, but you clearly can see that DSP has generated reply:
>
> ```
> [ 5126.955337] [...]

can you elaborate here, I am not familiar with the IPC so I don't get your explanation.

> It only reproduced with SDW, so I would look there.

@YvonneYang2 @ClarexZhou can you try with MSI disabled. I also see random timeouts on my side, maybe once a day, so maybe we still have an issue here.

plbossart, 2019-10-30:
@bardliao I wonder if we go one step further than you initial solution and merge the SoundWire handler with the IPC handler.

sinahuang, 2019-10-31:
Dmesg shows "ipc timed out for 0x80010000 size 120" when ran multipipline stress test. Failed at 57th time.

Run with command: ./multiplex_pipeline-test.sh pcm0p pcm2p pcm3c pcm4c

<details>
 <summary>error logger</summary>

````
[ 2866.034808] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 2866.034810] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 2866.035235]  SDW0-Playback: ASoC: hw_params BE SDW0-Playback
[ 2866.035241] rt711 sdw:0:25d:711:0: rt711_pcm_hw_params rt711-aif1
[ 2866.035255] rt711 sdw:0:25d:711:0: Update DAI routes for rt711-aif1 playback
[ 2866.035259] rt711 sdw:0:25d:711:0: Connecting DAI route DP3 Playback -> DP3RX
[ 2866.035262] intel-sdw sdw-master-0: intel_hw_params: SDW0 Pin2: start
[ 2866.035289] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2866.537713] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[ 2866.537725] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve conte
xt
[ 2866.537744] sof-audio-pci 0000:00:1f.3: unknown ROM status value 01009701
[ 2866.538040] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x001d003c trace 0x00000000
[ 2866.538077] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0x00000000 rirb
 00
[ 2866.538082] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 2866.538104] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040000 mask
 0x00000000
[ 2866.538110] sof-audio-pci 0000:00:1f.3: error: failed to set DAI hw_params for link 0 dai->id 2
ALH 7
[ 2866.538117] intel-sdw sdw-master-0: intel_hw_params: SDW0 Pin2: done
[ 2866.538122] intel-sdw sdw-master-0: ASoC: can't set SDW0 Pin2 hw params: -110
[ 2866.538132]  SDW0-Playback: ASoC: hw_params BE failed -110
[ 2866.538135]  SDW0-headphone: ASoC: hw_params BE failed -110
[ 2866.538147]  SDW0-headphone: ASoC: hw_free FE SDW0-headphone
[ 2866.538153] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 0
[ 2866.538249]  DMIC16k: ASoC: found 1 audio capture paths
[ 2866.538254] rt711_rt1308 rt711_rt1308: ASoC: find BE for widget DMIC1.IN
[ 2866.538256] rt711_rt1308 rt711_rt1308: ASoC: try BE (not set)
[ 2866.538258] rt711_rt1308 rt711_rt1308: ASoC: try BE ALH3.IN
[ 2866.538259] rt711_rt1308 rt711_rt1308: ASoC: try BE (not set)

````
</details>

Test recipe:
Kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 9f1bd5b
FW: https://github.com/thesofproject/sof/commits/master commit: 2ca6a4a
tplg: Same with TGL FW branch commit: sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg

Logs:
[multipipeline_test_57_dmesg.log](https://github.com/thesofproject/sof/files/3791778/multipipeline_test_57_dmesg.log)

Scripts:
[multiplex_pipeline-test.txt](https://github.com/thesofproject/sof/files/3791789/multiplex_pipeline-test.txt)

bardliao, 2019-10-31:
> @bardliao I wonder if we go one step further than you initial solution and merge the SoundWire handler with the IPC handler.

@plbossart Yeah I think it also. Let me think about how to do it.

tlauda, 2019-10-31:
@plbossart @YvonneYang2 At least we no longer see IPC timeouts without MSI. However it seems after a while you start to send to FW ALH stream_id = 0xFFFF.

plbossart, 2019-10-31:
> @plbossart @YvonneYang2 At least we no longer see IPC timeouts without MSI. However it seems that after a while you start to send to FW ALH stream_id = 0xFFFF.

that's a feature we use on hw_free, this was agreed on with @slawblauciak 

````c
static int sdw_free_stream(struct device *dev,
			   struct sdw_intel_stream_free_data *free_data)
{
	struct snd_sof_dev *sdev = dev_get_drvdata(dev);
	struct snd_soc_dai *d = free_data->dai;
	struct sof_ipc_dai_config config;
	struct sof_ipc_reply reply;
	int link_id = free_data->link_id;
	int ret;
	u32 size = sizeof(config);

	memset(&config, 0, size);
	config.hdr.size = size;
	config.hdr.cmd = SOF_IPC_GLB_DAI_MSG | SOF_IPC_DAI_CONFIG;
	config.type = SOF_DAI_INTEL_ALH;
	config.dai_index = (link_id << 8) | d->id;
	config.alh.stream_id = 0xFFFF; /* invalid value on purpose */

	/* send message to DSP */
	ret = sof_ipc_tx_message(sdev->ipc,
				 config.hdr.cmd, &config, size, &reply,
				 sizeof(reply));
	if (ret < 0) {
		dev_err(sdev->dev,
			"error: failed to free stream for link %d dai->id %d\n",
			link_id, d->id);
	}

	return ret;
}
````

slawblauciak, 2019-10-31:
Looking at the code, the stream_id errors occur, because of a fairly recent change in dai_config, which immediately tries to get the handshake on dai params. However, this is only needed for IMX ESAI DAI.
Regardless, those errors are not fatal and do not indicate to any actual problem in this scenario.

plbossart, 2019-10-31:
> Looking at the code, the stream_id errors occur, because of a fairly recent change in dai_config, which immediately tries to get the handshake on dai params. However, this is only needed for IMX ESAI DAI.
> Regardless, those errors are not fatal and do not indicate any actual problem in this scenario.

you should not flag an IPC error if there isn't one? I don't understand the answer here.

q009, 2019-11-01:
@plbossart that isn't an IPC error, because it's not propagated through an IPC in dai_prepare. It's only being logged/traced. At least, in THIS particular case, due to a change somebody made in dai_prepare.

paulstelian97, 2019-11-01:
@q009 I'm on that side, should I move my get_handshake call to the actual usage of the ESAI?

q009, 2019-11-01:
@paulstelian97 yeah, sure.

paulstelian97, 2019-11-01:
Alright, will do later, hopefully today.

paulstelian97, 2019-11-01:
@q009 You can check #2043

plbossart, 2019-11-01:
> @plbossart that isn't an IPC error, because it's not propagated through an IPC in dai_prepare (in DAI component). It's only being logged/traced. At least, in THIS particular case, due to a change somebody made in dai_prepare.
> The only reason why we started to see this error in the first place, is because dai_prepare now has an additional call to retrieve the handshake information, regardless of the case and the type of the DAI. The erroneous information returned by that function call is not even used by ALH at that point, that is why the traced error is essentially irrelevant.

@q009 @slawblauciak do you mind pointing me to a commit ID, I don't see any changes in dai_prepare and the changes suggested by @paulstelian97 don't seem to have any sort of impact on Intel DAIs? Really hard to follow the explanations, sorry.

paulstelian97, 2019-11-01:
@plbossart The change I discussed about is only to remove a trace which isn't relevant and might confuse those who attempt to debug this issue. It doesn't directly resolve the issue itself.

q009, 2019-11-01:
@plbossart apologies, it's actually dai_config
https://github.com/thesofproject/sof/pull/1977/commits/1e103942c62311cb83795da0e4ac08daa13cefb6#diff-9f8de02ec82000817dd1a9bbccac158cR685

plbossart, 2019-11-01:
> @plbossart apologies, it's actually dai_config
> [1e10394#diff-9f8de02ec82000817dd1a9bbccac158cR685](https://github.com/thesofproject/sof/commit/1e103942c62311cb83795da0e4ac08daa13cefb6#diff-9f8de02ec82000817dd1a9bbccac158cR685)

ok, so that adds a misleading trace, yes?

q009, 2019-11-01:
That's correct.

YvonneYang2, 2019-11-06:
"ipc timed out for 0x80010000 size 120" still can be reproduced. Failed at 64th time.
<details>
 <summary>dmesg</summary>

````
[ 5415.420886] sof-audio-pci 0000:00:1f.3: pcm: free stream 6 dir 0
[ 5415.420888]  iDisp2: ASoC: hw_free BE iDisp2
[ 5415.420895] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5415.425892]  DMIC: ASoC: pre trigger FE DMIC cmd 0
[ 5415.425895] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 4 dir 1 cmd 0
[ 5415.927821] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[ 5415.927834] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[ 5415.927845] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 5415.927993] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 5415.928006] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000002 intlctl 0xc0000003 rirb 00
[ 5415.928012] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[ 5415.928023] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[ 5415.928030] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[ 5415.928337] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928342] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928346] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928351] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928358] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928363] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for iDisp2
[ 5415.928369] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928372] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928380]  iDisp2: ASoC: close BE iDisp2
[ 5415.928474]  HDMI2: ASoC: close FE HDMI2
[ 5415.928535] sof-audio-pci 0000:00:1f.3: pcm: close stream 6 dir 0
[ 5415.928552]  iDisp2: ASoC: BE iDisp2 event 2 dir 0
[ 5415.928599]  HDMI2: ASoC: BE playback disconnect check for iDisp2
[ 5415.928605]  HDMI2: freed DSP playback path HDMI2 -> iDisp2
[ 5415.928933] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928946] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[ 5415.928950] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928958] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928961] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928968] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.928995] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929268] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929372] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929395] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929409] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929413] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929423] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929606] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929804] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929847] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929859] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929861] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929921] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.929969] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930109] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930170] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930274] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930276] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930280] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930287] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930345] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930533] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930586] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930656] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930721] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930769] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.930924] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.931026] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.931034] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.931078] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.931123] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.931138] sof-audio-pci 0000:00:1f.3: error: trace IO error
[... 18 lines cut ...]
[ 5415.932093] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5415.932130] sof-audio-pci 0000:00:1f.3: error: trace IO error
[ 5416.431787] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[ 5416.431799] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[ 5416.431811] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[ 5416.432048] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[ 5416.432061] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0xc0000002 intlctl 0xc0000003 rirb 01
[ 5416.432066] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[ 5416.432079] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x90020000 mask 0x00000003
[ 5416.432083] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[... cut ...]
tplg: same path as FW branch, sof-cml-rt711-rt1308-mono-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715 

[multipipeline_test_64_dmesg.log](https://github.com/thesofproject/sof/files/3812490/multipipeline_test_64_dmesg.log)
[multipipeline_test_64_logger_error.log](https://github.com/thesofproject/sof/files/3812491/multipipeline_test_64_logger_error.log)
[multipipeline_test_64_all_logger.log](https://github.com/thesofproject/sof/files/3812492/multipipeline_test_64_all_logger.log)

YvonneYang2, 2019-11-11:
Checked it 300 times with legacy interrupt mode, issue cannot be reproduced.

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 391b5f9c 
FW: https://github.com/thesofproject/sof/commits/master  commit: fb068e6a
tplg: same path as FW branch, sof-cml-rt711-rt1308-mono-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

lgirdwood, 2019-11-11:
@YvonneYang2 are you confirming issue is resolved with legacy IRQ mode but still valid fro MSI IRQ mode ?

YvonneYang2, 2019-11-12:
@lgirdwood  
Yes, issue still can be reproduced with MSI IRQ mode while cannot be reproduced with legacy IRQ mode.

Tested it 300 times with legacy IRQ mode , issue cannot reproduced, 
Tested it with MSI IRQ mode, failed at 4th time.

**dmesg with MSI mode**
```
[   69.685130] sdw_disable_stream: SDW1 Pin2-Playback: start
[   69.685131] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   69.687705]  DMIC: ASoC: pre trigger FE DMIC cmd 0
[   69.687707] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 4 dir 1 cmd 0
[   70.190207] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   70.190217] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   70.190228] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   70.190505] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   70.190514] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000002 intlctl 0xc0000003 rirb 00
[   70.190517] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[   70.190523] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[   70.190525] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   70.190542] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for iDisp2
[   70.190546] sof-audio-pci 0000:00:1f.3: error: trace IO error
[   70.190548]  iDisp2: ASoC: trigger BE failed -110
[   70.190551]  HDMI2: ASoC: trigger FE cmd: 0 failed: -110
[   70.190558]  HDMI2: ASoC: hw_free FE HDMI2
[   70.190564] sof-audio-pci 0000:00:1f.3: pcm: free stream 6 dir 0
[   70.190579] sof-audio-pci 0000:00:1f.3: error: trace IO error
[   70.190581] sof-audio-pci 0000:00:1f.3: error: trace IO error
[   70.190587] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[   70.190591] sof-audio-pci 0000:00:1f.3: error: trace IO error
```

[multipipeline_test_4_dmesg.log](https://github.com/thesofproject/sof/files/3834135/multipipeline_test_4_dmesg.log)
[sof-trace.log](https://github.com/thesofproject/sof/files/3834136/sof-trace.log)

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 391b5f9c
FW: https://github.com/thesofproject/sof/commits/master commit: fb068e6
tplg: same path as FW branch, sof-cml-rt711-rt1308-mono-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

YvonneYang2, 2019-11-12:
Error not seen after 150 times multipipline stress test with latest integration/soundwire-latest and master. Will increase the stress iterations to verify if issue can be reproduced or not.

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 2933f73 
FW: https://github.com/thesofproject/sof/commits/master  commit: d1dcdfe
tplg: same path as FW branch, sof-cml-rt711-rt1308-mono-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

sinahuang, 2019-11-12:
Original issue can't be reproduced in MSI mode, but there are some other errors show "Link power down failed: -11" in dmesg info.

Test for 500 times, failed at 83th time.

Dmesg shows with MSI mode:

```
[ 5887.621195] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5887.621293] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 5887.621295] intel-sdw sdw-master-3: intel_hw_free: SDW3 Pin2: done
[ 5887.621298]  SDW3-Capture: ASoC: close BE SDW3-Capture
[ 5887.621300] intel-sdw sdw-master-3: intel_shutdown: SDW3 Pin2: start
[ 5887.621303] intel-sdw sdw-master-3: intel_shutdown: SDW3 Pin2: done
[ 5887.621331]  DMIC: ASoC: close FE DMIC
[ 5887.621334] sof-audio-pci 0000:00:1f.3: pcm: close stream 4 dir 1
[ 5887.621415] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 3607 <= 0000
[ 5887.621512] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 3627 <= 0000
[ 5887.621676] rt715 sdw:3:25d:715:0: [rt715_sdw_write] 3501 <= 0003
[ 5887.621694]  SDW3-Capture: ASoC: BE SDW3-Capture event 2 dir 1
[ 5887.621715]  DMIC: ASoC: BE capture disconnect check for SDW3-Capture
[ 5887.621718]  DMIC: freed DSP capture path DMIC <- SDW3-Capture
[ 5888.374937] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 5888.375254] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[ 5890.951473] intel-sdw sdw-master-1: intel_suspend_runtime start
[ 5890.951476] intel-sdw sdw-master-0: intel_suspend_runtime start
[ 5890.951541] intel-sdw sdw-master-1: intel_suspend_runtime done
[ 5890.951543] intel-sdw sdw-master-3: intel_suspend_runtime start
[ 5890.951602] intel-sdw sdw-master-3: intel_suspend_runtime done
[ 5890.952051] intel-sdw sdw-master-0: Link power down failed: -11
[ 5892.775776]  HDMI2: ASoC: pop wq checking: Playback status: inactive waiting: yes
[ 5892.775845]  Headphone: ASoC: pop wq checking: Playback status: inactive waiting: yes
[ 5892.775905]  SDW1-speakers: ASoC: pop wq checking: Playback status: inactive waiting: yes

```
Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 2933f73
FW: https://github.com/thesofproject/sof/commits/master commit: d1dcdfe
tplg: same path as FW branch, sof-cml-rt711-rt1308-mono-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

[dmesg.log](https://github.com/thesofproject/sof/files/3835988/dmesg.log)
[sof_error_trace.log](https://github.com/thesofproject/sof/files/3835995/sof_error_trace.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/3835997/sof_trace.log)

YvonneYang2, 2019-11-13:
Tested it with MSI IRQ mode, issue cannot be reproduced. But there's two lines error info from time to time shows in dmesg. 

**Error info**
```
[  245.350092] rt1308 sdw:1:25d:1308:0: Parity error detected
```
```
[  245.358867] rt715 sdw:3:25d:715:0: Bus clash error detected
```

[multipipeline_test_19_dmesg.log](https://github.com/thesofproject/sof/files/3839985/multipipeline_test_19_dmesg.log)
[multipipeline_test_19_all_logger.zip](https://github.com/thesofproject/sof/files/3839987/multipipeline_test_19_all_logger.zip)

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: fb40b85 + PR1451
FW: https://github.com/thesofproject/sof/commits/master  commit: 835d876
tplg: same path as FW branch, sof-cml-rt711-rt1308-mono-rt715.tplg
platform: CML SDW Laptop with ALC711+ ALC1308+ ALC715

YvonneYang2, 2019-11-13:
Issue can still be reproduced without https://github.com/thesofproject/linux/pull/1451, failed at 67th time.

```
[  732.917935]  Headset mic: ASoC: pre trigger FE Headset mic cmd 0
[  732.917938] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 1 dir 1 cmd 0
[  733.416823] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[  733.416837] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  733.416847] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  733.417107] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[  733.417120] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000006 intlctl 0xc0000007 rirb 00
[  733.417126] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[  733.417138] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[  733.417143] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  733.417485] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  733.417489] sof-audio-pci 0000:00:1f.3: error: trace IO error
```
[multipipeline_test_67_all_logger.log](https://github.com/thesofproject/sof/files/3840327/multipipeline_test_67_all_logger.log)
[multipipeline_test_67_logger_error.log](https://github.com/thesofproject/sof/files/3840328/multipipeline_test_67_logger_error.log)
[multipipeline_test_67_dmesg.log](https://github.com/thesofproject/sof/files/3840329/multipipeline_test_67_dmesg.log)