thesofproject/sof #1937: [BUG] DSP panic when using SSP0 on BYT

plbossart, 2019-10-14:
**Describe the bug**

For BYT-CR, we need to use SSP0 instead of SSP2. A tentative kernel patch is suggested here:
https://github.com/thesofproject/linux/pull/1312

The PR generates a DSP panic, which can be reproduced with the regular kernel with partial changes to the topology.

It's likely that I don't fully get the topology changes needed, but regardless a DSP panic is a really bad thing ...

**To Reproduce**

use SOF master for kernel and firmware

kernel: 5edcfdf3907354de7a260a451ba611f22c5e6a07
firmware: 40bf5e070e9fb6a61174b26d401f1eb787f959d1

compile the kernel in nocodec mode with the attached config 
[sof-dev-defconfig.txt](https://github.com/thesofproject/sof/files/3725868/sof-dev-defconfig.txt)

boot and modprobe snd-sof-acpi, then play on hw:0,0

Then add the attached diff, which completely replaces the use of SSP2 by SSP0
[nocodec_ssp0_ok.diff.txt](https://github.com/thesofproject/sof/files/3725873/nocodec_ssp0_ok.diff.txt)

Try with this change, there should be no issue

Then revert the changes, and apply the diff 
[nocodec_ssp0_nok.diff.txt](https://github.com/thesofproject/sof/files/3725874/nocodec_ssp0_nok.diff.txt)

That generates a DSP panic, see log attached 
[dmesg_ssp0_dsp_panic.log](https://github.com/thesofproject/sof/files/3725886/dmesg_ssp0_dsp_panic.log)

Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
100%

**Expected behavior**

Error reported with no DSP panic

**Impact**

DSP panic -> P1 in my book.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5edcfdf3907354de7a260a451ba611f22c5e6a07
    * SOF: 40bf5e070e9fb6a61174b26d401f1eb787f959d1

2) Name of the topology file
    * Topology: sof-byt-nocodec
3) Name of the platform(s) on which the bug is observed.
    * Platform: baytrail

**Screenshots or console output**

no sof-logger output available

plbossart, 2019-10-14:
@ranj063 could you take a look at the topology file changes to see if there is anything obvious that should be fixed?

tlauda, 2019-10-15:
@plbossart Could you check with https://github.com/thesofproject/sof/pull/1942?

plbossart, 2019-10-15:
@tlauda PR #1942 does improve things, I don't get a DSP panic but a regular IPC error.

````
[   51.694423] sof-audio-acpi 80860F28:00: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   51.695165] sof-audio-acpi 80860F28:00: error: ipc error for 0x60010000 size 20
[   51.695172] sof-audio-acpi 80860F28:00: error: hw params ipc failed for stream 0
[   51.695177] sof-audio-acpi 80860F28:00: ASoC: 80860F28:00 hw params failed: -22
[   51.695182]  Low Latency: ASoC: hw_params FE failed -22
````

now I need to figure out how to replace SSP2 by SSP0, which is about to test my understanding of topology stuff (or lack thereof rather).

ranj063, 2019-10-15:
> @ranj063 could you take a look at the topology file changes to see if there is anything obvious that should be fixed?

@plbossart the topology changes look good. Im not sure why you're seeing this error. I had a similar error on my BYT this morning that got resolved by updated my FW with master

tlauda, 2019-10-15:
@plbossart As I already mentioned, you're getting this error because you are running the stream without previously configuring SSP 0 with DAI config. I suspect that this is the reason:
```
-DAI_CONFIG(SSP, 2, 2, NoCodec-2,
+DAI_CONFIG(SSP, 0, 2, NoCodec-2,
```

Wild guess, but maybe the second number is the actual dai index?

tlauda, 2019-10-15:
Edit: To be precised, you're sending dai config to the dai, which doesn't exist.

plbossart, 2019-10-15:
no, the first number is definitively the DAI index, and the second the dai link index.

the error does not happen while loading the topology but when playing audio. I believe I am sending the SSP0 index during the IPC_DAI_CONFIG, but I am not sure what happens during the hw_params (can't figure out how the DAI IPC happens).

There are additional issues with the topology, doing a blind change for SSP2->SSP0 isn't going to work I am afraid, there are a couple of shortcuts between DAI and DAI-links.

tlauda, 2019-10-16:
@plbossart You're changing dai index just in DAI_CONFIG but not in DAI_ADD macros. Basically you are configuring SSP 0, but you have DAI components connected to SSP 2.

tlauda, 2019-10-16:
@plbossart Try #1951. It should return an error on dai config.

tlauda, 2019-10-17:
@plbossart Can we close the issue?