thesofproject/sof #1958: [BUG]No errors reported when a new pipeline with an already existing pipeline ID is created

ranj063, 2019-10-16:
**Describe the bug**
When a new pipeline is created, the FW checks to see if the comp_id exists but doesnt check if the pipeline_id already exists.

The problem is that when creating topology, there is no way to verify that the pipeline ID's are unique. The issue is compounded when we load multiple topologies with each topology oblivious about the pipeline ID's in the other topology.

**To Reproduce**
Modify any topology with 2 pipelines with same pipeline ID.

**Reproduction Rate**
100%

**Expected behavior**
The FW should throw an error when the second pipeline with a duplicate pipeline ID is created.

**Impact**
Show-stopper.
The topology loading shows no errors but the pipeline data is corrupted with 2 pipelines having the same pipeline ID.

ranj063, 2019-10-16:
@tlauda there are 2 ways to fix this problem and I suppose we need both. 
The FW should reject a new pipeline request with a duplicate pipeline ID.
Also, in the driver we could have a monotonically increasing pipeline ID's that make sure that the ID's are always unique. But this would mean that the pipeline ID that the topology provides is only a reference and could be overwritten by the driver.

What do you think?

tlauda, 2019-10-16:
@ranj063 Agree on the first one, however I'm not sure how would you like to resolve the second one. In topology we define DAI apart from the rest of the pipeline and "connect" them together based on pipeline id. How would the driver know which DAIs are connected to which pipeline and how to pair them? In theory we can change PIPELINE macros in topologies, so it will take DAI as an argument. This way we wouldn't need to assign ids at all and let the driver do it.

ranj063, 2019-10-16:
> @ranj063 Agree on the first one, however I'm not sure how would you like to resolve the second one. In topology we define DAI apart from the rest of the pipeline and "connect" them together based on [...]

The pipeline ID's are only used to find out the name of the widget to connect to. So for example, PIPELINE_SINK_1 should translate to BUF1.1 and the SSP2.out would connect to BUF1.1. So in the end what we are left with is just connections between widget names. The driver deals with this already. With multiple topologies, it is going to be hard to put down the requirement that the pipeline ID's have to be unique because you never know which combination of topologies would get loaded.

tlauda, 2019-10-16:
@ranj063 If the driver can handle your idea, then go for it. I will take care of the first issue.

plbossart, 2019-10-16:
On 10/16/19 1:20 PM, Tomasz Lauda wrote:
> @ranj063 <https://github.com/ranj063> If the driver can handle your 
> idea, then go for it. I will take care of the first issue.

I am not hot on the driver mucking with the topology...

ranj063, 2019-10-16:
> I am not hot on the driver mucking with the topology...

@plbossart the solution to require different topologies to have unique pipeline ID's would not scale very well. We might be OK today with a limited number of topologies but this would be very hard to monitor when there different kinds of clients. The "mucking" part that I am proposing is actually not evil at all. Each topology file can have pipeline ID's starting at 1 but in reality they are just offset based on what other other topologies have been loaded previously. Let me propose my solution and then you can take a call on it?

plbossart, 2019-10-16:
> @plbossart <https://github.com/plbossart> the solution to require
> different topologies to have unique pipeline ID's would not scale very
> well. We might be OK today with a limited number of [...]

well, you need to talk to Cezary here, since the topology does define 
the buffer IDs that will be used for the probe. So if the driver starts 
reallocating IDs, it'll make the probe usage more complicated.

ranj063, 2019-10-16:
hmm what do you mean by buffer ID here? 
The buffer widget in the conf file only has the information as below. The widget ID is actually allocated in the driver. The "index" part is what bothers me a bit in my solution. Each widget comes associated with a pipeline ID and this information needs to be updated in the driver if I want to modify the pipeline ID's. But like I said, I think it might be possible. Let me give it a try. 

# Playback Buffers
SectionVendorTuples."BUF1.0_tuples" {
        tokens "sof_buffer_tokens"
        tuples."word" {
                SOF_TKN_BUF_SIZE        "768"
                SOF_TKN_BUF_CAPS        "  113"
        }
}
SectionData."BUF1.0_data" {
        tuples "BUF1.0_tuples"
}
SectionWidget."BUF1.0" {
        index "1"
        type "buffer"
        no_pm "true"
        data [
                "BUF1.0_data"
        ]
}
SectionVendorTuples."BUF1.1_tuples" {
        tokens "sof_buffer_tokens"
        tuples."word" {
                SOF_TKN_BUF_SIZE        "576"
                SOF_TKN_BUF_CAPS        "  113"
        }
}
SectionData."BUF1.1_data" {
        tuples "BUF1.1_tuples"
}
SectionWidget."BUF1.1" {
        index "1"
        type "buffer"
        no_pm "true"
        data [
                "BUF1.1_data"
        ]
}

tlauda, 2019-10-17:
@ranj063 Issue fixed on FW side. Let's move this discussion outside of this bug.

closed 2019-10-17 by tlauda