thesofproject/sof #1973: [BUG] ipc timeout when play via speaker pipeline with demux tplg

aiChaoSONG, 2019-10-21:
**Describe the bug**
On CML Chrome(hatch), when play via speaker pipeline, first hw params install failed in the  terminal, check dmesg, you can see DSP panic and ipc timeout

**To Reproduce**
run aplay on speaker pipeline: `aplay -Dhw:0,6 -f dat -c 2 -i -vvv ./long.wav`

**Reproduction Rate**
%100

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {97ccba77}
    * SOF: {3d5d3146}
2) Name of the topology file
    * Topology: {sof-cml-demux-rt5682-max98357a.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML Chrome(hatch)}

***dmesg***
```
[  190.047786] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  190.048007] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[  190.048011] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[  190.048016] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  190.048091] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[  190.048092] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  190.048099] sof-audio-pci 0000:00:1f.3: error: trace IO error
[  190.551547] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60010000 size 108
[  190.551560] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  190.551573] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  190.551682] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xbe00897c trace 0x00004000
[  190.551700] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[  190.551706] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  190.551719] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  190.551725] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  190.551733] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[  190.551741] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
[  190.551750]  Speakers: ASoC: hw_params FE failed -110
[  190.551763] sof-audio-pci 0000:00:1f.3: pcm: free stream 6 dir 0
[  190.552250] sof-audio-pci 0000:00:1f.3: pcm: close stream 6 dir 0
```
***sof-logger***
```
 CORE  LEVEL   COMP_ID    TIMESTAMP            DELTA            FILE_NAME	             CONTENT
   0      1     COMP     77526649.270833      4.739583          of/audio/component.h:421     comp_cmd() error: invalid version, data->magic = 0, data->abi = 65536
   0      1     IPC      3003167.864583       3003167.750000    src/ipc/handler.c:855 	     ipc: comp 2 cmd 3 failed -22
```

[dmesg.log](https://github.com/thesofproject/sof/files/3748902/dmesg.log)

tlauda, 2019-10-30:
@aiChaoSONG Please verify with #2009.

dengyangchao, 2019-10-31:
tried  #2009 on CML Chrome
boot up the device and occur DSP panic, also hw params install failed

<details>
 <summary>Dmesg</summary>

````
[   40.132058] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 3 dir 0
[   40.132062] sof-audio-pci 0000:00:1f.3: generating page table for 0000000003363ba9 size 0x4b00 pages 5
[   40.132076] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   40.132100] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   40.132106] sof-audio-pci 0000:00:1f.3: period_bytes:0x12c0
[   40.132109] sof-audio-pci 0000:00:1f.3: periods:4
[   40.132127] sof-audio-pci 0000:00:1f.3: stream_tag 2
[   40.132139] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   40.132250] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[   40.132266] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[   40.132284] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.132390] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   40.132396] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   40.132402] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   40.132408] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   40.132415] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   40.132422] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000044 PS       0x00060b25 SAR     0x00000000
[   40.132429] sof-audio-pci 0000:00:1f.3: EPC1     0xbe00ee9c EPC2     0xbe01cce5 EPC3     0x00000000 EPC4    0x00000000
[   40.132435] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   40.132442] sof-audio-pci 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   40.132448] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   40.132454] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05eb30
[   40.132471] sof-audio-pci 0000:00:1f.3: 0xbe05eb30: be00ee9c 00000000 00000000 00000000
[   40.132488] sof-audio-pci 0000:00:1f.3: 0xbe05eb34: 00000000 00000000 be056300 be05ee20
[   40.132500] sof-audio-pci 0000:00:1f.3: 0xbe05eb38: c382ba00 f886f1ae 9195b828 ffffa442
[   40.132508] sof-audio-pci 0000:00:1f.3: 0xbe05eb3c: 9447d500 ffffa442 93fc4a00 ffffa442
[   40.132516] sof-audio-pci 0000:00:1f.3: 0xbe05eb40: a76bbb12 ffffffff 00000000 00000000
[   40.132523] sof-audio-pci 0000:00:1f.3: 0xbe05eb44: 92b89c00 ffffa442 9447d500 ffffa442
[   40.132530] sof-audio-pci 0000:00:1f.3: 0xbe05eb48: 00000000 00000000 00000000 00000000
[   40.132537] sof-audio-pci 0000:00:1f.3: 0xbe05eb4c: 00000000 00000000 00000000 00000000
[   40.132542] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   40.639428] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60010000 size 108
[   40.639439] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[   40.639449] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   40.639549] sof-audio-pci 0000:00:1f.3: error: runtime exception
[   40.639553] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[   40.639557] sof-audio-pci 0000:00:1f.3: error: panic at :0
[   40.639562] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[   40.639566] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[   40.639572] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000044 PS       0x00060b25 SAR     0x00000000
[   40.639577] sof-audio-pci 0000:00:1f.3: EPC1     0xbe00ee9c EPC2     0xbe01cce5 EPC3     0x00000000 EPC4    0x00000000
[   40.639582] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[   40.639587] sof-audio-pci 0000:00:1f.3: EPS2     0x00060b20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[   40.639591] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[   40.639595] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05eb30
[   40.639603] sof-audio-pci 0000:00:1f.3: 0xbe05eb30: be00ee9c 00000000 00000000 00000000
[   40.639608] sof-audio-pci 0000:00:1f.3: 0xbe05eb34: 00000000 00000000 be056300 be05ee20
[   40.639614] sof-audio-pci 0000:00:1f.3: 0xbe05eb38: daab2d00 91f99033 9195b828 ffffa442
[   40.639619] sof-audio-pci 0000:00:1f.3: 0xbe05eb3c: 80e939e8 ffffbab7 00000000 00000000
[   40.639624] sof-audio-pci 0000:00:1f.3: 0xbe05eb40: 933289e8 ffffa442 933289f0 ffffa442
[   40.639629] sof-audio-pci 0000:00:1f.3: 0xbe05eb44: 92f95400 ffffa442 9195b828 ffffa442
[   40.639634] sof-audio-pci 0000:00:1f.3: 0xbe05eb48: 84de6a00 ffffa442 daab2d00 0000006c
[   40.639639] sof-audio-pci 0000:00:1f.3: 0xbe05eb4c: 00000000 00000000 00000000 00000000
[   40.639656] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[   40.639660] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   40.639672] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   40.639677] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[   40.639682] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[... cut ...]
    0      2          DMA          67118600.781250         5.885417 intel/cavs/hda-dma.c:587 	hda-dmac: 5 channel 1 -> config
    0      2       VOLUME          67118608.645833         7.864583 udio/volume/volume.c:244 	volume_params()

````
</details>

[amixer.txt](https://github.com/thesofproject/sof/files/3792385/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/3792386/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3792387/logger.log)
[logger-error.log](https://github.com/thesofproject/sof/files/3792388/logger-error.log)

tlauda, 2020-03-13:
@dengyangchao Still reproduced?

aiChaoSONG, 2020-03-13:
@tlauda The only hatch in PRC is broken, we are unable to verify this issue.

tlauda, 2020-04-22:
@aiChaoSONG Demux component and topology changed. Please reopen if ever retested.

closed 2020-04-22 by tlauda