thesofproject/sof #1974: [BUG]

YvonneYang2, 2019-10-21:
**Describe the bug**
Paplay playback stream drained when wake up from s3 during paplay.

**To Reproduce**
1.Paplay speaker/HDMI/DP via command "paplay wav/paplay wav/48California_Dreaming.wav -vvv"
2.Do s3 via "sudo rtcwake -m mem -s 10"

**frequency**
4/5

**Expected result**
Paplay output normally after wake up from s3

**Actual result**
paplay stopped and playback stream drained after wake up from s3.
```
ubuntu@sh-icl-rvp-hda-001:~$ paplay wav/48California_Dreaming.wav -vvv
Opening a playback stream with sample specification 's16le 2ch 48000Hz' and channel map 'front-left,front-right'.
Connection established.
Stream successfully created.
Buffer metrics: maxlength=4194304, tlength=384000, prebuf=380164, minreq=3840
Using sample spec 's16le 2ch 48000Hz', channel map 'front-left,front-right'.
Connected to device alsa_output.pci-0000_00_1f.3-platform-sdw_rt711_rt1308_rt715.HiFi__hw_sofsdwrt7111308_6__sink (index: 2, suspended: no).
Stream started.
Stream underrun.
Stream started.
Stream underrun.
Stream started.
Stream underrun.
Stream started.
Stream underrun.
Stream started.
Stream underrun.
Stream started.
Stream underrun.
Stream started.
Playback stream drained.y: 68829416 usec.
Draining connection to server.
```

**Environment**
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 29debf8
FW: https://github.com/thesofproject/sof/commits/master  commit: 3d5d314
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

[drained_dmesg.log](https://github.com/thesofproject/sof/files/3749370/drained_dmesg.log)
[drained_logger.log](https://github.com/thesofproject/sof/files/3749371/drained_logger.log)

sinahuang, 2019-10-21:
issue still can be reproduced on CML RVP with codec ALC700 in SDW mode

tlauda, 2019-10-30:
@YvonneYang2 @sinahuang FW just receives STOP command from kernel. There are no xruns or anything wrong from FW perspective. Please move this bug to linux repo.

plbossart, 2019-10-30:
Is this really a SoundWire problem? Have you checked that this sequence works with I2S or HDMI?

dengyangchao, 2019-10-31:
issue also can reproduce on CML Chrome I2S

Environment
platform: CML Chrome I2S
Firmware: 59c93383 https://github.com/thesofproject/sof/commits/master
Kernel: f5e70ce https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-cml-rt5682-max98357a.tplg same as firmware

[dmesg.log](https://github.com/thesofproject/sof/files/3791879/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3791880/logger.log)

tlauda, 2019-10-31:
@dengyangchao Still the same case - nothing wrong from the FW perspective. Please move to linux repository.

closed 2019-10-31 by tlauda