thesofproject/sof #1975: [BUG] [SDW] Headset and DMIC name should update in "aplay -l" and "arecord -l"

ClarexZhou, 2019-10-21:
**Describe the bug**
Headset and DMIC pipline name should update.

**To Reproduce**
1. aplay -l to check the pipline name
2. arecord -l to check the pipline name

**frequency**
100%

**Actual result**
'SDW0-speakers (*)' is shown for headset.
SDW3-mics (*) is shown for DMIC
```
ubuntu@sh-icl-rvp-hda-001:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 1: SDW0-mics (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 4: SDW3-mics (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

ubuntu@sh-icl-rvp-hda-001:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 0: SDW0-speakers (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 2: SDW1-speakers (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofsdwrt7111308 [sof-sdw-rt711-1308-715], device 3: SDW2-speakers (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```
**Expected result**
Headset and DMIC name should be listed. Take below on GLK for example.
```
test@test-Bobba:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofglkda7219max [sof-glkda7219max], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofglkda7219max [sof-glkda7219max], device 99: DMIC01 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice 

test@test-Bobba:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofglkda7219max [sof-glkda7219max], device 0: Speakers (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofglkda7219max [sof-glkda7219max], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

**Environment**
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 29debf8
FW: https://github.com/thesofproject/sof/commits/master  commit: 3d5d314
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

bardliao, 2019-10-21:
@ClarexZhou The name is define on topology. Is there any rule that we can follow?  Actually, I don't know why "SDW0-mics" is acceptable but "SDW3-mics" is not.

plbossart, 2019-10-21:
yeah, it's not a bug, it's just a naming convention in the topology file

We can follow the convention 'headset', 'speakers', 'mics' as needed.

But even for GLK, a playback stream called 'headset' is not very good if you use a headphone with no mics.

PR #1982 (2019-10-29): update topology for soundwire cml
Update topology for soundwire cml.

Fixes #1975
closed 2019-10-29 by tlauda