thesofproject/sof #1980: [BUG] SOF FW compiling fail when disable TRACE in menuconfig

keyonjie, 2019-10-22:
**Describe the bug**
SOF FW compiling fail when disable TRACE in menuconfig

**To Reproduce**
run make menuconfig, de-select the TRACE, save and build.

**Reproduction Rate**
100%
**Expected behavior**
build should success.
**Impact**
It block debugging the FW behavior on removing trace.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: any
    * SOF: master, latest
2) Name of the topology file
    * Topology: any
3) Name of the platform(s) on which the bug is observed.
    * Platform: {cnl}

paulstelian97, 2019-10-22:
i.MX is also affected. I'm getting compile error from kpb.c,

paulstelian97, 2019-10-22:
Tentative fix pushed.

closed 2019-10-22 by tlauda

paulstelian97, 2019-10-22:
@keyonjie Can you please check the issue is resolved? (if not please reopen)

PR #1985 (2019-10-22): sof: Fix compile errors when traces are disabled.
This fixes #1980 on the i.MX platform and should fix it on the others.

I cannot check platform specific code in other platforms, and CI doesn't
test this case. @keyonjie Can you please check if the code builds with this PR?