thesofproject/sof #1981: [BUG][SDW] Xrun error occurred when run aplay(RT711) stress test.

YvonneYang2, 2019-10-22:
**Describe the bug**
Xrun error occurred when run aplay stress test at 486 times. Terminate it and restart aplay can recover.

**To Reproduce**
Run aplay stress test via "./sof-test-playback.sh" (aplay0,0)

script: 
[sof-test-playback.sh.txt](https://github.com/thesofproject/sof/files/3753717/sof-test-playback.sh.txt)

**frequency**
1/486

**Expected result**
Aplay normaly

**Actual result**
Aplay stopped at 486th.

**Environment**
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 29debf8
FW: https://github.com/thesofproject/sof/commits/master  commit: 1dbd14b
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

<details>
 <summary>error logger</summary>

````

 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 1.4     3354725574.427083              nan      src/audio/dai.c:605 	dai_check_for_xrun() error: underrun due to no data available
    0      1         COMP         3354725584.322917         9.895833 of/audio/component.h:743 	comp_underrun() error: dev->comp.id = 4, source->avail = 1152, copy_bytes = 0
    0      1         PIPE         3354725682.447917        98.125000 src/audio/pipeline.c:838 	pipeline_copy() error: ret = -61
    0      1         PIPE 1.5     3354725692.708333        10.260417 src/audio/pipeline.c:1042	pipeline_task(): xrun recover failed! pipeline will be stopped!

````
</details>

[aplay_error_486_logger_all.log](https://github.com/thesofproject/sof/files/3753704/aplay_error_486_logger_all.log)
[aplay_error_486_logger_error.log](https://github.com/thesofproject/sof/files/3753705/aplay_error_486_logger_error.log)
[aplay_error_486_dmesg.log](https://github.com/thesofproject/sof/files/3753706/aplay_error_486_dmesg.log)

tlauda, 2019-10-22:
@YvonneYang2 Does it reproduce with linux#1277 included?

YvonneYang2, 2019-10-22:
Test it with linux#1277, issue not reproduced after running 500 times.  Same with https://github.com/thesofproject/linux/issues/1329

dengyangchao, 2019-10-22:
when use headset start a playback/or repeat playback several times, it will occur I/O error
on ICL RVP, with onboard codec ALC700 in SDW mode

kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest  29debf8
sof: https://github.com/thesofproject/sof/commits/master  1dbd14b
Tplg:  sof-icl-rt700.tplg

<details>
 <summary>Logger-error</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DAI 1.4      94001208.802083  94001208.000000      src/audio/dai.c:605 	dai_check_for_xrun() error: underrun due to no data available
    0      1         COMP          94001218.489583         9.687500 of/audio/component.h:743 	comp_underrun() error: dev->comp.id = 4, source->avail = 1152, copy_bytes = 0
    0      1         PIPE          94001315.052083        96.562500 src/audio/pipeline.c:838 	pipeline_copy() error: ret = -61
    0      1         PIPE 1.5      94001325.156250        10.104167 src/audio/pipeline.c:1042	pipeline_task(): xrun recover failed! pipeline will be stopped!

````
</details>

[dmesg.log](https://github.com/thesofproject/linux/files/3754163/dmesg.log)
[headset-playback_logger_error.log](https://github.com/thesofproject/linux/files/3754164/headset-playback_logger_error.log)
[logger.log](https://github.com/thesofproject/linux/files/3754165/logger.log)

YvonneYang2, 2019-10-23:
Issue not reproduced with test recipe as below (linux#1277 has been merged) 

Test recipe:
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 7c85d69
FW: https://github.com/thesofproject/sof/commits/master  commit: 8b059f6
tplg: Same with FW branch, sof-icl-rt711-rt1308-rt715-hdmi.tplg  rename  to  sof-icl-rt711-rt1308-rt715.tplg
platform: ICL RVP, with codec ALC711 -RT1308, RT715 in SDW mode

closed 2019-10-23 by YvonneYang2