thesofproject/sof #1988: [BUG][BYT] alsabat test fails due to waveform distortion

fredoh9, 2019-10-22:
**Describe the bug**
alsabat frequency analysis fails occasionally due to waveform distortion in BYT_MB_NOCODEC platform. This is not 100% but easy to reproduce, 1-2 failures out of 10 test. NOCODEC mode use SOF firmware SSP loopback.

**To Reproduce**
Run alsabat test and check test result

1, simple test
alsabat -Dhw:0,0 -c 2 -r 48000 -f s16_le

2. save to file and standalone analysis
 alsabat -Phw:0,0 -c 2 -r 48000 -f s16_le --standalone -n 240000 &
 sleep 0.2
 arecord -Dhw:0,0 -c 2 -r 48000 -f s16_le rec.wav;sync
 alsabat --local --file rec.wav

**Reproduction Rate**
1-2/10

**Expected behavior**
Alsabat test should return success, 0.

**Impact**
DEVICE_TEST will fail for PR test and daily test

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: fd274c2b
    * SOF: 1dbd14b3
2) Name of the topology file
    * Topology: sof-byt-nocodec.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: baytrail, BYT_MB_NOCODEC

**Screenshots or console output**
Attached multiple pictures
[image]
[image]
[image]
[image]

sof debugging files are attached as a zip
[test_outputs.zip](https://github.com/thesofproject/sof/files/3758177/test_outputs.zip)

tlauda, 2019-10-22:
Could you check whether #1984 improves it?

fredoh9, 2019-10-22:
@tlauda sure, i can pick up #1984 and update the results here

fredoh9, 2019-10-22:
@tlauda I updated only sof build. booting fine loading firmware and tplg is fine. But any aplay have hw_param failed with IPC error

> [   81.806856] sof-audio-acpi 80860F28:01: error: ipc error for 0x60010000 size 20
> [   81.806865] sof-audio-acpi 80860F28:01: error: hw params ipc failed for stream 0
> [   81.806898] sof-audio-acpi 80860F28:01: ASoC: 80860F28:01 hw params failed: -12
> [   81.806904]  Low Latency: ASoC: hw_params FE failed -12

[1022_new_pr1984_ipc_error.log](https://github.com/thesofproject/sof/files/3758718/1022_new_pr1984_ipc_error.log)

tlauda, 2019-10-22:
Weird. It passes CI flawlessly.

tlauda, 2019-10-23:
@fredoh9 Could you retry? Nobody else sees this problem with #1984.

fredoh9, 2019-10-23:
Sure, i will plan to build your PR and re-test again. I will update the result here

fredoh9, 2019-10-23:
I'm able to test with #1984, tlauda:topic/dma-separate-buffers-new.
Got better and less occurrence than before, but still I'm seeing  alsabat frequency analysis failure.

>  ------------->>>>> LOOP 48 <<<<<--------------
> alsa-utils version 1.1.3
> 
> Entering playback thread (ALSA).
> Get period size: 2064  buffer size: 16368
> Playing generated audio sine wave
> Entering capture thread (ALSA).
> Get period size: 4080  buffer size: 16320
> Recording ...
> Playback completed.
> Capture completed.
> 
> BAT analysis: signal has 65536 frames at 48000 Hz, 2 channels, 2 bytes per sample.
> 
> Channel 1 - Checking for target frequency 997.00 Hz
> Amplitude: 31078.1; Percentage: [94]
> Detected peak at 988.77 Hz of 20.25 dB
>  Total 20.3 dB from 988.77 to 988.77 Hz
>  FAIL: Peak freq too low 988.77 Hz
> Detected peak at 996.83 Hz of 37.61 dB
>  Total 41.3 dB from 990.23 to 1005.62 Hz
>  PASS: Peak detected at target frequency
> Detected peak at 1007.08 Hz of 20.25 dB
>  Total 41.3 dB from 1007.08 to 1007.08 Hz
>  FAIL: Peak freq too high 1007.08 Hz
> Detected at least 3 signal(s) in total
> 
> Return value is -1003
>

tlauda, 2019-10-23:
@fredoh9 Could you attach recording screenshot? Does it look the same as previous ones?

fredoh9, 2019-10-23:
@tlauda yes and no. I found a missing samples problem. The others wasn't noticeable waveform like that. I will attach wavefiles and logs.

fredoh9, 2019-10-23:
I ran 100 loop on two BYT_MB_NOCODEC devices. failures are 6/100 and 8/100. It was much worse before.

1. fred-mb (6/100, attached 3 files)
[fred-mb.zip](https://github.com/thesofproject/sof/files/3764519/fred-mb.zip)
Some of them after zoom in, i could see missing samples. alsarec_140341.wav is not visible.

2. jf-byt-mb3(8/100, attached 2 files)
[jf_mb3.zip](https://github.com/thesofproject/sof/files/3764527/jf_mb3.zip)
one or two samples seems missing.

Hope this help @tlauda . you can reproduce this easily with the command in the description. Alsabat generate 997Hz sinewave by default.

tlauda, 2020-04-29:
@fredoh9 Does this bug still exist? I haven't seen any failure in CI for a long time.

lgirdwood, 2020-05-07:
Ack, lets close.

closed 2020-05-07 by lgirdwood