thesofproject/sof #1989: [BUG] PM status is active for non-existence SDW device

RanderWang, 2019-10-23:
On Comet-lake with 3 SDW codecs, rt715 on link0,  rt1308 on link1,  rt715 on link3 and no device on link2.  After reboot, SDW2 is always active, and this makes SOF is always active.  

The expect behavior is if there is one link without any devices, their pm_runtime status should be marked as ‘Unsupported’, and that should not prevent the system from going to sleep nor should it have any impact on other devices

tlauda, 2019-10-23:
What does it have to do with firmware?

plbossart, 2019-10-23:
yes, it's a pm_runtime bug on the kernel side. @RanderWang please move this to the Dark Side

RanderWang, 2019-10-24:
sorry,  I should created this bug at kernel side. I will move it to kernel side

RanderWang, 2019-10-24:
I made a mistake , it should be reported in kernel side

closed 2019-10-24 by RanderWang