thesofproject/sof #1992: [BUG][TGL] Can't change Dmic capture volume through volume controler

dengyangchao, 2019-10-23:
**Describe the bug**
change Dmic volume through volume controler but it is no change to fact volume
and the Dmic  volume bar seems a boost, adjust the volume once, it gain 10 dB
[image]

**To Reproduce**
1.reboot system via "sudo reboot"
2.wait system to suspend status
3. Arecord via"arecord -Dhw:0,3 -r48000 -c4 -fs32_le test.wav -vv -i" or "arecord -Dhw:0,4 -r48000 - 
        c4 -fs16_le test1.wav -vv -i"
4.use command"alsamixer" to open volume controler and adjust Dmic capture volume

**Reproduction Rate**
10/10

**Expected behavior**
can change Dmic capture volume successfully

**Impact**
can't change Dmic capture volume

**Environment**
kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 7c85d69
FW: https://github.com/thesofproject/sof/commits/master commit: 8b059f6
tplg: Same with TGL FW branch commit: sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg
platform: TGL RVP, with codec ACL711, ALC1308 in SDW mode

<details>
 <summary>Dmesg</summary>

````
[   17.083701]  SDW0-headphone: ASoC: pop wq checking: Playback status: inactive waiting: yes
[   17.083767]  Speaker: ASoC: pop wq checking: Playback status: inactive waiting: yes
[   17.083779]  SSP2-Codec: ASoC: pop wq checking: AIF1 Playback status: inactive waiting: yes
[   79.077377]  DMIC01: ASoC: found 1 audio capture paths
[   79.077384] rt711_rt1308 rt711_rt1308: ASoC: find BE for widget DMIC0.IN
[   79.077386] rt711_rt1308 rt711_rt1308: ASoC: try BE (not set)
[   79.077388] rt711_rt1308 rt711_rt1308: ASoC: try BE ALH3.IN
[   79.077390] rt711_rt1308 rt711_rt1308: ASoC: try BE (not set)
[   79.077391] rt711_rt1308 rt711_rt1308: ASoC: try BE DMIC0.IN
[   79.077394]  DMIC01: connected new DPCM capture path DMIC01 <- dmic01
[   79.077407]  DMIC01: ASoC: found 1 new BE paths
[   79.077409]  dmic01: ASoC: open capture BE dmic01
[   79.099255] sof-audio-pci 0000:00:1f.3: snd_sof_runtime_resume
[   79.099266] sof-audio-pci 0000:00:1f.3: sof_resume
[   79.099288] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   79.099305] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000040 at  00000048
[   79.099324] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   79.102840] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000048
[   79.102855] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000040 at  00000048
[   79.102879] sof-audio-pci 0000:00:1f.3: loading firmware
[   79.102901] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000002 at  00000048
[   79.102908] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000000 at  00000048
[   79.102922] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   79.102928] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000004 at  00000044
[   79.102930] sof-audio-pci 0000:00:1f.3: booting DSP firmware
[   79.103001] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   79.103071] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   79.103079] sof-audio-pci 0000:00:1f.3: period_bytes:0x0
[   79.103081] sof-audio-pci 0000:00:1f.3: periods:1
[   79.103191] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010f0f successful
[   79.103222] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010f0e successful
[   79.103228] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[   79.103232] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
[   79.103927] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
[   79.103944] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[   79.104068] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   79.169438] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
[   79.184454] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
[   79.184464] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[... 793 lines cut ...]
[  104.315752]  dmic01: ASoC: hw_free BE dmic01
[  104.315821]  dmic01: ASoC: close BE dmic01
[  104.315865]  DMIC01: ASoC: close FE DMIC01
[  104.315868] sof-audio-pci 0000:00:1f.3: pcm: close stream 3 dir 1
[  104.315996]  dmic01: ASoC: BE dmic01 event 2 dir 1
[  104.316012]  DMIC01: ASoC: BE capture disconnect check for dmic01
[  104.316015]  DMIC01: freed DSP capture path DMIC01 <- dmic01
[  105.201156] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  105.201175] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  106.618856] sof-audio-pci 0000:00:1f.3: snd_sof_runtime_suspend
[  106.618870] sof-audio-pci 0000:00:1f.3: sof_suspend
[  106.619555] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  106.665451] sof-audio-pci 0000:00:1f.3: sof_send_pm_ipc
[  106.665465] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  106.665727] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  106.665751] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010f0f successful
[  106.665768] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[  106.665772] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[  106.665859] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044

````
</details>

[dmesg.log](https://github.com/thesofproject/sof/files/3761528/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3761529/logger.log)

slawblauciak, 2019-10-29:
Does this occur specifically with SDW? Have you tried I2S?

dengyangchao, 2019-10-30:
tried on GLK Chrome, issue also can reproduce

[image]

**Environment**
platform: GLK Chrome
Firmware: 2ca6a4af https://github.com/thesofproject/sof/commits/master
Kernel: f5e70ce https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware

[amixer.txt](https://github.com/thesofproject/sof/files/3787873/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/3787874/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3787875/logger.log)

dengyangchao, 2019-10-30:
> Does this occur specifically with SDW? Have you tried I2S?

@slawblauciak ICL RVP HDA can adjust the Dmic Volume but it have another issue https://github.com/thesofproject/sof/issues/1756

tlauda, 2019-10-30:
@dengyangchao Please try with #2033.

sinahuang, 2019-10-31:
Verified the issue with master ( commit :59c93383 ) + #2033, original issue can not be reproduced on TGL RVP, with codec ACL711, ALC1308 in SDW mode

Test recipe:
Kernel: https://github.com/thesofproject/linux/tree/integration/soundwire-latest commit: 9f1bd5b 
FW: https://github.com/thesofproject/sof/commits/master commit: 59c93383 + #2033
Tplg: Same with TGL FW branch commit: sof-tgl-rt711-rt1308-nohdmi.tplg rename to sof-tgl-rt711-rt1308.tplg
Platform: TGL RVP, with codec ACL711, ALC1308 in SDW mode

dengyangchao, 2019-10-31:
@tlauda tried #2034 on GLK Chrome
There is a option PGA4.0 to adjust DMIC capture volume, and it also can adjust the actual capture volume.
 But i observed another thing, when adjust the DMIC capture volume to zero, the arecod terminl volume still shows 48%
[image]

[amixer.txt](https://github.com/thesofproject/sof/files/3792526/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/3792527/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/3792528/logger.log)

tlauda, 2019-10-31:
@dengyangchao But doesn't you have different bug for this #1756? If so, then this issue can be closed.

PR #2033 (2019-10-31): topology: sof-tgl-rt711-rt1308: add volume to DMIC pipelines

Fixes #1992.
closed 2019-10-31 by jajanusz