thesofproject/sof #2000: [BUG] CNL: tool-chain: LPS doesn't work in S0ix when building with old 2016.4 tool chain.

keyonjie, 2019-10-24:
**Describe the bug**
LPS doesn't work in S0ix when building with old 2016.4 tool chain, the FW will crash after resumed from S0ix.

**To Reproduce**
1. checkout the cml-008-drop-stable branch commit 569c9495
2. build the FW with XTENSA_CORE="X6H3CNL_2016_4_linux", and XTENSA_TOOLS_VERSION="RF-2016.4-linux", which is the default setting for the building script:
https://github.com/thesofproject/sof/blob/89ef0af07612c0bcf1a85fec3871d0c31e60087b/scripts/xtensa-build-all.sh#L202
3. cp the generate FW binary to the DUT, reboot the DUT.
4. after DUT rebooted, run 'echo freeze > /sys/power/state', the system will enter S0Ix.
5. press a keyboard to wake the system up.
6. dmesg will show you that the FW is crashed.

**Reproduction Rate**
100%

**Expected behavior**
The FW should survive.

**Impact**
Looks the old tool chain is unreliable, or there is bug in the firmware source.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/keyword_detect
    * SOF: cml-008-drop-stable branch commit 569c9495
2) Name of the topology file
    * Topology: any
3) Name of the platform(s) on which the bug is observed.
    * Platform: cnl/cml

dmesg logs as below:

[  896.462899] PM: Finishing wakeup.
[  896.462900] OOM killer enabled.
[  896.462903] Restarting tasks ... done.
[  896.472686] PM: suspend exit
[  896.628006] cml_rt1011_rt5682 cml_rt1011_rt5682: ASoC: resume work completed
[  896.882834] rt5682 i2c-10EC5682:00: jack_type = 0
[  898.019012] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  898.019063] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000
[  898.153031] asix 1-2.2:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
[  898.522387] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[  898.522407] sof-audio-pci 0000:00:1f.3: unknown ROM status value 80000000
[  898.522688] sof-audio-pci 0000:00:1f.3: invalid header size 0xffffffff. FW oops is bogus
[  898.522695] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xecc00302 trace 0x00000000
[  898.522719] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  898.522725] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  898.522742] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  898.522749] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[  898.522792] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  898.522804] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[  898.522810] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  898.522898] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
localhost ~ # packet_write_wait: Connection to 10.54.75.131 port 22: Broken pipe

lgirdwood, 2019-11-04:
@keyonjie is this now fixed with correct toolchain ?

keyonjie, 2019-11-04:
> @keyonjie is this now fixed with correct toolchain ?

Yes, using 2017.8 fixes the issue and the script has been changed to use that, let me close the issue since nobody has schedule to work on the root cause why it fails with 2016.4 tool chain.

closed 2019-11-04 by keyonjie