thesofproject/sof #2002: [BUG][CML][cml-008-drop-stable] ipc error in ipc flood test

aiChaoSONG, 2019-10-25:
**Describe the bug**
running ipc flood test with cml-008-drop-stable branch(fw provided by Poland team), there will be an errer message: `bash: line 0: echo: write error: Invalid argument`, check dmesg, you will see ipc error.

**To Reproduce**
1. reboot system
2. run command `bash -c "echo 10000 > /sys/kernel/debug/sof/ipc_flood_count"
`

**Reproduction Rate**
10/10

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev {c9b2e8f9}
    * SOF: cml-008-drop-stable {N/A}
2) Name of the topology file
    * Topology: {sof-cml-rt5682-max98357a.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML Chrome with Ubuntu1804}

```
[  315.146674] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  315.146890] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40020000: GLB_PM_MSG: CTX_RESTORE
[  315.146901] sof-audio-pci 0000:00:1f.3: ipc tx: 0xb0010000: GLB_TEST_MSG: IPC_FLOOD
[  315.147058] sof-audio-pci 0000:00:1f.3: error: ipc error for 0xb0010000 size 12
[  315.147066] sof-audio-pci 0000:00:1f.3: error: ipc flood test failed at 0 iterations
[  315.640922] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  315.640950] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  316.140891] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  316.140919] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
[dmesg.log](https://github.com/thesofproject/sof/files/3770656/dmesg.log)

***Note: this issue not repoduce with kernel:sof-dev + sof:master***

tlauda, 2019-10-25:
@aiChaoSONG Test IPC is only available when FW is built with DEBUG option enabled in config. Pleasy verify your config.

aiChaoSONG, 2019-10-25:
Closed for not valid

closed 2019-10-25 by aiChaoSONG