thesofproject/sof #2004: [BUG] Glitch and extra buffers output happens in the end of playback with media pipe

singalsu, 2019-10-25:
**Describe the bug**
This bug was noticed when playing back 1s and 30s long low frequency sine wave test clip. Just before the heard output end there's an audible glitch. It appears in recorded audio as a discontinuity in the waveform. The glitch is about 100ms before the end. Also when capturing the analog line-out it is seen that the glitch happens at 1s that is length of playback clip. The continued audio playback after the glitch is apparently copy of old buffers content.

The issue happens when playing back other than 48 kHz content. The glitch looks similar with e.g. 32 kHz an 44.1 kHz content.

The pipeline consists of components PCM - PGA - SRC - Mixer - PGA - DAI. A simpler test pipeline with  PCM - SRC - PGA - DAI does not have this problem.

**To Reproduce**
Use UP^2 board with HifiBerry DAC+ and default topology sof-apl-pcm512x.tplg. Play back wav clip with aplay to device hw:0,4 (card 0: sofbxtpcm512x [sof-bxt-pcm512x], device 4: Media Playback 4 (*) []).  Listen and record the output.

**Reproduction Rate**
Always.

**Expected behavior**
Clean end of playback without annoying glitches.

**Impact**
Annoyance.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 21147939958a35363dfd01563d004094198c917d
    * SOF: ffe904c9f9013b87d2c163eab0ca03ef34560315
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: UP^2 with HiFiBerry DAC+

**Screenshots or console output**
[image]

tlauda, 2019-10-25:
@singalsu Could you verify whether it happens with nocodec on loopback with the same pipeline? It looks like codec problem.

tlauda, 2020-03-03:
@singalsu Ping.

singalsu, 2020-03-04:
@tlauda It's not a codec issue since it depends pipeline and happens with media pipe that inputs other than 48 kHz content. Though this issue is old so I should re-test if this still happens.

tlauda, 2020-04-29:
@singalsu Could you verify whether the bug still exists?

singalsu, 2020-09-24:
I just checked that this issue does not happen with current SOF git repositories:

Kernel Branch: topic/sof-dev
Kernel Commit: 40e44b3a8841
SOF Branch: master
SOF Commit: c4aa4c176

[image]

This issue can be closed!

closed 2020-09-24 by singalsu