thesofproject/sof #2006: [BUG][WoV] draining data is missing in S0ix

keqiaozhang, 2019-10-25:
**Describe the bug**
The draining data will be lost if trigger the WoV in S0ix. When checking the recorded wav file, I found that all the draining data before triggering is missing. Both S16_LE and S24_LE have same issue. But I feel S24_LE is worse, about 1s data after triggering is missing too. 
BTW, there's no obvious errors in dmesg or trace log. Related logs are attached.
I also checked this case in S0, but failed to reproduce such issue.

**To Reproduce**
1. switch to superuser.
2. run wov test via arecord and put it to the background. Do S0ix after 10s, After system is freezed trigger the wov by clapping. 
`$ arecord -Dhw:0,8 -M -N -r 16000 -c 2 -f s24_le  wov.wav -vvv -d 10  & sleep 10 ; echo freeze > /sys/power/state`
3. check the wov.wav via aplay, then you can only hear the sound after clapping.
`aplay -Dplughw:0,0 wov.wav`

**Reproduction Rate**
100%.

**Expected behavior**
The draining data also can be saved.

**Impact**
The draining part is missing.

**Environment**
1) kernel: https://github.com/thesofproject/linux/tree/chrome-4.19-backport commit: a61bd1
2) FW: The sof binary is shared form Firmware team 
Topology: sof-cml-rt1011-rt5682.tplg 
3) Platform: CML Chromebook

[drain-data-missing-dmesg.txt](https://github.com/thesofproject/sof/files/3772171/drain-data-missing-dmesg.txt)
[drain-data-missing-logger-t.txt](https://github.com/thesofproject/sof/files/3772173/drain-data-missing-logger-t.txt)

plbossart, 2019-10-25:
What do you mean by 'draining data'? it seems your wording suggest that we are missing data buffered prior to the wake-up?

keqiaozhang, 2019-10-28:
@plbossart 
Yes, the 2.1s buffer data for wake-up.

lgirdwood, 2019-10-28:
@keqiaozhang, @keyonjie do we have a standard script that we use for WoV testing ? It would be good to add this to our testing directory so that we can all align on the correct and supported usage.

keyonjie, 2019-10-28:
@lgirdwood I have local scripts for that but the are actually 'arecord' only, but need to specify "-M -N" to using mmap mode and non-blocking mode.

keyonjie, 2019-10-30:
@mrajwa can you help debug this with larger buffer-size, e.g. '--buffer-size=64000'?

mrajwa, 2019-11-05:
@lgirdwood, @keqiaozhang  I debugged this issue and here is the summary:

With current setup it looks like the time needed for host to wake up after power down is about 2s. During this time firmware performs draining. So when host buffer is capable of keeping only one second of data, by the time its up and running firmware already did overwrite history data with real time stream. That's why this issue is reporting "missing data". The only sane solution here is to increase host buffer size. Good news is we contacted client and they are using host buffer which is four times bugger then the one we use for tests.

@lgirdwood, @keqiaozhang  can we therefore close this issue?

lgirdwood, 2019-11-05:
@mrajwa I guess so, but we should specify this constrain in the topology file alongside perhaps similar comments in the code this way it obvious to anyone using the feature.

mrajwa, 2019-11-05:
@lgirdwood, I agree for the constrain in fact this is what I am going to do next. However I feel like this constrain should end up in the FW namely KPBs function validate_host_params which is run on hw_params(). The reason for that is host buffer is dynamic value, in user space one can specify it as he whish and this value propagates down the line to FW in the form of host_buffer_size. 
Does it make sense for you?

lgirdwood, 2019-11-05:
@mrajwa host buffer min/max size is set in topology, so can be set dynamically by applications oroviding if falls within min/max.

mrajwa, 2020-05-18:
@keqiaozhang @mengdonglin can we close this issue?

slawblauciak, 2020-08-11:
Buffer size increased accordingly and issue has been solved.

closed 2020-08-11 by slawblauciak