thesofproject/sof #2007: [BUG][WoV] WoV is unable to handle the extreme period_sizes

keqiaozhang, 2019-10-25:
**Describe the bug**
Looks like WoV is unable to handle the extreme period_size. 
For S16_LE, the range of processing is (48, 4032), if period size is bigger than 4032, then trigger failure. if period size is less than 48, xrun and IPC errors occur. but wov still can be run in next time. 
The range for S24_LE is (48 , 2040), if period size is bigger than 2040, DSP firmware oops occurs. If period is less than 48, then xrun and IPC errors occur. 

**To Reproduce**
1. S16_LE :
`arecord -Dhw:0,8 -M -N -r 16000 -c 2 -f s16_le  --period-size=$value wov.wav -vvv `
2. S24_LE:
`arecord -Dhw:0,8 -M -N -r 16000 -c 2 -f s16_le  --period-size=$value wov.wav -vvv`

**Reproduction Rate**
100%

**Expected behavior**
Ideally, WoV should handle various period-sizes.

**Impact**
WoV trigger failure or kinds of IPC errors

**Environment**
1. kernel: https://github.com/thesofproject/linux/tree/chrome-4.19-backport commit: a61bd1
2. FW: The sof binary is shared form Firmware team
3. Topology: sof-cml-rt1011-rt5682.tplg
4. Platform: CML Chromebook
 
**dmesg**
1.  S24_LE: period-size < 48
```
[  232.534540] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  232.534860] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  232.534866] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[  232.534870] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  232.535059] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60030000 size 12
[  232.535064]  DMIC16k: ASoC: trigger FE cmd: 0 failed: -16
[  232.535170] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 8 dir 1 cmd 1
[  232.535389] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x24001e successful
[  232.535396] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
[  232.535598] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60040000 size 12
[  232.535603]  DMIC16k: ASoC: trigger FE cmd: 1 failed: -22
[  232.535608] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 8 dir 1 cmd 0
[  232.535611] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  232.535791] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[  232.535798] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[  232.535800]  DMIC16k: ASoC: trigger FE cmd: 0 failed: -22
[  232.535848] sof-audio-pci 0000:00:1f.3: pcm: free stream 8 dir 1
[  232.535851] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  232.537677] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  232.537734] sof-audio-pci 0000:00:1f.3: pcm: close stream 8 dir 1
```
2. S24_LE: period_size > 2040
```
[  216.053100] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 8 dir 1 cmd 0
[  216.053112] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  216.053518] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  216.053532] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[  216.053542] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  216.053736] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60030000 size 12
[  216.053750]  DMIC16k: ASoC: trigger FE cmd: 0 failed: -16
[  216.053812] sof-audio-pci 0000:00:1f.3: pcm: free stream 8 dir 1
[  216.053820] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  216.053849] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[  216.053861] sof-audio-pci 0000:00:1f.3: panic: dsp_oops_offset 788480 offset 788480
[  216.053868] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  216.053970] sof-audio-pci 0000:00:1f.3: error: runtime exception
[  216.053974] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  216.053978] sof-audio-pci 0000:00:1f.3: error: panic at :0
[  216.053983] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
```
**Trace**
1.  S24_LE: period-size < 48
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM         277600598.593750 277600608.000000      src/lib/alloc.c:273    error: 155 blocks needed for allocation but only 154 blocks are remaining
    0      1         PIPE         290659270.677083  13058672.000000 src/audio/pipeline.c:757    pipeline_reset() error: ret = -16, host->comp.id = 40
    0      1         COMP         290659930.520833       659.843750 rc/audio/component.c:146    comp_set_state() error: wrong state = 1, COMP_TRIGGER_START
    0      1          IPC         290659948.750000        18.229166 src/audio/pipeline.c:701    pipeline_trigger() error: ret = -22, host->comp.id = 40, cmd = 1
    0      1          IPC         290659968.020833        19.270834    src/ipc/handler.c:489    ipc: comp 40 trigger 0x40000 failed -22
    0      1         COMP         290660170.156250       202.135422 rc/audio/component.c:167    comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
    0      1          IPC         290660188.333333        18.177084 src/audio/pipeline.c:701    pipeline_trigger() error: ret = -22, host->comp.id = 40, cmd = 0
    0      1          IPC         290660207.343750        19.010416    src/ipc/handler.c:489    ipc: comp 40 trigger 0x
```
2. S24_LE: period_size > 2040
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME    CONTENT
    0      1          MEM         265245499.843750 265245504.000000      src/lib/alloc.c:273    error: 155 blocks needed for allocation but only 154 blocks are remaining
    0      1         PIPE         270058796.510417   4813296.500000 src/audio/pipeline.c:757    pipeline_reset() error: ret = -16, host->comp.id = 40
    0      2          KPB         269741051.197917              nan      src/audio/kpb.c:1098   kpb: update history_depth by 112640
    0      2          IPC         270058174.270833    317123.062500    src/ipc/handler.c:455    ipc: comp 40 -> trigger cmd 0x50000
    0      2         PIPE 8.47    270058232.916667        58.645832 src/audio/pipeline.c:674    pipeline_trigger()
    0      2         HOST 8.40    270058246.197917        13.281250     src/audio/host.c:283    host_trigger()
    0      2          DMA         270058260.416667        14.218750 intel/cavs/hda-dma.c:538    hda-dmac: 4 channel 1 -> stop
    0      2          KPB         270058276.510417        16.093750      src/audio/kpb.c:322    kpb_trigger()
    0      2       VOLUME         270058291.145833        14.635417 udio/volume/volume.c:531    volume_trigger()
    0      2          DAI 8.46    270058305.625000        14.479167      src/audio/dai.c:494    dai_comp_trigger(), command = 0
    0      2          DAI 8.46    270058318.385417        12.760417      src/audio/dai.c:558    dai_comp_trigger(), PAUSE/STOP
    0      2          DMA         270058332.187500        13.802083 src/drivers/dw/dma.c:353    dw_dma_stop(): dma 0 channel 2 stop
    0      2         DMIC         270058348.229167        16.041666 rs/intel/cavs/dmic.c:1442   dmic_trigger()
    0      2         DMIC         270058360.364583        12.135417 rs/intel/cavs/dmic.c:1367   dmic_stop()
    0      2         DMIC         270058374.895833        14.531250 rs/intel/cavs/dmic.c:1393   dmic_stop(), dmic_active_fifos = 1
    0      2          IPC         270058692.343750       317.447906    src/ipc/handler.c:344    ipc: comp 40 -> free
    0      2         PIPE 8.47    270058742.708333        50.364582 src/audio/pipeline.c:752    pipeline_reset()
    0      2         HOST 8.40    270058755.572917        12.864583     src/audio/host.c:717    host_reset()
    0      2          KPB         270058782.708333        27.135416      src/audio/kpb.c:475    kpb_reset(): resetting from state 6, state log 56565656
    0      1         PIPE         270058796.510417        13.802083 src/audio/pipeline.c:757    pipeline_reset() error: ret = -16, host->comp.id = 40
```
[wov-dmesg_s24_lower.txt](https://github.com/thesofproject/sof/files/3772363/wov-dmesg_s24_lower.txt)
[wov-dmesg_s24_upper.txt](https://github.com/thesofproject/sof/files/3772364/wov-dmesg_s24_upper.txt)
[wov-s24_lower_sof_err.txt](https://github.com/thesofproject/sof/files/3772365/wov-s24_lower_sof_err.txt)
[wov-s24_upper_sof_error.txt](https://github.com/thesofproject/sof/files/3772366/wov-s24_upper_sof_error.txt)
[wov-s24_upper_sof_trace.txt](https://github.com/thesofproject/sof/files/3772368/wov-s24_upper_sof_trace.txt)

paulstelian97, 2019-10-28:
`    0      1          MEM         277600598.593750 277600608.000000      src/lib/alloc.c:273    error: 155 blocks needed for allocation but only 154 blocks are remaining`

Is this line relevant?

lgirdwood, 2019-10-28:
@paulstelian97 yep, not enough memory, but we also have a bug with the IPC timeout here.
@keqiaozhang period size limited by available free memory and topology constraints for PCM.

paulstelian97, 2019-10-29:
Funny how -16 means -EBUSY, at least according to errno-base.h on my system. Reset returning -EBUSY wasn't special case behavior for kpb reset?

lgirdwood, 2019-11-05:
@keqiaozhang can you confirm 
1) period size used by test is permitted in the topology (please also check buffer size is permitted too, based on number of periods * period size).
2) DSP panic still present ?
I suspect here we need to constrain the params in topology for memory size and latency.

tlauda, 2020-04-29:
@keqiaozhang Could you reproduce the issue with current master? There were many changes in KPB. Probably some of the reported problems don't reproduce anymore.

plbossart, 2021-06-10:
@lgirdwood @keqiaozhang I don't think this bug is still relevant, keyword detection was not enabled in CML products.

keqiaozhang, 2021-06-11:
@plbossart , yes. this bug is is invalid for CML, close it.

closed 2021-06-11 by keqiaozhang