thesofproject/sof #2010: [BUG][APL] KD with SSP - IPC timeout on TplgCompFree on DAI component

zrombel, 2019-10-25:
**Describe the bug**
When running test Keyword Detection with SSP interface on APL RVP, when whole test is passed, when sending TplgCompFree to DAI, IPC timeouts. 

**To Reproduce**
Run test scripts with slim driver
test cases:
tests/sof/fw_07_kd/test_02_kd_with_ssp.py::TestSspKd16000Hz16b16b2ch

**Reproduction Rate**
50%

**Expected behavior**
FW should free DAI component and send response IPC

**Environment**
FW master branch 89ef0af

mrajwa, 2019-11-05:
@zrombel, can I have an error log for that? If I understand correctly test has passed but you see IPC timeout at the very end right?

jdrzaszcz, 2020-03-13:
Can't reproduce on Sof master branch: 15736c57667e541e619ec82618c322b7977f5fbd

closed 2020-03-13 by jdrzaszcz