thesofproject/sof #2011: [BUG][WHL] KD with SSP - IPC response error -16 on StreamPmcFree

zrombel, 2019-10-25:
**Describe the bug**
When running test Keyword Detection with SSP interface on WHL RVP, after stoping all streams, when sending StreamPmcFree to pipeline with KPB component FW responds with IPC error -16

**To Reproduce**
Run test scripts with slim driver
test cases:
tests/sof/fw_07_kd/test_02_kd_with_ssp.py::TestSspKd16000Hz24b32b2ch

**Reproduction Rate**
10%

**Expected behavior**
Steam should be freed without IPC error

**Environment**
FW master branch 89ef0af

**Trace Log**
[07_02_TestSspKd16000Hz24b32b2ch_trace.log](https://github.com/thesofproject/sof/files/3772677/07_02_TestSspKd16000Hz24b32b2ch_trace.log)

mrajwa, 2019-11-05:
@zrombel, KPB returned EBUSY because you attempted to reset it when it was impossible in single step. Such condition can occur when KPB is performing draining or buffering. You should really resend the StreamPmcFree() if previous one returned BUSY. Does it make sense to you?

lgirdwood, 2019-11-05:
@zrombel @mrajwa can we update the script to cater for this situation and have some comment that desribes this are normal behaviour given the circumstances.

zrombel, 2019-11-06:
@mrajwa But how KPB can be buffering or draining when previously we stopped all streams and nothing should be copied through pipelines any more, so can KPB be in draining or buffering state ?

mrajwa, 2019-11-06:
@zrombel, because your stream stop does trigger pipeline_reset() and I see in logs:
	

> kpb_reset(): resetting from state 6, state log ‭56565656‬
>         pipeline_reset() error: ret = -16, host->comp.id = 6

So as you can see at the time piepline_reset() was requested kpb was constantly changing its state between 5 and 6 - meaning it was draining and buffering back and forth.
Currently I am in a discussion on kernel repo as to whether we should repeat IPC message if DSP was busy or ignore it or maybe return success even if reset needs more time...

zrombel, 2019-11-06:
@mrajwa I wonder if the root cause is somewhere else. I've checked time stamp of kpb_notification, draining_done and trigger_pause ipc. In correct iterations maximum time between kpb_notification and draining_done is 1285ms, and time between kpb_notification and trigger_pause ipc is ~1460ms so there is a minimum 180ms gap. In failed iteration draining in not completed in 1460ms.
And there are two questions:
1) Why sometimes draining is taking so long?
2) Does KPB need to end draining to be properly reset/freed? If so maybe host should be notified when draining is done?

mrajwa, 2019-11-06:
1. With your FW version draining last long because at that time we implemented it with assumption of 1 second host buffer. This made us prone to XRUNS so we had to limit the draining speed accordingly.
Newer FW version make use of 4 second host buffer so draining is much faster - aprox < 100 ms.

2. No, you can reset KPB during draining however we also want to avoid data corruption because of that. So what we is we register reset request and do it at right point so no corruption occur.

I will probably return success to the very first reset request instead of BUSY - quick test with full stack shows it doesn't cause any problem.

tlauda, 2020-03-12:
Fixed some time ago.

closed 2020-03-12 by tlauda