thesofproject/sof #2013: [BUG][ICL][HDA] It tips"Trying resume.Failed.Restarting stream.Done.....write error:Input/out error" after suspend/resume at audio pause state when aplaying audio

Liviali155, 2019-10-28:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/howtos/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

tlauda, 2019-10-28:
@Liviali155 What is this? No description, no logs, nothing...

Liviali155, 2019-10-28:
> @Liviali155 What is this? No description, no logs, nothing...

I am sorry,I have updated it

tlauda, 2019-10-28:
@plbossart @ranj063 State machine transitions are messed up in this one. Pipeline is in COMP_STATE_PAUSED in firmware, but instead of COMP_TRIGGER_RELEASE the kernel sends COMP_TRIGGER_START. Either you should send COMP_TRIGGER_RELEASE or COMP_TRIGGER_STOP/HW params/COMP_TRIGGER_START.
@Liviali155 Please move this issue to kernel.

closed 2019-10-28 by tlauda