thesofproject/sof #2014: [CML] Plug in Two TypeC HDMI/DP simultaneously, no audio playback on neither monitors.

macchian, 2019-10-28:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
The current Pin Mux settings we set on Chrome, it works with MST audio hub. The sound can play separately when switching audio between display monitors.

		cset "name='Pin5-Port0 Mux' 1"
		cset "name='Pin5-Port1 Mux' 2"
		cset "name='Pin5-Port2 Mux' 3"
		cset "name='Pin6-Port0 Mux' 1"
		cset "name='Pin6-Port1 Mux' 2"
		cset "name='Pin6-Port2 Mux' 3"
		cset "name='Pin7-Port0 Mux' 1"
		cset "name='Pin7-Port1 Mux' 2"
		cset "name='Pin7-Port2 Mux' 3"

However, the pin mux is not compatible to SST audio mode.
Instead I modified the Pin mux as below, it works for SST mode as the issue report. 
But it's not working for MST audio.

		cset "name='Pin5-Port0 Mux' 2"
		cset "name='Pin5-Port1 Mux' 2"
		cset "name='Pin5-Port2 Mux' 2"
		cset "name='Pin6-Port0 Mux' 1"
		cset "name='Pin6-Port1 Mux' 1"
		cset "name='Pin6-Port2 Mux' 1"
		cset "name='Pin7-Port0 Mux' 3"
		cset "name='Pin7-Port1 Mux' 3"
		cset "name='Pin7-Port2 Mux' 3"

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)
1. plug in typeC DP cable
2. confirm play sound working on display 1.
3. plug in typeC HDMI cable
4. display 2 audio didn't show up in Chrome UI. Still display 1 exist alone.
5. no audio playback from display 1 & 2.

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
100%

**Expected behavior**
A clear and concise description of what you expected to happen.
Sound can playback when switching audio between display 1 and display 2.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
no sound playback in SST audio mode.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 4.19
    * SOF: 0.7
2) Name of the topology file
    * Topology: sof-cml-da7219-max98357a.tplg , sof-cml-rt5682-max98357a.tplg

3) Name of the platform(s) on which the bug is observed.
    * Platform: CML

plbossart, 2019-10-28:
what does 'SST mode" mean? Is this a typo for MST, or did you mean the SKL driver?

lgirdwood, 2019-10-28:
@macchian looks like a driver bug and not a FW bug ? (you have logged under FW)

macchian, 2019-10-29:
> what does 'SST mode" mean? Is this a typo for MST, or did you mean the SKL driver?
SST here is Single Stream Transport (SST). 
By plugging in two TypeC DP/HDMI cable(SST) sequentially, I saw the errors as below:
[  407.066262] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  407.100519] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:0
[  407.113769] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:0
[  407.114286] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:1
[  407.114783] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 7:1
[  407.115035] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 5:2
[  407.115529] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 7:2

[  423.349112] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_present_sense: disconnect for pin:port 6:0

[hdmi_dmesg.txt](https://github.com/thesofproject/sof/files/3782478/hdmi_dmesg.txt)

macchian, 2019-10-29:
> @macchian looks like a driver bug and not a FW bug ? (you have logged under FW)

I'm not sure, I assume it's driver issue when enabled more logs in hdac_hdmi.c. and found the errors as below:

2019-10-28T11:03:01.085470+00:00 DEBUG kernel: [  163.215341] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_cvt_output_widget_event: widget: Converter 2 event: 1
2019-10-28T11:03:01.085472+00:00 DEBUG kernel: [  163.216047] HDMI HDA Codec ehdaudio0D2: num_mux_nids 3 for pin:port 5:0
2019-10-28T11:03:01.085473+00:00 DEBUG kernel: [  163.216054] HDMI HDA Codec ehdaudio0D2: hdac_hdmi_verify_connect_sel_all_pins: cvt 2 set connect 5 -> 0
2019-10-28T11:03:01.085474+00:00 DEBUG kernel: [  163.216296] HDMI HDA Codec ehdaudio0D2: num_mux_nids 3 for pin:port 5:1
2019-10-28T11:03:01.085474+00:00 WARNING kernel: [  163.216546] HDMI HDA Codec ehdaudio0D2: No connections found for pin:port 6:1

[messages.txt](https://github.com/thesofproject/sof/files/3782539/messages.txt)

Any suggestions to address the issue?

plbossart, 2019-10-29:
@libinyang @kv2019i is this really possible to have the same mixer settings with MST and two connectors (aka SST)?  And is this really an SOF-specific issue or all drivers would have the same issue, right?

libinyang, 2019-10-29:
> @libinyang @kv2019i is this really possible to have the same mixer settings with MST and two connectors (aka SST)? And is this really an SOF-specific issue or all drivers would have the same issue, right?

@plbossart 
It depends on the HW whether it is possible to have the same mixer setting with MST and 2 connectors (aka SST). And there are conflicts in most cases. This is one of the reasons why we want to switch to legacy hdmi audio driver.

The below setting is used for DPMST (This is not a good setting because it easily causes PCM assignment conflict. Let's image one monitor is connected pin5-port0, It will use the Mux 1. And another monitor is connected pin6-port0. It will also use Mux 1. This is wrong).
```
	cset "name='Pin5-Port0 Mux' 1"
	cset "name='Pin5-Port1 Mux' 2"
	cset "name='Pin5-Port2 Mux' 3"
	cset "name='Pin6-Port0 Mux' 1"
	cset "name='Pin6-Port1 Mux' 2"
	cset "name='Pin6-Port2 Mux' 3"
	cset "name='Pin7-Port0 Mux' 1"
	cset "name='Pin7-Port1 Mux' 2"
	cset "name='Pin7-Port2 Mux' 3"
```

And the below setting is used for SST
```
 	cset "name='Pin5-Port0 Mux' 2"
	cset "name='Pin5-Port1 Mux' 2"
	cset "name='Pin5-Port2 Mux' 2"
	cset "name='Pin6-Port0 Mux' 1"
	cset "name='Pin6-Port1 Mux' 1"
	cset "name='Pin6-Port2 Mux' 1"
	cset "name='Pin7-Port0 Mux' 3"
	cset "name='Pin7-Port1 Mux' 3"
	cset "name='Pin7-Port2 Mux' 3"
```

@plbossart @lgirdwood @macchian 
From the description, this should not be a FW driver. It should be the driver issue, or more likely the amixer setting issue.

kv2019i, 2019-10-29:
Ack on @libinyang's explanation, this is a driver issue. Recommend fix is to move to common HDMI driver (PR1155). @yongzhi1 already tested it on Chrome devices.

tlauda, 2019-10-30:
@macchian Please move this bug to https://github.com/thesofproject/linux repo. Thanks!

closed 2019-10-30 by tlauda

macchian, 2019-11-04:
Hi @yongzhi1 , we are trying to backport the 

> @macchian Please move this bug to https://github.com/thesofproject/linux repo. Thanks!

updated to the existing issue PR1426.